microsoft/AirSim #3213: Mavsdk python package with AirSim simulator

AIKUUM, 2020-12-14:
## Question
### What's your question?

### Include context on what you are trying to achieve

#### Context details

### Include details of what you already did to find answers

Below screenshot represents the details of my system:
[image]

AirSim version (1.3.1)
Unreal version(4.25)
python version (3.8.3)

Im using AirSim simulator for my test cases. As of now I want communication to happen between 2 drones in an automated simulation environment of AirSim. I also want to know if mavsdk python package can be used along with airsim. 

Details of what I already tried to check if Mavsdk python package can be used along with AirSim simulator:
1) I searched for "mavsdk+airsim" in google. I could then access the link "https://dev.px4.io/v1.9.0/en/simulation/airsim.html". But I couldnt find anything related to combination of airsim and mavsdk. I couldnt find anything related to combination of airsim and mavsdk even in the link "https://github.com/microsoft/AirSim/blob/master/docs/px4_sitl.md".
2) I searched for "mavsdk" in the main search of "https://microsoft.github.io/AirSim/", but i couldnt find anything in it.

AIKUUM, 2020-12-15:
> ## Question
> ### What's your question?
> ### Include context on what you are trying to achieve
> #### Context details
> ### Include details of what you already did to find answers
> Below screenshot [...]

Hi @sytelus, Which python package is better for communication between 2 drones in AirSim? Mavros or Mavsdk?

AIKUUM, 2020-12-19:
> ## Question
> ### What's your question?
> ### Include context on what you are trying to achieve
> #### Context details
> ### Include details of what you already did to find answers
> Below screenshot [...]

Hi @rajat2004, Do you have any idea about this?

ahmed-elsaharti, 2020-12-21:
Hello @AIKUUM , I've done some testing and can confirm that mavsdk's example scripts (I tested `takeoff_and_land.py`) will work with Airsim in PX4 SITL mode (haven't tested HITL yet).
All I had to do to make this work was set the `ControlPort` in [settings.json](https://github.com/microsoft/AirSim/blob/master/docs/settings.md) to PX4's Onboard port:
[image]
then run [PX4 SITL](https://github.com/microsoft/AirSim/blob/master/docs/px4_sitl.md) normally.

Here's my settings.json for reference:
```json
{
  "SettingsVersion": 1.2,
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SimMode": "Multirotor",
  "DefaultVehicleConfig": "PX4",
  "Vehicles": {
    "PX4": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPort": 14580
        }
  }
}
```

and here's a screenshot of the example script connecting and running:
[image]

Hope this helps

jonyMarino, 2020-12-21:
Hi @AIKUUM! You are continuing the topic of #3145 here, so I suggest you close there.
The messages that happen by mavlink between AirSim and PX4 are for simulation. Specifically for PX4 to simulate drone control. It is not what you are looking for for your use case. If you want to use mavsdk, instantiate a mavlink node for each drone and communicate between them. You would do all this from your script. For example, you can have a thread for each drone, and each thread has a mavlink node. Within each thread, you will make calls to the AirSim API for the drone that corresponds to it (you will specify the drone's name each time you call the API).

AIKUUM, 2020-12-22:
> Hi @AIKUUM! You are continuing the topic of #3145 here, so I suggest you close there.
> The messages that happen by mavlink between AirSim and PX4 are for simulation. Specifically for PX4 to simulate [...]

Hi @jonyMarino, I have closed the issue #3145

AIKUUM, 2020-12-22:
> Hi @AIKUUM! You are continuing the topic of #3145 here, so I suggest you close there.
> The messages that happen by mavlink between AirSim and PX4 are for simulation. Specifically for PX4 to simulate [...]

Hi @jonyMarino, I'm sorry for disturbing you again. But is PX4 in SITL compulsorily required along with AirSim for UAV communication? or just AirSim is sufficient?

jonyMarino, 2020-12-22:
Just AirSim is sufficient.

AirSim uses mavlink to integrate the PX4 flight controller. But what you want is to use mavlink for other purposes. You would not have requirements from the AirSim side because you will simulate the communication on the client-side (your script). You are doing this for research, or do you plan to deploy this in a real drone?

AIKUUM, 2020-12-23:
> Just AirSim is sufficient.
>
> AirSim uses mavlink to integrate the PX4 flight controller. But what you want is to use mavlink for other purposes. You would not have requirements from the AirSim side [...]

What I want to do is for a project.

AIKUUM, 2020-12-23:
> Hello @AIKUUM , I've done some testing and can confirm that mavsdk's example scripts (I tested `takeoff_and_land.py`) will work with Airsim in PX4 SITL mode (haven't tested HITL yet).
> All I had to [...]

Hi @ahmed-elsaharti, How are you using mavsdk? I mean did you install the module through pip? Or did u download mavsdk-python package from https://github.com/mavlink/MAVSDK-Python? I have done both. Can you suggest me which would be better to use?

ahmed-elsaharti, 2020-12-23:
> Hi @ahmed-elsaharti, How are you using mavsdk? I mean did you install the module through pip? Or did u download mavsdk-python package from https://github.com/mavlink/MAVSDK-Python? I have done both. Can you suggest me which would be better to use?

I remember installing it through pip and cloning the https://github.com/mavlink/MAVSDK-Python repo just for the example scripts. The general difference between the pip version and the github version of any library would be that the version on github is the one being updated with every commit while the pip one awaits publishing by the creator

AIKUUM, 2020-12-23:
> Hello @AIKUUM , I've done some testing and can confirm that mavsdk's example scripts (I tested `takeoff_and_land.py`) will work with Airsim in PX4 SITL mode (haven't tested HITL yet).
> All I had to [...]

I tried executing the same python file (takeoff_and_land.py) as you did with PX4 in SITL. But there are some issues with the execution as in the screenshot below:
[image]
As you can see in Anaconda prompt "Waiting for drone to connect..." is there and then nothing is displayed. Morover there are some errors in PX4 too.

These are settings that I have used in settings.json file:
`
{
 "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
 "SettingsVersion": 1.2,
 "SimMode": "Multirotor",
 "ClockSpeed": 1,
 "LocalHostIp": "127.0.0.1",
 "CameraDefaults": {
  "CaptureSettings": [
   {
    "ImageType": 0,
    "Width": 3000,
    "Height": 2000,
    "FOV_Degrees": 90,
    "AutoExposureSpeed": 100,
    "AutoExposureBias": 0,
    "AutoExposureMaxBrightness": 0.64,
    "AutoExposureMinBrightness": 0.03,
    "MotionBlurAmount": 0,
    "TargetGamma": 1.0,
    "ProjectionMode": "",
    "OrthoWidth": 5.12
   }
  ]
 },
 "Vehicles": {
  "Drone1": {
   "VehicleType": "PX4Multirotor",
   "UseSerial": false,
   "UseTcp": true,
   "TcpPort": 4560,
   "ControlPort": 14580,
   "Parameters": {
    "NAV_RCL_ACT": 0,
    "NAV_DLL_ACT": 0,
    "LPE_LAT": 47.641468,
    "LPE_LON": -122.140165,
    "COM_OBL_ACT": 1
   },
   "X": 0,
   "Y": 0,
   "Z": 0
  }
 }
}
`

Can you help me to resolve the issues?

ahmed-elsaharti, 2020-12-23:
@AIKUUM try getting rid of the 
```json
"Parameters": { "NAV_RCL_ACT": 0, "NAV_DLL_ACT": 0, "LPE_LAT": 47.641468, "LPE_LON": -122.140165, "COM_OBL_ACT": 1 },
```
section in your settings.json and make sure the anaconda prompt is not blocked by a firewall or an antivirus

AIKUUM, 2020-12-23:
> @AIKUUM try getting rid of the
> 
> ```json
> "Parameters": { "NAV_RCL_ACT": 0, "NAV_DLL_ACT": 0, "LPE_LAT": 47.641468, "LPE_LON": -122.140165, "COM_OBL_ACT": 1 },
> ```
> 
> section in your settings.json and make sure the anaconda prompt is not blocked by a firewall or an antivirus

Hi @ahmed-elsaharti, I tried what you suggested. But nothing appears after "Waiting for drone to connect..." in Anaconda Prompt. 
And this is what thats displayed in PX4:
`
px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2020-12-23/06_34_26.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
INFO  [ecl/EKF] 3423094: EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] 7970499: EKF commencing GPS fusion
`
Nothing appears in PX4 after this.

ahmed-elsaharti, 2020-12-23:
What is the port on line 10 of the takeoff-land example script?

ahmed-elsaharti, 2020-12-23:
If its `udp://:14540` change it to `udp://:14550`

AIKUUM, 2020-12-24:
> If its `udp://:14540` change it to `udp://:14550`
> 
> Edit: or change the `ControlPort` in settings.json to 14570

Hi @ahmed-elsaharti, I tried exactly as you have suggested. 

But I'm still getting these errors in PX4:
`
ERROR [sensors] Accel #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Accel #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Accel #0 priority: 1
ERROR [sensors] Accel #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Accel #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Accel #0 priority: 1
ERROR [sensors] Gyro #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Gyro #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Gyro #0 priority: 1
ERROR [sensors] Accel #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Accel #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Accel #0 priority: 1
ERROR [sensors] Gyro #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Gyro #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Gyro #0 priority: 1
`

In Anaconda prompt, this is all that gets displayed. And then nothing happens even in the Blocks environment. 

settings.json contents:
`
{
 "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
 "SettingsVersion": 1.2,
 "SimMode": "Multirotor",
 "ClockSpeed": 1,
 "LocalHostIp": "127.0.0.1",
 "CameraDefaults": {
  "CaptureSettings": [
   {
    "ImageType": 0,
    "Width": 3000,
    "Height": 2000,
    "FOV_Degrees": 90,
    "AutoExposureSpeed": 100,
    "AutoExposureBias": 0,
    "AutoExposureMaxBrightness": 0.64,
    "AutoExposureMinBrightness": 0.03,
    "MotionBlurAmount": 0,
    "TargetGamma": 1.0,
    "ProjectionMode": "",
    "OrthoWidth": 5.12
   }
  ]
 },
 "Vehicles": {
  "Drone1": {
   "VehicleType": "PX4Multirotor",
   "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPort": 14570,
   "X": 4,
   "Y": 0,
   "Z": 0
  }
 }
}
`

Contents of takeoff_and_land.py:
`
#!/usr/bin/env python3

import asyncio
from mavsdk import System

async def run():

    drone = System()
    await drone.connect(system_address="udp://:14550")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break

    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok:
            print("Global position estimate ok")
            break

    print("-- Arming")
    await drone.action.arm()

    print("-- Taking off")
    await drone.action.takeoff()

    await asyncio.sleep(5)

    print("-- Landing")
    await drone.action.land()

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())

`

Can you please help me? I'm stuck with this issue for enough time now.

ahmed-elsaharti, 2020-12-24:
> If its `udp://:14540` change it to `udp://:14550`
> 
> Edit: or change the `ControlPort` in settings.json to 14570

@AIKUUM as I mentioned, you need to do **ONE** of them or the other. What you did was that you changed both the control port and the port in mavsdk.

It should be either:
A. `udp://:14540` in the python script + `"ControlPort": 14570` in settings.json
**OR**
B. `udp://:14550` in the python script + `"ControlPort": 14580` in settings.json
for mavsdk to work

AIKUUM, 2020-12-24:
> `udp://:14540` in the python script + `"ControlPort": 14570` in settings.json

Hi @ahmed-elsaharti, I followed procedure A. Execution of "takeoff_and_land.py" is successful. In PX4, I can also see takeoff detection and landing logs. 
But at the same time in between the successful logs, I'm still getting errors in PX4:
ERROR [sensors] Accel #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Accel #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Accel #0 priority: 1
ERROR [sensors] Gyro #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Gyro #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Gyro #0 priority: 1

Is it ok to get these errors?

AIKUUM, 2020-12-24:
> Just AirSim is sufficient.
>
> AirSim uses mavlink to integrate the PX4 flight controller. But what you want is to use mavlink for other purposes. You would not have requirements from the AirSim side [...]

Hi @jonyMarino, But I'm not sure about what parameters I need to consider to differentiate between 2 drones (IP address, port numbers etc). Can you suggest?

ahmed-elsaharti, 2020-12-24:
> Hi @ahmed-elsaharti, I followed procedure A. Execution of "takeoff_and_land.py" is successful. In PX4, I can also see takeoff detection and landing logs.
> But at the same time in between the [...]

These are probably more related to #3236. As long as the flight is not affected I wouldn't think they're an issue. If you get an error that causes the drone to lose GPS and crash (I've been getting those recently with SITL) that would be problematic.

AIKUUM, 2020-12-24:
Hi @ahmed-elsaharti, what changes can I make in "takeoff_and_land.py", so that both the drones takeoff, when 2 drones are configured in settings.json?

ahmed-elsaharti, 2020-12-24:
@AIKUUM tbh I wouldn't really know since I haven't looked into it that much but this would be a very good question for the folks over at the [Mavsdk repo](https://github.com/mavlink/MAVSDK-Python). However, I do know you to set up two drones in SITL you'll need to run multiple instances of SITL in addition to setting them up in settings.json (ie: having two drones in settings.json but one SITL instance wont work, you'll need one SITL instance per drone). Have a look at #3222 for some docs on how to do that.

AIKUUM, 2020-12-24:
> @AIKUUM tbh I wouldn't really know since I haven't looked into it that much but this would be a very good question for the folks over at the [Mavsdk repo](https://github.com/mavlink/MAVSDK-Python). However, I do know that for you to set up two drones in SITL you'll need to run multiple instances of SITL in addition to setting them up in settings.json (ie: having two drones in settings.json but one SITL instance wont work, you'll need one SITL instance per drone). Have a look at #3222 for some docs on how to do that.

Hi @ahmed-elsaharti, I tried this in PX4 console:
$ ./Tools/sitl_multiple_run.sh 2
killing running instances
starting instance 0 in /cygdrive/e/PX4.Windows.Cygwin.Toolchain.0.8/PX4.Windows.Cygwin.Toolchain.0.8/home/PX4/Firmware/build/px4_sitl_default/instance_0
starting instance 1 in /cygdrive/e/PX4.Windows.Cygwin.Toolchain.0.8/PX4.Windows.Cygwin.Toolchain.0.8/home/PX4/Firmware/build/px4_sitl_default/instance_1

But nothing happens in Anaconda prompt after this:
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...

Can you help me?

ahmed-elsaharti, 2020-12-24:
Was airsim startedbafter this and was it able to connect to the px4 sitl instances?

AIKUUM, 2020-12-24:
> Was airsim started after this and was it able to connect to the px4 sitl instances?

Hi @ahmed-elsaharti, I opened PX4 console first and did this:
[image]

And then, I opened Blocks environment and these logs were present on the screen:
[image]

And then i executed the python script.

ahmed-elsaharti, 2020-12-24:
What does your settings.json look like?

AIKUUM, 2020-12-25:
> What does your settings.json look like?

Hi @ahmed-elsaharti, contents of settings.json:
`
{
 "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
 "SettingsVersion": 1.2,
 "SimMode": "Multirotor",
 "ClockSpeed": 1,
 "LocalHostIp": "127.0.0.1",
 "CameraDefaults": {
  "CaptureSettings": [
   {
    "ImageType": 0,
    "Width": 3000,
    "Height": 2000,
    "FOV_Degrees": 90,
    "AutoExposureSpeed": 100,
    "AutoExposureBias": 0,
    "AutoExposureMaxBrightness": 0.64,
    "AutoExposureMinBrightness": 0.03,
    "MotionBlurAmount": 0,
    "TargetGamma": 1.0,
    "ProjectionMode": "",
    "OrthoWidth": 5.12
   }
  ]
 },
 "Vehicles": {
  "Drone1": {
   "VehicleType": "PX4Multirotor",
   "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPort": 14570,
			"VehicleSysID": 10,
   "X": 0,
   "Y": 0,
   "Z": 0
  },
  "Drone2": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4561,
            "ControlPort": 14570,
"VehicleSysID": 20,			
            "X": 2, "Y": 0, "Z": 0
        }
 }
}
`

AIKUUM, 2020-12-25:
Hi @ahmed-elsaharti, There is one more issue which I forgot to mention yesterday. When I execute hello_drone.py (link:https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/hello_drone.py), sometimes the drone goes to specified location based on (x,y,z in code), sometimes the drone becomes unstable during the flight and doesnt go to the specified location. Even after the execution of code completes, the drone doesnt land, but lands after some time automatically. 

Can you tell me why this could be happening?

jonyMarino, 2020-12-28:
Hi @AIKUUM, can you open a new issue filling the bug report, please?

AIKUUM, 2020-12-29:
> Hi @AIKUUM, can you open a new issue filling the bug report, please?

Hi @jonyMarino, I have done, Links:
https://github.com/microsoft/AirSim/issues/3259
https://github.com/microsoft/AirSim/issues/3260

jonyMarino, 2021-05-20:
Hi @AIKUUM! PX4 is now working. See:
https://github.com/microsoft/AirSim/issues/2477#issuecomment-845451529
[Here you have the new binaries with the fixes](https://github.com/microsoft/AirSim/releases)

closed 2021-05-20 by jonyMarino