microsoft/AirSim #3214: Invalid uncomressed images when recording from Unreal editor

alonfaraj, 2020-12-14:
## Bug report 

- AirSim Version/#commit: 1.3.1
- UE/Unity version: 4.25.4
- autopilot version: N/A
- OS Version: Win10

### What's the issue you encountered?

Recording uncompressed images from Unreal editor using rec button, output invalid images which can't be open when set `"Compress": false`. 
When record compress images - it works fine.

[image]

### Settings

Example:
```
  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 1,
    "Cameras": [
        { "CameraName": "0", "ImageType": 0, "PixelsAsFloat": false, "Compress": false },
        { "CameraName": "0", "ImageType": 5, "PixelsAsFloat": false, "Compress": false }
    ]
  }

```
### How can the issue be reproduced?

1. Add uncompress camera to recording section in setting file
2. Record from unreal editor using 'R'
3. Check the images folder

rajat2004, 2020-12-15:
This is a known bug, mentioned here - https://github.com/microsoft/AirSim/pull/2861#issuecomment-664036371
Copied from the above linked comment - UE is doing the conversion to PNG format, and then we're just writing it as a binary file. For uncompressed, writing array values as binary doesn't work. I can see 2 options, either go for an external library such as `libpng` which will do the conversion and all (format's complex, so doesn't seem to be possible to write it ourselves), or use a different format such as PPM which is simpler and can be written similar to how PFM file is being written currently.
I haven't worked on it more since then, maybe will pick it up again, I guess PPM will be better than nothing, external libraries are somewhat painful to inlcude, especially for cross-platform.

alonfaraj, 2020-12-15:
thank you @rajat2004

PR #2861 (2021-03-01): Extend Recording to multiple vehicles
Replaces #2847 (branch got messed up)

Closes #3029
Closes #3123
Closes #3214

Other related issues - #2307

New Additions -

1. Info of all vehicles is recorded (with additional column `VehicleName`), `VehicleName` parameter in `Cameras` field to specify specific vehicle to use the camera from
2. `Folder` field in settings to change the directory to store data in
3. Allow Recording with no images (just pose, etc data) (#3123)
4. `Enabled` parameter for starting recording when AirSim starts without needing to press R or call API (#3029) (Any naming suggest [... cut]
closed 2021-03-01 by zimmy87