microsoft/AirSim #3229: Rotation matrix to convert AirSim local coordinates to global coordinates

AngelTang190, 2020-12-18:
## Question
### What's your question?

Is there a rotation matrix to convert local AirSim coordinates to global coordinates and vice versa?

### Include context on what you are trying to achieve

N/A

#### Context details

This question is only applicable to AirSim version only.

### Include details of what you already did to find answers
I tried to search online about what is the rotation matrix to convert between AirSim coordinates and global coordinates, but I couldn't find any, seems like it needs to compare the direction of AirSim and the Earth. For example, the North of AirSim may be different from the True North (which is the Earth's North), so the rotation matrix synchronizes both the North, but I'm not sure about this. Can anyone please help? Thank you!

ahmed-elsaharti, 2020-12-18:
Hi @AngelTang190 ,
AFAIK, what you're trying to achieve will mainly depend on how the Unreal Engine level/environment is set up. i.e.: if your unreal environment is set up so that it resembles an actual location on earth, the angle between where (in the map/environment) north would be and the x-axis (for example) would mainly depend on how your environment is rotated within Unreal Engine.

Things are slightly simpler when it comes to GPS points though. You specify where exactly you want your drone to spawn ([X, Y and Z](https://github.com/microsoft/AirSim/blob/master/docs/settings.md#common-vehicle-setting)) within the map in [settings.json](https://github.com/microsoft/AirSim/blob/master/docs/settings.md) (or in the Unreal Editor by moving `PlayerStart`) and assign that spawn location to a real-world GPS coordinate (see [OriginGeopoint](https://github.com/microsoft/AirSim/blob/master/docs/settings.md#origingeopoint))

In general, if you are trying to line up the map's North and AirSim's North **visually**, you might need to do that from within the environment itself.
But if you don't care how it looks in the viewport, AFAIK, AirSim defaults the environment's x-axis to correspond to North. So If you track GPS data and move the drone in the x-direction, you'll see that it's moving in the north direction.

In the test below I used Droneshell to move the drone in the x-direction and it corresponded to motion in the north direction from where my OriginGeopoint/spawn-point was 
[image]

AngelTang190, 2020-12-18:
May I know how can I know the orientation of AirSim environment? Is there an API that I can use ?

ahmed-elsaharti, 2020-12-18:
@AngelTang190 the [getMultirotorstate](https://microsoft.github.io/AirSim/apis/#getmultirotorstate) and/or simGetGroundTruthKinematics APIs would provide the orientation in quaternions

AngelTang190, 2020-12-18:
Can you please give an example? As in when an orientation has a x value of 0 initially, what does it mean?

ahmed-elsaharti, 2020-12-18:
@AngelTang190 a yaw orientation of 0 means that the front of the vehicle is aligned with the World's/environment's x axis. Note that the multirotor state produces orientations in [quaternions](https://en.wikipedia.org/wiki/Quaternion) (for conversion see [this](https://quaternions.online/))

Here's an example that should clarify everything:

Lets say we have a blank unreal engine environment/level:
[image]

This environment would have a set of x,y and z axes that start from a point (the world origin). I'll represent the environment's top view in my diagrams with this grid:
[image]

Now, when u place the AirSim plugin into the Unreal Engine project, the x-axis is considered north (keep in mind that there is no map or anything here yet, its just an empty environment, and north is just relative to the environment's coordinate system):
[image]
And here is our setup with a drone placed at (3,4,0) (Note that if you set an OriginGeopoint in your settings.json like: 
```
"OriginGeopoint": {
    "Latitude": 47.641468,
    "Longitude": -122.140165,
    "Altitude": 122
  }
``` 
your drone spawn point will be set to the coordinates you enter):
[image]

Again, this is all without having anything related to 'real earth' in our environment.

Now, since the angle between the forward direction of the drone (the blue arrow below) and the world's x axis is 0. Your yaw value would be 0 and your orientation in [quaternions](https://quaternions.online/) would be [ 0, 0, 0, 1 ].
[image]

If the drone is spawned at an angle (pitch, yaw or roll):
[image]
the angles between the drone and the world's coordinate system will be reported in the state. (-45 degree yaw in the diagram).

Now this behavior should not change regardless of what we place in the environment. For example, if I place assets in the UE environment that represents a particular area on earth but these assets/map-items are not alligned with the world coordinate system:
[image]
There would be a difference between the world/game's coordinate system's North and the 'maps' "true north". If you know that difference (assuming you are the one building the level/environment so you would know what the offset is) you can always add it to your calculations. 
OR you could just build your environment so that it's lined up correctly:
[image]

Hope this clarifies things. I might be mistaken in how exactly the aligning works within UE but this is how my understanding of it is.

AngelTang190, 2020-12-21:
Alright, thank you very much! So let's say if the yaw is 0, then the rotation matrix will just be the identity matrix correct? Since there is no rotation?

ahmed-elsaharti, 2020-12-21:
@AngelTang190 if you're refering to a 2D rotation matrix about the Z axis then yes. But if you're talking about an overall 3D rotation matrix then pitch and roll would have to be taken into consideration

AngelTang190, 2020-12-21:
For 3D rotation matrix, should pitch also be aligned with the North? Then which direction should roll be aligned to?

ahmed-elsaharti, 2020-12-21:
Pitch would be measured based on the horizon or the vehicle being upright. Same thing with roll. They would both be relative to the drone's upright orientation

AngelTang190, 2020-12-21:
May I know what does upright mean here?

petergerten, 2021-05-08:
I am currently using AirSim together with the Cesium for Unreal plugin. Cesium puts a real world map into the Unreal project. I used the same coordinates for the georeference origin in AirSim and Cesium. This works, the drone shows at the correct position and GPS data (PX4 SITL) is correct. But when I fly into any directions - they don't line up. I am a bit lost where I would configure that. Any hints?

ahmed-elsaharti, 2021-05-08:
@petergerten by 'they don't line up' do you mean there's an angle offset? or is it a distance offset?

petergerten, 2021-05-08:
@ahmed-elsaharti I need to evaluate it more closely. But it seems distance is correct but when I fly south in the airsim real-world environment - my GPS coordinates in AirSim/PX4 go east.

petergerten, 2021-05-09:
@ahmed-elsaharti I tried to change the rotation of player start - but it does not seem to have an impact on the world rotation.

petergerten, 2021-05-09:
ok, after rotating the Cesium World Terrain 90 degrees - everything lines up :)