microsoft/AirSim #3232: Can't get AirSim to run with VehicleType ArduCopter

MoBaT, 2020-12-18:
## Bug report 

- AirSim Version/#commit: 5513f19
- UE/Unity version: ++UE4+Release-4.18
- autopilot version: master/f670f7a
- OS Version: Ubuntu 18.04

### What's the issue you encountered?

Getting the error below and AutoPilot is not able to connect and send data
[image]
[image]

### Settings

```
{
  "SettingsVersion": 1.2,
  "LogMessagesVisible": true,
  "SimMode": "Multirotor",
  "OriginGeopoint": {
    "Latitude": -35.363261,
    "Longitude": 149.165230,
    "Altitude": 583
  },
  "Vehicles": {
    "Copter": {
      "VehicleType": "ArduCopter",
      "UseSerial": false,
      "LocalHostIp": "127.0.0.1",
      "UdpIp": "127.0.0.1",
      "UdpPort": 9003,
      "ControlPort": 9002
    }
  },

  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 1024,
        "Height": 576,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 100,
        "AutoExposureBias": 0,
        "AutoExposureMaxBrightness": 0.64,
        "AutoExposureMinBrightness": 0.03,
        "MotionBlurAmount": 0,
        "TargetGamma": 1.0,
        "ProjectionMode": "",
        "OrthoWidth": 5.12
      }
    ]
  }
}
```

### How can the issue be reproduced?

1. Run airsim docker with these commands: ./run_airsim_image_binary.sh airsim_binary:10.2-devel-ubuntu18.04 Blocks/Blocks.sh -windowed -ResX=1080 -ResY=720

rajat2004, 2020-12-25:
I'm not able to reproduce this error, the current latest Blocks release is working for me, the settings are correct.
From the script `run_airsim_image_binary.sh` you're using, I think you would have used the corresponding `download_blocks_env_binary.sh` script as well? That will download `v1.2.0` binary, which is very old and does not have ArduPilot support, you'll need to download the latest release `v1.3.1` and use that, there's also a PR https://github.com/microsoft/AirSim/pull/2942 which updates the various Docker scripts

MoBaT, 2021-01-05:
@rajat2004 I updated Blocks to 1.3.1. When I run Unreal with the Blocks environment, I see error messages like these. I'm not too sure if it's affecting my startup. But it still did not work.

```
[2021.01.05-15.52.42:570][876]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:573][877]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:577][877]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:578][877]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:583][878]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:587][878]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:589][878]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:591][879]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:595][879]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:597][879]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:600][880]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:603][880]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:605][881]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:609][881]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:611][881]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:615][882]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:618][882]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:621][883]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:625][883]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:629][884]LogTemp: Error: Exception occurred while updating world
Exception occurred while updating world
[2021.01.05-15.52.42:632][884]LogTemp: Error: Exception occurred while updating world
```

closed 2021-01-06 by MoBaT