microsoft/AirSim #3235: UE4 Engine Version clarification (minimum and suggested)

petergerten, 2020-12-19:
Is there a definite place that clarifies the minimum and recommended engine version to be used with AirSim?
I have been using UE4.24.3 for example and it stopped working due to new uasset files that are not downward 
compatible anymore. But I don't know what the new minimum engine version is.

Example:
https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Content/HUDAssets/CameraDistortion.uasset
does not work with 4.24.3 anymore.

saihv, 2020-12-19:
This information is maintained in https://github.com/microsoft/AirSim/blob/master/docs/build_windows.md and https://github.com/microsoft/AirSim/blob/master/docs/build_linux.md. We have recently updated the docs to reflect the change to 4.25 as the minimum requirement.

b0b023, 2020-12-20:
@saihv great, thanks! Sorry I missed that particular doc.

petergerten, 2020-12-20:
@saihv thanks. I just tried to upgrade to 4.25.4-release - but it seems this version does not work for different reasons. See https://github.com/microsoft/AirSim/issues/3200

Maybe it would be good to be more specific in the recommended version?

saihv, 2020-12-22:
Please see discussion at #3200

closed 2020-12-28 by jonyMarino

dromero12, 2021-10-14:
> This information is maintained in https://github.com/microsoft/AirSim/blob/master/docs/build_windows.md and https://github.com/microsoft/AirSim/blob/master/docs/build_linux.md. We have [...]

It means 4.25.0?, what about 4.25.4, same problems than the commented previously. Is AirSim upgradable? UE 4.27.1 is the current version.

jonyMarino, 2021-10-14:
Hi @dromero12! Yes, 4.25.4 is fine. We haven't tested with 4.27.1, but we should support it also.