microsoft/AirSim #3236: ERROR [sensors] Accel #0 fail:  TIMEOUT!

AIKUUM, 2020-12-21:
## Question
### What's your question?

### Include context on what you are trying to achieve

#### Context details

### Include details of what you already did to find answers

Below is my system details:
https://user-images.githubusercontent.com/72294070/102069277-90db4a00-3e23-11eb-8f71-d14e10181433.PNG

AirSim version (1.3.1)
Unreal version(4.25)
python version (3.8.3)

When I try to open an AirSim environment with PX4 (SITL), i get error as in the below screenshot:
[image]

Can anyone help me to proceed further?

ahmed-elsaharti, 2020-12-21:
@AIKUUM what versions of PX4 and Cygwin/PX4-Toolkit are you using? Also, are you simulating on Unity or Unreal Engine?

AIKUUM, 2020-12-21:
> @AIKUUM what versions of PX4 and Cygwin/PX4-Toolkit are you using? Also, are you simulating on Unity or Unreal Engine?

Hi @ahmed-elsaharti, Im simulating on Unreal Engine. Im using PX4.Windows.Cygwin.Toolchain.0.8
Below are the settings in settings.json file:
`
{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "Vehicles": {
        "Drone1": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPort": 14580,
            "X": 0, "Y": 1, "Z": 0
        },
        "Drone2": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4561,
            "ControlPort": 14580,          
            "X": 0, "Y": -1, "Z": 0
        }
    },
    "OriginGeopoint": {
        "Latitude": 47.641468,
        "Longitude": -122.140165,
         "Altitude": 0
        }
  }
`

ahmed-elsaharti, 2020-12-21:
@AIKUUM and are you using the latest repo or are you simulating with one of the precompiled binaries?
Also, what is the version of the PX4 firmware being compiled? is it the latest version (1.11.2) or the recommended 1.10.1?

The reason I'm asking is that I'm trying to recreate the issue on my machine.

AIKUUM, 2020-12-21:
> @AIKUUM and are you using the latest repo or are you simulating with one of the precompiled binaries?
> Also, what is the version of the PX4 firmware being compiled? is it the latest version (1.11.2) [...]

Hi @ahmed-elsaharti, I'm using PX4 v1.10.1.
And which repo are you talking about? AirSim repo?

ahmed-elsaharti, 2020-12-21:
@AIKUUM Yup, I was refering to the Airsim repo. Were you using the latest repo when you got these errors or were you using one of the recompiled binaries in the latest release from back in April?

AIKUUM, 2020-12-21:
> @AIKUUM Yup, I was refering to the Airsim repo. Were you using the latest repo when you got these errors or were you using one of the recompiled binaries in the latest release from back in April?

Hi @ahmed-elsaharti, I'm using one of the recompiled binaries from the link https://github.com/microsoft/AirSim/releases

ahmed-elsaharti, 2020-12-21:
@AIKUUM I was able to recreate the error with one of the precompiled binaries. Try switching over to the newest Airsim repo. There probably was a bug that was fixed at some point between April and now.

You could test out the blocks environment in the latest repo by following the steps [here](https://microsoft.github.io/AirSim/unreal_blocks/)

AIKUUM, 2020-12-21:
> @AIKUUM I was able to recreate the error with one of the precompiled binaries. Try switching over to the newest Airsim repo. There probably was a bug that was fixed at some point between April and [...]

Hi @ahmed-elsaharti, I had tested blocks environment without PX4 few months ago. But to accomplish my current task, I need some simulator environment which is similar to a real time environment. I cant see simulator environments in the latest repository. 
Can you now tell me how to proceed?

ahmed-elsaharti, 2020-12-21:
Hi @AIKUUM, If the blocks environment is not suitable you will either have to wait until the maintainers publish a new release that will include binaries built from the latest (fixed) repo OR you could use Airsim with a custom environment.

To use the latest repo of Airsim with a custom UE4 environment, you will first need to [build Airsim](https://microsoft.github.io/AirSim/build_windows/). After you have done that you can follow [this guide](https://microsoft.github.io/AirSim/unreal_custenv/) to add Airsim to any UE4 environment you want.

AIKUUM, 2020-12-21:
> Hi @AIKUUM, If the blocks environment is not suitable you will either have to wait until the maintainers publish a new release that will include binaries built from the latest (fixed) repo OR you [...]

Hi @ahmed-elsaharti, Im not manually simulating the environments. I'm using my Python codes based on the available AirSim APIs to achieve my task. I had already built AirSim as in the link https://microsoft.github.io/AirSim/build_windows/ few months ago. 
By any chance am I getting the error in PX4 because of the configurations I have used in settings.json (link:https://github.com/microsoft/AirSim/issues/3236#issuecomment-748846975)

ahmed-elsaharti, 2020-12-21:
@AIKUUM it is not likely that the errors are because of the settings. I got the same errors when I used one of the precompiled binaries, and when I switched over to the latest repo the errors were gone.

AIKUUM, 2020-12-21:
> @AIKUUM it is not likely that the errors are because of the settings. I got the same errors when I used one of the precompiled binaries, and when I switched over to the latest repo the errors were gone.

Hi @ahmed-elsaharti, How did you simulate? manually or automated?
Did you simulate on Windows or on any other OS?
Are you using Unreal Engine or Unity?

ahmed-elsaharti, 2020-12-21:
@AIKUUM I cloned the latest repo and tested the Blocks environment in Unreal Engine as shown [here](https://microsoft.github.io/AirSim/unreal_blocks/). This was all done on Windows 10 and used PX4 1.10.1 with Cygwin 0.8 and resulted in no errors overall (GPS and sensors do glitch sometimes but it doesn't always happen).

[image]

jonyMarino, 2021-05-20:
Hi @AIKUUM! PX4 is now working. See:
https://github.com/microsoft/AirSim/issues/2477#issuecomment-845451529
[Here you have the new binaries with the fixes](https://github.com/microsoft/AirSim/releases)

closed 2021-05-20 by jonyMarino