microsoft/AirSim #3238: error:cdo Constructor(PIPcamera):failed to find Material'AirSim/HUDAssets/CameraDistortion.CameraDistortion'

Yannisxm, 2020-12-21:
error:cdo Constructor(PIPcamera):failed to find Material'AirSim/HUDAssets/CameraDistortion.CameraDistortion' 
[image]

jonyMarino, 2020-12-21:
Hi! Please, next time, fill the Bug Report template. Incomplete reports would be closed immediately.
We have updated to UE 4.25.
Update your UE version, and the error will disappear

denmonz, 2021-03-12:
If anyone else is encountering this issue on UE 4.24 and can't upgrade your project to UE 4.25 (like myself) I was able to go into the source code ([UE4_project]/Plugins/AirSim/Source) and commented-out the code in PIPCamera.cpp and PIPCamera.h that mentions the distortion effect (variable distortion_material_static_ and function setDistortionMaterial()). Then I rebuilt my project from source via Visual Studio 2019. It's working thus far for me, but I have yet to test whether something like this is viable and won't cause issues later.