microsoft/AirSim #3243: AirSim repository with all the simulator environments for Windows

AIKUUM, 2020-12-23:
## What feature are you suggesting?
### Overview:

### Smaller Details: 

### Nature of Request:  

- Addition
- Change
- Removal
  - 

## Why would this feature be useful?

Can anyone give me an idea on by when all the simulator environments for Windows (link:https://github.com/microsoft/AirSim/releases) will be included in AirSim repository? I would appreciate it if that happens as soon as possible, because "blocks" environment is not a suitable environment (would want to use an environment which is similar to some real time environment) and "LandscapeMountains" environment does not give a fixed position for drones (sometimes drones will slide down the slope of mountains, sometimes drones will be on a flat surface of a mountain, sometimes it will be on the peak of a mountain, sometimes errors occur too for invalid location of drones) even when the GPS location is fixed.

ahmed-elsaharti, 2020-12-23:
@AIKUUM, to address the issue with LandscapeMountains, you can programatically reposition the drone in a compiled binary by changing the X, Y and Z values in settings.json. If you're compiling your own binary then you can place the drone in any location you like in LandscapeMountains or any other environment as shown in [this guide](https://microsoft.github.io/AirSim/unreal_custenv/), [this video](https://www.youtube.com/watch?v=1oY8Qu5maQQ&) and [this video](https://www.youtube.com/watch?v=y09VbdQWvQY&feature). 
In general, the use of AirSim is not limited to LandscapeMountains, Blocks and the precompiled binaries

AIKUUM, 2020-12-24:
> @AIKUUM, to address the issue with LandscapeMountains, you can programatically reposition the drone in a compiled binary by changing the X, Y and Z values in settings.json. If you're compiling your [...]

Hi @ahmed-elsaharti, even for fixed values of X,Y,Z, the drone(s) land in different areas in LandscapeMountains environment every time I open the environment. This is what i have seen for both precompiled version and build version.

saihv, 2020-12-24:
@AIKUUM Out of curiosity, could you please try this (old) binary of Landscape Mountains and check if the issue you mentioned still happens?

https://github.com/microsoft/AirSim/releases/download/v.1.2.2/LandscapeMountains.zip

AIKUUM, 2020-12-24:
> @AIKUUM Out of curiosity, could you please try this (old) binary of Landscape Mountains and check if the issue you mentioned still happens?
> 
> https://github.com/microsoft/AirSim/releases/download/v.1.2.2/LandscapeMountains.zip

Hi @saihv, I even have the precompiled version of "landscapeMountains" whose link you have shared. For fixed values of (x,y,z) I had been facing the issue.

xxEoD2242, 2020-12-24:
@AIKUUM the binaries won't help. 
Your problem is that there are multiple Player Start objects in the binaries. You need to build your own, delete all of them and have only 1 Player Start. Then recompile the binaries and it'll work the way you want

AIKUUM, 2020-12-24:
> @AIKUUM the binaries won't help.
> Your problem is that there are multiple Player Start objects in the binaries. You need to build your own, delete all of them and have only 1 Player Start. Then recompile the binaries and it'll work the way you want

Hi @xxEoD2242, The issue which I'm facing now is for built version of "LandscapeMountains". Can you tell me what the reason could be?

ahmed-elsaharti, 2020-12-24:
@xxEoD2242 Interesting. I was not aware that one of the binaries had multiple playerstart objects. I can confirm that this is the case with the binary in release 1.3.1. 
However, that doesn't seem to be the case with the binary @saihv linked to. I just tested it out and the drone spawned at the same point every time (I did 5 tests to be very specific).
All in all though, seems like a custom compilation of the LandscapeMountains environment is the way to go.

AIKUUM, 2020-12-24:
> @AIKUUM Out of curiosity, could you please try this (old) binary of Landscape Mountains and check if the issue you mentioned still happens?
> 
> https://github.com/microsoft/AirSim/releases/download/v.1.2.2/LandscapeMountains.zip

Hi @saihv, I just assumed the binary to be from v1.3.1. I just checked for the version whose link you have shared for "LandscapeMountains". The issue which I was facing is resolved.

saihv, 2020-12-24:
Tagging @jonyMarino for visibility: issues with latest binaries vs. old binaries of Landscape Mountains

jonyMarino, 2020-12-28:
Thanks, Sai! Answering the feature request, the other environments in the binaries are private, so we can not add their sources to the repository. If you still need them, you can download them from its provider.

jonyMarino, 2022-01-12:
Now Landscape shouldn't have that stochastic behavior.

closed 2022-01-12 by jonyMarino