microsoft/AirSim #3249: NedTransform::toVector3r why not apply - to y?

litingzhen, 2020-12-27:
## Question
### What's your question?

Unreal World  Coordinate system: x is right of the screen，y is down of the screen, z is towards to user. 
Global NED: x is right of the screen, y is up of the screen, z is back to user.
so (x,y,z) from UU to Global NED is (x,-y,-z).
why the code NedTransform::toVector3r not apply - to y?

### Include context on what you are trying to achieve

#### Context details

### Include details of what you already did to find answers

ahmed-elsaharti, 2020-12-27:
Hi @litingzhen , Firstly, the Unreal World Coordinate system you described is a standard [left-handed](https://www.evl.uic.edu/ralph/508S98/coordinates.html) coordinate system and the directions of the axes depend on how your viewport is oriented (although their directions relative to eachother are the same):
[image]
[image]

Now, if you look at the map from the top view without any rotation, X will point upwards, Y will point right and Z will point towards the camera:
[image]
From what I remember (and from your description), the UAV's coordinate system is a [right-handed coordinate system](https://www.evl.uic.edu/ralph/508S98/coordinates.html). Where from the top view (with the drone pointing up in the view), the drone's X will be pointing upwards, Y will point right and Z will point AWAY from the camera. To convert from a right handed coordinate system to a left-handed one you'll need to multiply Z by -1 
[image]

litingzhen, 2020-12-28:
Thank you very much.

closed 2020-12-28 by litingzhen