microsoft/AirSim #3251: [Bug Report] msgpack-python is deprecated, use msgpack

wangwwno1, 2020-12-28:
## Bug report 

- AirSim Version: a0d4d0e9672900e6182aab5f6cb5b59a09a85509
- UE/Unity version: N/A
- autopilot version: N/A
- OS Version: N/A

### What's the issue you encountered?

> This package is deprecated. Install msgpack instead.

https://pypi.org/project/msgpack-python/
When users install both msgpack and airsim they will have both msgpack and msgpack-python installed, which conflict.

### Settings

N/A

### How can the issue be reproduced?

1. Install airsim, then install msgpack by `pip install msgpack`
2. Run the code below and you will get an error
```
import airsim
client = airsim.MultirotorClient()
client.ping()
```
### Include full error message in text form

```
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/home/USER/anaconda3/envs/VeriFC/lib/python3.8/site-packages/airsim/client.py", line 44, in ping
    return self.client.call("ping")
  File "/home/USER/anaconda3/envs/VeriFC/lib/python3.8/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/USER/anaconda3/envs/VeriFC/lib/python3.8/site-packages/msgpackrpc/session.py", line 51, in send_request
    self._transport.send_message([message.REQUEST, msgid, method, args])
  File "/home/USER/anaconda3/envs/VeriFC/lib/python3.8/site-packages/msgpackrpc/transport/tcp.py", line 89, in send_message
    self.connect()
  File "/home/USER/anaconda3/envs/VeriFC/lib/python3.8/site-packages/msgpackrpc/transport/tcp.py", line 98, in connect
    socket = ClientSocket(stream, self, self._encodings)
  File "/home/USER/anaconda3/envs/VeriFC/lib/python3.8/site-packages/msgpackrpc/transport/tcp.py", line 53, in __init__
    BaseSocket.__init__(self, stream, encodings)
  File "/home/USER/anaconda3/envs/VeriFC/lib/python3.8/site-packages/msgpackrpc/transport/tcp.py", line 12, in __init__
    self._packer = msgpack.Packer(encoding=encodings[0], default=lambda x: x.to_msgpack())
  File "msgpack/_packer.pyx", line 124, in msgpack._cmsgpack.Packer.__init__
TypeError: __init__() got an unexpected keyword argument 'encoding'
```

**What's better than filing an issue? Filing a pull request :).**

ahmed-elsaharti, 2020-12-28:
@wangwwno1 I tested this on a new machine and everything worked as expected when following [the docs](https://microsoft.github.io/AirSim/apis/) and installing `msgpack-rpc-python`
[image]

The error did appear when i tried to install `msgpack`. But its not what the docs recommend anyway.

wangwwno1, 2020-12-28:
@ahmed-elsaharti 
Please install msgpack by `pip install msgpack==1.0.2` and try again.
`msgpack` and `msgpack-python` are two different packages, and the latter one is deprecated.

ahmed-elsaharti, 2020-12-28:
@wangwwno1 it shouldn't matter (as far as the scope of Airsim goes) if either of those packages produces errors though as long as the one recommended in the docs works.

If you're trying to make this work just for airsim i would recommend uninstalling the packages mentioned below and installing msgpack-rpc-python which should work.

I'll also do some testing on one more machine to make sure the results are consistent.

wangwwno1, 2020-12-28:
@ahmed-elsaharti 
Thanks for your suggestion, but I need the [Ray](https://github.com/ray-project/ray) library to parallelize the data collection process and `msgpack-python` is conflict with it.
I resolve this issue by:
1. install msgpack-rpc-python in this PR msgpack-rpc/msgpack-rpc-python#23 and
2. Remove argument `pack_encoding` and `unpack_encoding` in `VehicleClient`, which locatetd in airsim/client.py

closed 2021-03-22 by jonyMarino