microsoft/AirSim #3254: help needed! compiling error occurred during build.cmd under Unity directory

eminem1984, 2020-12-28:
Hi, i am wondering if anyone can help me on this issue - I pretty stuck in the building process of AirSim,  I am now inside of the Unity directory and tried to build the project, but then this error pops up: 

> LINK : fatal error LNK1181: cannot open input file 'MavLinkCom.lib' [C:\AirSim\Unity\AirLibWrapper\AirsimWrapper\Airsim

Wrapper.vcxproj]

I tried to manually add the MavLinkCom.lib to the VS project as additional dependencies, but still got the same error. What have I missed?  Thanks a lot. 

For your information - i am using windows10, VS2019 and Unity2019.

rajat2004, 2020-12-30:
Is this happening when running `build.cmd` from the Unity directory? That would be strange, since it seems to working correctly on the Github Actions CI
Could you post the entire output? The above message would happen if there was a compilation failure earlier and hence it wasn't able to generate the library. Update AirSim to the latest master and try again

jonyMarino, 2021-01-04:
Hi @eminem1984, and welcome!
Please, next time, fill the issue template.
It seems that you are trying to compile AirsimWrapper.sln. Can you use `build.cmd`  from the Unity directory only?

RoonLoe, 2021-01-19:
Hi @eminem1984 can you tell me where you found MavLinkCom.lib. I am running into, perhaps, a similar issue, but I cannot find even find the file. 
[image]

jonyMarino, 2021-02-01:
@eminem1984, are you still having this issue?

closed 2021-02-02 by eminem1984

eminem1984, 2021-02-02:
> @eminem1984, are you still having this issue?

many thanks for the replies, just got overwhelmed by other stuff and didn't get time to look into this anymore, so better to close the issue for now.