microsoft/AirSim #3255: Is there a way of setting the GPS Position while the simulator is running?

MarvinKlemp, 2020-12-28:
## Question
### What's your question?
Is there a way of setting the GPS Position while the simulator is running?

### Include context on what you are trying to achieve
I'm trying to set a vehicle to different positions while the simulator is running. I did not find a method to set the position of the car other than restarting with a different settings.json (requires to restart the simulator each time).

### Include details of what you already did to find answers
Search the repository. But could not really find a solution.

rajat2004, 2020-12-29:
[`simSetVehiclePose`](https://microsoft.github.io/AirSim/api_docs/html/#airsim.client.VehicleClient.simSetVehiclePose) might be what you're looking for, you can move the vehicle as required in the environment, and the GPS position will change accordingly.
The other interpretation of your issue would be to change just the GPS values, and not the position in the environment itself, which doesn't make much sense

MarvinKlemp, 2021-01-04:
Thanks that's exactly what I was looking for

closed 2021-01-04 by MarvinKlemp