microsoft/AirSim #3256: How to enable Roadwetness, RoadSnow and RoadLeaf in a Custom Unreal Environment?

GilmarCorreia, 2020-12-28:
## Question
### How to enable Roadwetness, RoadSnow, and RoadLeaf in a Custom Unreal Environment?

Hello developers community,

I'm trying to create a Custom Unreal Environment, it is working pretty well, but when I'm trying to enable Roadwetness, RoadSnow, or even RoadLeaf it doesn't seem to work. Is there any prop that should I put in Unreal to enable these features?

### Include context on what you are trying to achieve

N/A

#### Context details
OS: Windows 10
AirSim: 1.3
Python:  3.8.5
Unreal: 4.24

### Include details of what you already did to find answers
N/A

ahmed-elsaharti, 2020-12-28:
@GilmarCorreia were you able to get weather effects to work with the blocks environment or any of the binaries?

I've tested adding weather effects to multiple of the pre-built binaries (and a blocks binary built a week ago) but the F10 button doesn't bring up the weather menu anymore. 
The weather menu shows up when running the sim using the UE editor but is blank:
[image]
I also did receive a couple of compilation errors for the BP responsible for that menu:
[image]
[image]

@jonyMarino @zimmy87 any ideas?

GilmarCorreia, 2020-12-28:
Hello @ahmed-elsaharti, I'm facing the same problem. It seems that the weather widgets for UI are disabled and not showing when you hit F10. To enable it I'm using a python client to set the configurations for each weather parameter.

saihv, 2020-12-29:
@ahmed-elsaharti That is likely because there was an error in the options menu blueprint, which #3240 was trying to fix. Could you perhaps try with the latest master?

ahmed-elsaharti, 2020-12-29:
@saihv Oh wow! My apologies, I did NOT see that PR. (I had a feeling I was going to regret not testing the latest master earlier 😅) 
On testing with the latest master and UE 4.25 (within the UE Editor):
[image]

saihv, 2020-12-29:
@ahmed-elsaharti Thanks for checking!

@GilmarCorrea To answer your original question, enabling Road-related
weather features for custom worlds is actually a bit involved and would
have to be done for each map. The concept generally involves using an
additional material over the texture of the road (example: white layer for
snow) and dynamically adjusting the blending of this material with the road
through the weather parameter. Similarly, road wetness can be done through
a water-like material and adjusting the roughness and specular properties
to get the right reflection. You might be able to find some Unreal Engine
tutorials online that talk about how to blend materials that way.

In the prebuilt binaries, Neighborhood is the only map (IIRC) that allows
users to change these options and see the related change visually appear.
Down the line, it might be possible to do this in a more level-agnostic
fashion but it hasn't been explored yet.

GilmarCorreia, 2020-12-30:
Hello @saihv!

Thank you for your explanation, after rebuilding my project with the unreal engine 4.25 the F10 menu is now showing. I will study more about blending material in road assets. 

Thank you!

closed 2020-12-30 by GilmarCorreia