microsoft/AirSim #3258: Segmentation images not appearing correctly from Blocks environment

uzgit, 2020-12-29:
## Question
### What's your question?

Why do the Segmentation images that I am extracting from the Blocks sample environment appear just as normal (darker) RGB images? I am very new to AirSim/Unreal so I kind of assume I am doing something naive or overlooking something simple.

### Include context on what you are trying to achieve

I am trying to collect Scene and Segmentation images from an edited version of the sample environment Blocks. The scene pictures are extracted as expected, but the Segmentation images appear just as slightly different RGB images. I am trying to extract Segmentation images where each mesh/object is given a single, solid color in order to create a mask so that I can train a model to detect certain objects within the scene.

#### Context details

OS: Ubuntu 20.04
Python: 3.8.5
SITL: ArduCopter (sim_vehicle.py)

### Include details of what you already did to find answers
I am using the following Python script to extract images from AirSim:
```
#!/usr/bin/python3

import sys
sys.path.insert(0, "/home/joshua/git/Airsim/PythonClient") # use the included AirSim Python client
import airsim
import numpy

client = airsim.MultirotorClient()

# try to exclude all objects from the segmentation image as per the documentation
# result = client.simSetSegmentationObjectID("[\w]*", -1, True)
# print("set all object ids to -1: {}".format(result))

# set the ID of the symbol cube to 99 instead of -1 to include it in the segmentation image
result = client.simSetSegmentationObjectID("symbol_cube", 99)
symbol_cube_id = client.simGetSegmentationObjectID("symbol_cube")
print("set symbol_cube id: {} -> {}".format(result, symbol_cube_id))

# verify IDs for all objects are -1 unless that object is the symbol cube
scene_objects = client.simListSceneObjects("[\w]*")
for scene_object in scene_objects:
    scene_object_id = client.simGetSegmentationObjectID(scene_object)
    # print("{} : {}".format(scene_object, scene_object_id))
    if( scene_object != "symbol_cube" ):
        assert scene_object_id == -1

# request Scene and Segmentation images
requests = [airsim.ImageRequest(camera_name=0, image_type=airsim.ImageType.Scene, pixels_as_float=False, compress=False),
            airsim.ImageRequest(camera_name=0, image_type=airsim.ImageType.Segmentation, pixels_as_float=False, compress=False)]
responses = client.simGetImages(requests)
print("response length: {}".format(len(responses)))

# for each returned image
for i in range(len(requests)):

    # make the image as a 1-dimensional numpy array
    img1d = numpy.frombuffer(responses[i].image_data_uint8, dtype=numpy.uint8)
    img_rgb = img1d.reshape(responses[i].height, responses[i].width, 3)

    # chatter about the image
    print("response dimensions: {} x {}".format(responses[i].height, responses[i].width))
    print("dimensions: {} x {}".format(img_rgb.shape[0], img_rgb.shape[1]))

    # save the image
    filename = "image_{}.png".format(i)
    airsim.write_png(filename, img_rgb)
```

I am using a settings.json file as follows
```
{
  "SettingsVersion": 1.2,
  "LogMessagesVisible": true,
  "SimMode": "Multirotor",
  "OriginGeopoint": {
    "Latitude": -35.363261,
    "Longitude": 149.165230,
    "Altitude": 583
  },
  "Vehicles": {
    "Copter": {
      "VehicleType": "ArduCopter",
      "UseSerial": false,
      "LocalHostIp": "127.0.0.1",
      "UdpIp": "127.0.0.1",
      "UdpPort": 9003,
      "ControlPort": 9002
    }
  },
  "CameraDefaults": {
	  "CaptureSettings": [
		  {
			  "ImageType": 0,
			  "Width": 640,
			  "Height": 480,
			  "FOV_Degrees": 160
		  },
		  {
			  "ImageType": 5,
			  "Width": 640,
			  "Height": 480,
			  "FOV_Degrees": 160
		  }
	  ]
  },
  "SegmentationSettings": {
    "InitMethod": "None",
    "MeshNamingMethod": "",
    "OverrideExisting": false
  }
}
```

When I run the Python script alongside AirSim/ArduPilot using the Blocks environment (running either in the UE4 Editor or in a standalone window) I get the following two images regardless of if I set object segmentation IDs to -1:
Scene:
[image]
Segmentation:
[image]

When I run the Python script alongside AirSim/ArduPilot using the Neighborhood environment (without setting segmentation IDs to -1) I get the following two images:
Scene:
[image]
Segmentation:
[image]

The segmentation image above, from the neighborhood environment is much more like what I am expecting to get in order to create a mask. Does anyone have any input on how to change my setup in order to get this sort of image? Thanks

ahmed-elsaharti, 2020-12-29:
Hi @uzgit , I did some testing with your script+settings but using simple flight instead of ArduCopter.
Firstly, the blocks environment doesn't contain an object with the name 'symbol_cube' (as far as I remember. There's a decent change I'm completely off here 😂).
From what I understood, I think you're trying to refer to the `TemplateCube` objects. 
On changing `symbol_cube` in your script to `TemplateCube[\w]*` and adding `True` to `result = client.simSetSegmentationObjectID("TemplateCube[\w]*", 99)` (ie: making it `result = client.simSetSegmentationObjectID("TemplateCube[\w]*", 99, True)` this is the result I get:
[image]
Is this what you're looking to do?

uzgit, 2020-12-29:
Hi @ahmed-elsaharti, this is very close to what I want to do! I should have mentioned that I have added the object "symbol_cube" to the Blocks environment as a Basic->Cube Actor with a WhyCode texture. It is the center-top (slightly right) of my first 2 images. However, when I run the command with the regex that you have provided, and the additional "True" parameter for is_name_regex I still get a Segmentation image like the following:
[image]
Could it be that adding an additional actor to the project changes things somehow? Or maybe that recompiling the project to a different version (this was done upon my first load of the project in UE4Editor) messes things up?

The texture which I have added to the cube is the .png file included below:
[image]
Thank you for your quick response!

ahmed-elsaharti, 2020-12-29:
@uzgit Ah! I see what you're doing now.
I've added a Cube into the environment (the default name was Cube_2 so I just went with it) and changed `symbol_cube` in your code to `Cube_2`. Here is what I got:
[image]
Not sure what the reason behind your results is but it could be the version thing you mentioned. I'd reclone and rebuild the repo with UE 4.25 (this is what i used for this test). Especially since the image you're getting doesnt seem to be a segmentation image at all.

uzgit, 2020-12-29:
@ahmed-elsaharti Alright, this is great info. I agree that what I'm getting doesn't seem to be a segmentation image at all. The segmentation image that you are getting is exactly what I'm looking for. It's very weird to me that it is working with the Neighborhood project and not the Blocks project. I will try recloning/recompiling soon. Thanks for testing it! Also, could you let me know your OS and Python version?

ahmed-elsaharti, 2020-12-29:
@uzgit I actually forgot to ask, are you using the blocks environment within the Unreal folder of the repo or are you using a precompiled one (ie: one of the releases just like with the neighborhood environment).
Also, this was tested on Windows 10 with Python 3.8.2 and a brand new clone of the repo.

Hope this helps ^_^

uzgit, 2020-12-29:
@ahmed-elsaharti I am using the Blocks environment at ./AirSim/Unreal/Environments/Blocks/Blocks.uproject of the AirSim cloned git repo (once the script ./install_run_all.sh is run). However, I have edited it to add the additional cube actor only. I am not using a precompiled one. Thanks for your effort! I will get back to you soon!

uzgit, 2020-12-29:
So I have re-cloned and re-compiled the whole AirSim/Unreal repo. This is still the Segmentation image that I am getting:
[image]

uzgit, 2020-12-29:
By the way, Segmentation images appear correctly in the Africa world as well, but not in the Blocks world:
[image]

ahmed-elsaharti, 2020-12-29:
@uzgit try testing the precompiled Blocks environment from the releases section and see if it does the same thing. Also if u enable the 2nd subwindow in the environment by clicking the '2' button you should be able to see what the segmentation camera view looks like before running the script (this will help us understand if the script has anything to do with the behaviour)

uzgit, 2020-12-29:
If I run my script on the precompiled Blocks environment, I do get a proper Segmentation image, as below:
[image]
Running it within the editor gives this still:
[image]

uzgit, 2020-12-30:
I believe I have found/solved the issue, which is that I was using UE4Editor version 4.24. This is the version that AirSim natively downloads and installs via the scripts install_run_all.sh and install_unreal.sh. I re-cloned the AirSim repo and changed the git clone commands in those scripts to use the 4.25 branch of the Unreal Engine repo, i.e.
```
git clone -b 4.24 https://github.com/EpicGames/UnrealEngine.git "$UnrealDir"
```
becomes
```
git clone -b 4.25 https://github.com/EpicGames/UnrealEngine.git "$UnrealDir"
```
and now the Segmentation images can be successfully pulled from the Editor using my original script, as shown below:
[image]

closed 2020-12-30 by uzgit