microsoft/AirSim #3259: Execution "takeoff_and_land.py" with 2 instances in PX4 SITL

AIKUUM, 2020-12-29:
## Bug report 

System details: https://user-images.githubusercontent.com/72294070/102069277-90db4a00-3e23-11eb-8f71-d14e10181433.PNG
AirSim version (1.3.1)
Unreal version(4.25)
python version (3.8.3)

### What's the issue you encountered?
I had done this in PX4 SITL for 2 drones:
https://user-images.githubusercontent.com/72294070/103081709-7ca30400-45fe-11eb-9dae-78919e6e6c40.png

After that i opened Unreal Engine environment, and executed "takeoff_and_land.py" file of mavsdk package. This is what that was appearing the unreal engine screen:
https://user-images.githubusercontent.com/72294070/103081760-99d7d280-45fe-11eb-86d5-f958e46a8ef7.png

### Settings
contents of settings.json file:
`
{ "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md", "SettingsVersion": 1.2, "SimMode": "Multirotor", "ClockSpeed": 1, "LocalHostIp": "127.0.0.1", "CameraDefaults": { "CaptureSettings": [ { "ImageType": 0, "Width": 3000, "Height": 2000, "FOV_Degrees": 90, "AutoExposureSpeed": 100, "AutoExposureBias": 0, "AutoExposureMaxBrightness": 0.64, "AutoExposureMinBrightness": 0.03, "MotionBlurAmount": 0, "TargetGamma": 1.0, "ProjectionMode": "", "OrthoWidth": 5.12 } ] }, "Vehicles": { "Drone1": { "VehicleType": "PX4Multirotor", "UseSerial": false, "UseTcp": true, "TcpPort": 4560, "ControlPort": 14570, "VehicleSysID": 10, "X": 0, "Y": 0, "Z": 0 }, "Drone2": { "VehicleType": "PX4Multirotor", "UseSerial": false, "UseTcp": true, "TcpPort": 4561, "ControlPort": 14570, "VehicleSysID": 20, "X": 2, "Y": 0, "Z": 0 } } }
`

### How can the issue be reproduced?

1.
2.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2021-01-05:
Hi @AIKUUM! And what happens if you use one PX4 drone without using the script?

AIKUUM, 2021-01-06:
> Hi @AIKUUM! And what happens if you use one PX4 drone without using the script?

Hi @jonyMarino,
PX4 console contents:
[image]

Unreal engine screen:
[image]

AIKUUM, 2021-01-10:
> > Hi @AIKUUM! And what happens if you use one PX4 drone without using the script?
> 
> Hi @jonyMarino,
> PX4 console contents:
> [image]
> 
> Unreal engine screen:
> [image]

Hi @jonyMarino, What could be the reasons behind these issues?

jonyMarino, 2021-05-20:
Already responded in #3213

closed 2021-05-20 by jonyMarino