microsoft/AirSim #3260: Execution of "hello_drone.py" with PX4 in SITL

AIKUUM, 2020-12-29:
## Bug report 

System details: https://user-images.githubusercontent.com/72294070/102069277-90db4a00-3e23-11eb-8f71-d14e10181433.PNG
AirSim version (1.3.1)
Unreal version(4.25)
python version (3.8.3)

### What's the issue you encountered?

When I execute hello_drone.py (link:https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/hello_drone.py) with PX4 in SITL and AirSim, sometimes the drone goes to specified location based on (x,y,z in code), sometimes the drone becomes unstable during the flight and doesnt go to the specified location. Even after the execution of code completes, the drone doesnt land, but lands after some time automatically.

### Settings
Contents of settings.json:
`
{
 "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
 "SettingsVersion": 1.2,
 "SimMode": "Multirotor",
 "ClockSpeed": 1,
 "LocalHostIp": "127.0.0.1",
 "CameraDefaults": {
  "CaptureSettings": [
   {
    "ImageType": 0,
    "Width": 3000,
    "Height": 2000,
    "FOV_Degrees": 90,
    "AutoExposureSpeed": 100,
    "AutoExposureBias": 0,
    "AutoExposureMaxBrightness": 0.64,
    "AutoExposureMinBrightness": 0.03,
    "MotionBlurAmount": 0,
    "TargetGamma": 1.0,
    "ProjectionMode": "",
    "OrthoWidth": 5.12
   }
  ]
 },
 "Vehicles": {
  "Drone1": {
   "VehicleType": "PX4Multirotor",
   "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPort": 14570,
			"VehicleSysID": 10,
   "X": 0,
   "Y": 0,
   "Z": 0
  }
  
}
}
`

### How can the issue be reproduced?

1.
2.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

ahmed-elsaharti, 2021-01-07:
Hi @AIKUUM , This seems to be a consistently recurring problem with PX4 SITL in Windows and Linux (in my testing at least). In my experience, it seems like some HilActuatorControlsMessage(s) do not get through causing the EKF estimate to eventually fall apart and the drone to just crash into the ground.

I'll be posting more testing images/video and maybe a more detailed bug report on this soon.

jonyMarino, 2021-05-20:
Hi @AIKUUM! PX4 is now working. See:
https://github.com/microsoft/AirSim/issues/2477#issuecomment-845451529
[Here you have the new binaries with the fixes](https://github.com/microsoft/AirSim/releases)

closed 2021-05-20 by jonyMarino