microsoft/AirSim #3264: camera settings inside settings.json is not working, only default settings are loaded

lknnnli, 2020-12-30:
Hello everyone,

I tried to set up stereo camera with airsim, and tried to set my own baseline.
I followed the suggestions from
[1770](https://github.com/Microsoft/AirSim/issues/1770)
[1790](https://github.com/microsoft/AirSim/issues/1790)
to modify my settings.json file. I copied the settings.json provided inside [1790](https://github.com/microsoft/AirSim/issues/1790).
However, the camera settings of left camera and right camera are not taken, still when i print the camera_position of simGetImage(), the results is the camera position of default camera setting.

Anyone has any opinion about this issue? Any help would be appreciated.
Best Regards

ahmed-elsaharti, 2020-12-30:
Hi @lknnnli , I just tested this out and got these results:

settings.json:
```json
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ClockSpeed": 1.0,
  "SubWindows": [
  {"WindowID": 0, "ImageType": 0, "CameraName": "front-left", "Visible": true},
  {"WindowID": 2, "ImageType": 0, "CameraName": "front-right", "Visible": true}
  ],
  "Vehicles": {
    "SimpleFlight": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      },
      "Cameras": {
        "front-left": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90
            }
          ],
          "X": 0.50, "Y": -20.0, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        },
        "front-right": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90
            }
          ],
          "X": 0.50, "Y": 20.0, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        }
      },
      "X": 0, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    }
  }
}
```

Test script:

```python
import airsim

import pprint
import tempfile
import os
import time

client = airsim.VehicleClient()

tmp_dir = os.path.join(tempfile.gettempdir(), "airsim_drone")
print ("Saving images to %s" % tmp_dir)

responses = client.simGetImages([
    airsim.ImageRequest("front-left", airsim.ImageType.Scene),
    airsim.ImageRequest("front-right", airsim.ImageType.Scene)])

for i, response in enumerate(responses):
    if response.pixels_as_float:
        print("Type %d, size %d, pos %s" % (response.image_type, len(response.image_data_float), pprint.pformat(response.camera_position)))
        airsim.write_pfm(os.path.normpath(os.path.join(tmp_dir, str(1) + "_" + str(i) + '.pfm')), airsim.get_pfm_array(response))
    else:
        print("Type %d, size %d, pos %s" % (response.image_type, len(response.image_data_uint8), pprint.pformat(response.camera_position)))
        airsim.write_file(os.path.normpath(os.path.join(tmp_dir, str(i), str(1) + "_" + str(i) + '.png')), response.image_data_uint8)
```

Camera views with a baseline of 40m (exaggerated to make the effect visible):

[image]

Script output:

[image]

Images saved:

[image]

Did you make sure to change the camera names in `ImageRequest` to `front-left` and `front-right` so that they match the camera names in settings.json?

lknnnli, 2020-12-31:
> Hi @lknnnli , I just tested this out and got these results:
>
> settings.json:
>
> ```json
> {
>   "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
>   "SettingsVersion": 1.2, [...]

Hi,  thank you for your kind reply, you are correct, it is because the camera name of  ImageRequest and  key inside setting.json are not matched. Thanks a lot for your help. It works.

closed 2021-01-13 by jonyMarino