microsoft/AirSim #3267: No pedestrians in City Environment 1.3.1

hturki, 2020-12-31:
## Bug report 

- AirSim Version/#commit: 1.3.1
- UE/Unity version: N/A
- autopilot version: N/A
- OS Version: N/A

### What's the issue you encountered?

The City environment has moving cars and pedestrians in 1.2.2, but they seem to have vanished in 1.3.1 outside of some parked cars. Is this intentional / intended as a "fix" to https://github.com/microsoft/AirSim/issues/2461?

### Settings

### How can the issue be reproduced?

1. Download City 1.3.1 binaries from website
2. Start exe

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

nightduck, 2021-01-25:
Same with the 1.4.0 release

ArthurYdalgo, 2021-09-07:
same with 1.5.0 release

closed 2021-10-06 by jonyMarino