microsoft/AirSim #3276: Some problem about the Unitydemo/carDemo

MiaoMiaoMeng, 2021-01-06:
https://github.com/microsoft/AirSim/blob/68ae8de7065d2225a1c211f656ec875ee2e622f6/Unity/UnityDemo/Assets/AirSimAssets/Scripts/Vehicles/VehicleCompanion.cs#L197

Hi, I have some question about this function, It seem like we didn't use this function. 
So I try to annotation this part of code.
But when I try to run the  carDemo, the Editor will crash.
From the Log i found

Loaded scene 'Assets/Scenes/CarDemo.unity'
	Deserialize:            27.018 ms
	Integration:            13.212 ms
	Integration of assets:  1.326 ms
	Thread Wait Time:       6.267 ms
	Total Operation Time:   47.824 ms
NullReferenceException: Object reference not set to an instance of an object
  at AirSimUnity.VehicleCompanion.Reset (System.String vehicleName) [0x00025] in C:\AirSim-master\Unity\UnityDemo\Assets\AirSimAssets\Scripts\Vehicles\VehicleCompanion.cs:146 

https://github.com/microsoft/AirSim/blob/68ae8de7065d2225a1c211f656ec875ee2e622f6/Unity/UnityDemo/Assets/AirSimAssets/Scripts/Vehicles/VehicleCompanion.cs#L146

But I didn't found which code use it, except “getfunctionpointerfordelegate”.

Is there anyone else have the same situation like me？

ahmed-elsaharti, 2021-01-07:
Hi @MiaoMiaoMeng , it looks like this was added as a placeholder for camera distortion in e5967a0c8d7a980761776b6a08d40c6014ef11c1 of #3039

MiaoMiaoMeng, 2021-01-08:
Thx for your help@ahmed-elsaharti, 
I read the part of code in e5967a0, but in this part:

        private static bool SetDistortionParam(string cameraName, string paramName, float value, string vehicleName) {
            var vehicle = Vehicles.Find(element => element.vehicleName == vehicleName);
            return vehicle.VehicleInterface.SetDistortionParam(cameraName, fov_degrees);
        }

the value which named as fov_degrees is not found. I think that is the 'value'
So I try to rename float 'value' to fov_degrees, but i didn't found the override function with two param.

BTW,I also didn‘t found any reference about this function.

MiaoMiaoMeng, 2021-01-08:
> Hi @MiaoMiaoMeng , it looks like this was added as a placeholder for camera distortion in [e5967a0](https://github.com/microsoft/AirSim/commit/e5967a0c8d7a980761776b6a08d40c6014ef11c1) of #3039

I read the src in this morning and set a try catch in Reset() function, from the result I think that Vehicles can't Find the element with 'vehicleName'

private static bool Reset(string vehicleName)
        {
            try
            {
                var vehicle = Vehicles.Find(element => element.vehicleName == vehicleName);
                vehicle.VehicleInterface.ResetVehicle();
            }
            catch (Exception e) {
                Debug.Log("vehicle is null");
                throw e;
            }
            return true;
        }

that's make me feel confuse, I think the **vehicleName** is already pass in _InitDelegators()_ when the _StartVehicleServer()_ been used.
So I can't find the reason why it has a NullReference in _Reset()_ function.

ahmed-elsaharti, 2021-01-08:
> Thx for your help@ahmed-elsaharti,
> I read the part of code in [e5967a0](https://github.com/microsoft/AirSim/commit/e5967a0c8d7a980761776b6a08d40c6014ef11c1), but in this part:
>
> ```
>     private [...]

My understanding is that camera distortion hasn't been implemented in Unity yet. I could be wrong though since I barely looked into the code within the PR. @saihv would probably be able to answer this better than me.

jonyMarino, 2021-01-12:
Yes, that is correct; the programmable camera distortion was not implemented for Unity.

MultiPsy, 2021-01-13:
Hi @MiaoMiaoMeng,
so how did u build the system? I tried to comment out the SetDistortionParam function and the build process seemed quite ok until Unity said "Problem in starting AirSim server".

kengregson, 2021-01-13:
Encountered same, last working Unity version I found  can be checked out via:
'git checkout 7f48ba3aa8f61b6082215353c1318ee9174e0759'

david9705, 2021-01-14:
Hello,  @MiaoMiaoMeng .
I also encounter same problem. 
Unity build is currently broken when annotate the `SetDistortionParam` function, but without annotation Unity report  'IVehicleInterface' does not contain a definition for 'SetDistortionParam' and no accessible extension method 'SetDistortionParam'

How do you solve those errors?

MiaoMiaoMeng, 2021-01-17:
> Hi @MiaoMiaoMeng,
> so how did u build the system? I tried to comment out the SetDistortionParam function and the build process seemed quite ok until Unity said "Problem in starting AirSim server".

Hi @MultiPsy,
I have not successfully built this system, as I said there is a null reference in Reset() function.
And for now, I didn't spend time on it. Because I need to deal with the transplantation of a slam system.
So if I have time after, I will find out how to avoid this problem.

zimmy87, 2021-01-25:
Hi @MiaoMiaoMeng and @MultiPsy, thank you for reporting and testing this issue. I just created a PR that should address both issues you are experiencing with the UnityDemo. Please apply this PR and test again and let me know if you are still seeing this issue.

MiaoMiaoMeng, 2021-01-26:
> Hi @MiaoMiaoMeng and @MultiPsy, thank you for reporting and testing this issue. I just created a PR that should address both issues you are experiencing with the UnityDemo. Please apply this PR and test again and let me know if you are still seeing this issue.

Thanks @zimmy87 ,

It works well.
I will close the issue.

Thank you again.

closed 2021-01-26 by MiaoMiaoMeng

kengregson, 2021-01-26:
@zimmy87, please accept my thanks as well!
Also verified under Ubuntu 20.04/Unity 2019.3.12f1 with PX4-SITL/QGroundControl.

Note:  Issue 1617 still seems to prevent proper operation of PX4 SITL unless settings.json and VehicleCompanion.cs are modified as described in the workaround.

PR #3343 (2021-02-02): Fix UnityDemo build

Fixes: #3276

## About
This change fixes the build break caused by VehicleCompanion.SetDistortionParam referencing a non-existent parameter. This also fixes an issue where the DllImport of InitVehicleManager was missing two parameters that were added to the API in the AirLibWrapper. 

## How Has This Been Tested?
AirLibWrapper was built with this PR applied and it was confirmed that both car and drone mode in the UnityDemo project can launch without issue.

## Screenshots (if appropriate):