microsoft/AirSim #3279: Adding a sensor to a VehicleType of ArduPilot causes it not to work

MoBaT, 2021-01-06:
## Bug report 

- AirSim Version/#commit: v1.3.1-linux
- UE/Unity version: 4.25.1-0+++UE4+Release-4.25 518 0
- autopilot version: b0fa5fa
- OS Version: Ubuntu 18.04

### What's the issue you encountered?

[image]

### Settings

https://gist.github.com/MoBaT/4dc5d1c2a836ee489820e7d6cb2e5a81

### How can the issue be reproduced?

1. Add a "Sensors" block and add any type of sensor to it. In my case I added a Lidar (6)

rajat2004, 2021-01-07:
This is occurring due to the GPS and IMU sensors not being present in the settings, they are required for AP to work, see the settings on this page - https://ardupilot.org/dev/docs/sitl-with-airsim.html#using-lidar

If the `Sensors` block is present in the settings, only the specified sensors will be created since AirSim allows creating vehicles without any sensors as well. There's some discussion which can be done as to ensure the minimum sensors required for the specific vehicle are always present, something like `DefaultSensors: true` maybe.

MoBaT, 2021-01-07:
@rajat2004 Thanks for this info! This worked for me. I was under the assumption that IMU's and GPS would be on there by default.

closed 2021-01-07 by MoBaT