microsoft/AirSim #3288: Airsim Only Performing Step Every Few Seconds

TomoCoder123, 2021-01-08:
## Question
### What's your question?
How do I speed up the steps in Airsim? (new to Airsim)

### Include context on what you are trying to achieve
I am not sure how many steps Airsim is supposed to perform every few seconds but because the steps are going so slow, my AI is unable to detect collisions. This is the case because it collides into an object and then the collision goes off but a step isn't happening at that moment so when it stops colliding with that object, sometimes a step happens so it does not detect the collision.

#### Context details
 Windows 10, Python 3.6, Unreal Engine 
 Settings.json is Multirotor

### Include details of what you already did to find answers

ahmed-elsaharti, 2021-01-08:
Not entirely sure if you're referring to the simulation clock or the API call speed when saying steps.
If you're referring to the simulation clock, you could slow down or speed up the sim clock using `ClockSpeed` in [settings.json](https://microsoft.github.io/AirSim/settings/) (for example, with a clock speed of 0.25, 1 real-life second = 0.25 simulation seconds. ie: more calculations per real life second, etc #3206).
If you're referring to the API call speed, in my testing they were pretty fast at about 0.003s per call.

TomoCoder123, 2021-01-10:
In the case of my project which is using reinforcement learning, I am pretty sure I am referring to API because when I put a print statement into the step function, it printed out every 4 seconds so I am not sure how to speed that up.

ahmed-elsaharti, 2021-01-10:
There's a good chance the delay is due to something within your function rather than the API call since when I tested this by calling getMultirotorState (as an example) multiple times in a row, the timestamp difference was ~0.003s

What API are you calling in your function?

TomoCoder123, 2021-01-10:
We are calling several APIS including self.drone.getMultirotorState().kinematics_estimated.linear_velocity and self.drone.moveByVelocityAsync.

ahmed-elsaharti, 2021-01-10:
So I just tested this using:
```python
t = time.time()
while True:
    print(client.getMultirotorState().kinematics_estimated.linear_velocity)
    client.moveByVelocityAsync(1, -0.67, -5,4)
    print(time.time() - t)
    t = time.time()
```
and this printed the linear velocity vector with a time difference of 0.00199s (I did the test this way to see the actual delay in execution rather than the messages' timestamp difference).

The only way I was able to get it to stop for 4 seconds before printing the next line was to have the duration argument of the `moveByVelocityAsync` set to 4s (as shown above) and calling join() to wait for the task to complete:
```python
t = time.time()
while True:
    print(client.getMultirotorState().kinematics_estimated.linear_velocity)
    client.moveByVelocityAsync(1, -0.67, -5,4).join()
    print(time.time() - t)
    t = time.time()
```

TomoCoder123, 2021-01-10:
I believe that was the reason. I copied the file drone_env and modified that so we did not have a full understanding of the code. Thank you so much, I really appreciate it.

TomoCoder123, 2021-01-10:
Yep it was the fix, thanks again.

ahmed-elsaharti, 2021-01-10:
Glad it fixed the problem :D 
In general, calling join() holds the execution of the code until the particular API task being called is complete.
If you've got any other issues we're all here to help ^_^

jonyMarino, 2021-01-12:
Hi, @TomoCoder123! And Welcome! I'm glad that @ahmed-elsaharti helped you solve this!

closed 2021-01-12 by jonyMarino