microsoft/AirSim #3291: Airsim crash while training

CoolCoder54323, 2021-01-11:
## Bug report 

- AirSim Version/#commit: 1.3.1
- UE/Unity version: 4.25.4
- OS Version: windows 10

### What's the issue you encountered?
The simulation crashes after any amount of training time, could be 2k steps or 100k. It gives this error:
[image]

### Settings
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings_json.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ClockSpeed": 1.0,
  "Vehicles": {
    "SimpleFlight": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      }
    }
  }
}

### How can the issue be reproduced?
It happens every time I train I have yet to get a full session in.

### Include full error message in text form
Unhandled exception at 0x00007FF981AA96AD (UE4Editor-AirSim-Win64-DebugGame.dll) in UE4Editor-Win64-DebugGame.exe: 0xC0000005: Access violation reading location 0x0000000000009074.

jonyMarino, 2021-01-11:
Hi @CoolCoder54323! Can you provide more details on how you are training the drone so we can reproduce the issue?

CoolCoder54323, 2021-01-11:
I am using the rl files that came with airsim I modeified the dron_env in airgym and made a custom file to run it called dronetrain
I also moved airgym into pythonclient/multirotor and that is all the changes I made.

Drone_Env: 
[drone_env.txt](https://github.com/microsoft/AirSim/files/5798413/drone_env.txt)

DroneTrain:
[DroneTrain.txt](https://github.com/microsoft/AirSim/files/5798419/DroneTrain.txt)

CoolCoder54323, 2021-01-11:
One more thing to add. It usually happens when I click into unreal engine after being tabbed out and interact with it, but not always it happened one time when I already had UE in focus and I was away from my computer. Hope this helps!

CoolCoder54323, 2021-01-12:
@jonyMarino Were you by any chance able to reproduce the error.

saihv, 2021-01-12:
@CoolCoder54323 Could you please run the game in DebugGameEditor mode (in Visual Studio), and show a screenshot of the call stack that appears in the bottom right when the crash happens?

CoolCoder54323, 2021-01-12:
Ok, no problem my take a few hours considering when it crashed in the past. Kind of annoying though because I remember looking at that and for some reason, I didn't screenshot it.

CoolCoder54323, 2021-01-13:
[image]

> @CoolCoder54323 Could you please run the game in DebugGameEditor mode (in Visual Studio), and show a screenshot of the call stack that appears in the bottom right when the crash happens?

Here you go. It looks like it came from an unreal file RenderRequest. I wonder if this is a problem with unreal and not Airsim.

saihv, 2021-01-14:
This is quite odd, I don't think I've come across this issue before. It seems like something is going wrong with the recording/image capture functionality. Are you using the Blocks environment? 

And I see you mentioned you were using 1.3.1, but not the binary, so are you working with an older commit? If yes, perhaps you could retry with the latest master?

CoolCoder54323, 2021-01-14:
I am using my own custom environment not blocks, and I don't know what you mean by binary (sorry I am new to this), but I last updated airsim was nov 25. I will try to reinstall and see what happens.

ahmed-elsaharti, 2021-01-14:
> I don't know what you mean by binary

@CoolCoder54323 binaries are the pre-compiled Unreal Engine environments found in the [releases section](https://github.com/microsoft/AirSim/releases). Those are ready-to-go executable environments that were built with the latest (at the time of release) Airsim commit

CoolCoder54323, 2021-01-14:
So do you think It is because of my environment? I will try it on blocks and see what happens.