microsoft/AirSim #3292: No Transmission Towers in Landscape Mountains Unreal Environment

edwardjan, 2021-01-11:
## Question
### What's your question?

I've been following the provided tutorial for RL with Quadrotor in https://microsoft.github.io/AirSim/reinforcement_learning/.
I proceeded to download and create a project in using the Landscape Mountains Unreal Environment.

However, the Unreal Environment doesn't seem to have transmission towers. Do they have to be added in? Or is there an environment with them available?

### Include context on what you are trying to achieve

I need to run the provided source code for RL with Quadrotor, so that I can see it in action personally.

#### Context details

OS: Windows 10
AirSim Version: 1.3.1
Python: 3.7.9
Unreal Version: 4.25.4

Setings.json:
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings_json.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ClockSpeed": 1.0,
  "Vehicles": {
    "SimpleFlight": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      }
    }
  }
}

### Include details of what you already did to find answers

jonyMarino, 2021-01-12:
Hi @edwardjan! and Welcome! The Landscape Mountains environment in 1.3.1 doesn't have power lines. But the one in 1.2 has it. So, try with the binary from that version and tell us if you get another issue (you will not be able to use the new API methods that were added since 1.2)

edwardjan, 2021-01-13:
Thank you for the response. I was able to get the drone up and running. For any other people with the same issue, you will need Python 3.6 since CNTK is needed and it can't be installed in Python 3.7 and onwards.

I think we can close this query, thanks again.

closed 2021-01-13 by edwardjan