microsoft/AirSim #3293: IMU Kalibr parameters for AirSim

alberto1995, 2021-01-11:
## Question
### What's your question?
Hi everyone,
I'm trying to run some VIO algorithms (Open-VINS and VINS-Mono) using a rosbag recorded with the drone of AirSim. I'm here to ask if someone knows, and can share, IMU noise parameters in [Kalibr](https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model) format.

### Include context on what you are trying to achieve

I would like to run VIO algorithms with the exact IMU noise parameters needed by Open-VINS and VINS-Mono (Gyroscope "white noise" - Accelerometer "white noise" - Gyroscope "random walk" - Accelerometer "random walk").

#### Context details

### Include details of what you already did to find answers
First of all, I have seen that In the [SimpleIMU](https://github.com/microsoft/AirSim/blob/master/AirLib/include/sensors/imu/ImuSimpleParams.hpp#L18) code there are four parameters setted to emulate the MPU-6000, but them don't match with the Kalibr format. Also a Datasheet of the IMU is given, but again I can't find IMU noise parameters in Kalibr format. Secondly, I have seen that other issues #2369 #3269 #2836 have been opened and are still pending, so the pubblication of those parameters can help a lot.

Thank you for your support,
sincerely,

Alberto

jonyMarino, 2021-01-12:
Hi @alberto1995! And welcome! I think you are asking for a feature request to add  IMU noise parameters in Kalibr format, that we do not have.

karnikram, 2021-02-04:
Hi @alberto1995, I had the same issue.

I was able to infer the noise model parameters from these two scripts - [ImuSimpleParams](https://github.com/microsoft/AirSim/blob/master/AirLib/include/sensors/imu/ImuSimpleParams.hpp), [ImuSimple](https://github.com/microsoft/AirSim/blob/238c8e708dab831bf5835b1c954abfb0136e9277/AirLib/include/sensors/imu/ImuSimple.hpp#L71)

`gyro.arw` is the `gyroscope_noise_density`, and is equal to `8.7266462e-5`
`gyro_bias_stability_norm` is the `gyroscope_random_walk`, and is equal to `9.9735023e-7`
`accel.vrw` is the `accelerometer_noise_density`, and is equal to `0.002353596`
`accel_bias_stability_norm` is the `accelerometer_noise_density`, and is equal to `1.2481827e-5`

I also verified these values using [kalibr_allan](https://github.com/rpng/kalibr_allan) after recording a four hour static sequence, and the values are very similar:

[image]
[image]

I'm not sure about Open-VINS, but VINS-Mono expects these parameters in discrete-time form. Meaning, the noise density parameters need to be divided by the square root of your sampling time, and random walk parameters need to be multiplied with the square root of your sampling time. Note that the airsim sampling time has an [upper limit](https://github.com/microsoft/AirSim/blob/238c8e708dab831bf5835b1c954abfb0136e9277/AirLib/include/sensors/imu/ImuSimple.hpp#L78) of 0.001 secs. More over in my case I also noticed a time offset between camera and IMU which I had to separately calibrate using Kalibr.

Hope this helps

GimpelZhang, 2021-02-05:
The two VIO, open_vins and vins_mono, expect different form of IMU parameters. 

* open_vins: continuous time noise parameters

[evidence:](https://github.com/rpng/open_vins/blob/master/ov_msckf/src/state/Propagator.cpp#L381-L392)

```
    // Construct our discrete noise covariance matrix
    // Note that we need to convert our continuous time noises to discrete
    // Equations (129) amd (130) of Trawny tech report
    Eigen::Matrix<double,12,12> Qc = Eigen::Matrix<double,12,12>::Zero();
    Qc.block(0,0,3,3) = _noises.sigma_w_2/dt*Eigen::Matrix<double,3,3>::Identity();
    Qc.block(3,3,3,3) = _noises.sigma_a_2/dt*Eigen::Matrix<double,3,3>::Identity();
    Qc.block(6,6,3,3) = _noises.sigma_wb_2*dt*Eigen::Matrix<double,3,3>::Identity();
    Qc.block(9,9,3,3) = _noises.sigma_ab_2*dt*Eigen::Matrix<double,3,3>::Identity();

    // Compute the noise injected into the state over the interval
    Qd = G*Qc*G.transpose();
    Qd = 0.5*(Qd+Qd.transpose());
```

where sigma_w_2/sigma_a_2/sigma_wb_2/sigma_ab_2 are the squares of the IMU param inputs: [from ros](https://github.com/rpng/open_vins/blob/master/ov_msckf/src/utils/parse_ros.h#L153-L157)

* vins_mono: as @karnikram pointed out, discrete time noise parameters

evidence:
[construct noise matrix](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/vins_estimator/src/factor/integration_base.h#L21-L27):
```
    noise = Eigen::Matrix<double, 18, 18>::Zero();
    noise.block<3, 3>(0, 0) =  (ACC_N * ACC_N) * Eigen::Matrix3d::Identity();
    noise.block<3, 3>(3, 3) =  (GYR_N * GYR_N) * Eigen::Matrix3d::Identity();
    noise.block<3, 3>(6, 6) =  (ACC_N * ACC_N) * Eigen::Matrix3d::Identity();
    noise.block<3, 3>(9, 9) =  (GYR_N * GYR_N) * Eigen::Matrix3d::Identity();
    noise.block<3, 3>(12, 12) =  (ACC_W * ACC_W) * Eigen::Matrix3d::Identity();
    noise.block<3, 3>(15, 15) =  (GYR_W * GYR_W) * Eigen::Matrix3d::Identity();
```

[use the noise in the covariance prediction](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/vins_estimator/src/factor/integration_base.h#L125):
```
    covariance = F * covariance * F.transpose() + V * noise * V.transpose();
```
where ACC_N/GYR_N/ACC_W/GYR_W are the IMU param inputs: [read settings](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/vins_estimator/src/parameters.cpp#L70-L73)

We can see in vins_mono, there is no division or multiplication of **dt**.

closed 2021-02-17 by alberto1995