microsoft/AirSim #3294: Build environment error :  LNK1181: cannot open input file 'rpc.lib' (VS2019, Unity)

david9705, 2021-01-11:
## Bug report 

- AirSim Version/#commit:  Latest Version AirSim download for github
- UE/Unity version: Unity 2019.4
- autopilot version: N/A
- OS Version: Windows 10
- Visual Studio 2019 Community

### What's the issue you encountered?

build environment error 
I follow the following website suggestion and fix all the bug except the last bug (**cannot open “rpc.lib”**) 

https://blog.csdn.net/yldmkx/article/details/108694287?fbclid=IwAR2mj22YrcQCJasPrEb-Y5LlnivKeYpJxpU6A4NyArAjUh_GaxHQupeOy6A

but I think the real problem is that rpc.lib is not put the correct path because I have generated the **rpc.lib** by CMAKE successfully.

Please tell me the correct path.

Thanks. 
   

<br><br>
Error Message 

[image]
<br>

Path of rpc.lib

[image]
 
<br>
Other file path

[image]

rajat2004, 2021-01-12:
I believe you need to use the Developer Command Prompt instead of the x86 Native Tools one

david9705, 2021-01-12:
@rajat2004  Developer Command Prompt cannot solve the problem. Thanks

rajat2004, 2021-01-12:
That's pretty strange, Unity build is working on Github Actions CI atleast, which is using the Developer Command Prompt
The error is occuring when running `build.cmd` correct? Could you post the entire output in text format if possible? 
I'll also try it out when back on a Windows machine

david9705, 2021-01-12:
Yes, the error is occuring when running `build.cmd`
<br> the follow is entire output and  use **x86 Native Tools** 

<br><br> <br>

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\Unity>build.cmd

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\Unity>REM //---------- copy binaries and include for MavLinkCom in deps ----------

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\Unity>msbuild AirLibWrapper\AirsimWrapper.sln  /target:Clean /target:Build  /property:Configuration=Release /property:Platform=x64
Microsoft (R) Build Engine for .NET Framework 16.8.2+25e4d540b 版
Copyright (C) Microsoft Corporation. 著作權所有，並保留一切權利。

在此解決方案中一次建置一個專案。若要啟用平行建置，請新增 "-m" 參數。
已經開始建置於 2021/1/12 下午 05:54:41。
節點 1 (Clean;Build 目標) 上的專案 "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\Unity\AirLibWrapper\AirsimWrapper.sln"。
ValidateSolutionConfiguration:
  建置方案組態 "Release|x64"。
專案 "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\Unity\AirLibWrapper\AirsimWrapper.sln" (1) 正在節點 1 (Clean 目標) 上建置 "C:\Program Files (x86)\Micros
oft Visual Studio\2019\Community\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj.metaproj" (2)。
專案 "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj.metaproj" (2) 正在節點 1 (Clean 目標) 上建置 "C:
\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\AirLib\AirLib.vcxproj" (3)。
專案 "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\AirLib\AirLib.vcxproj" (3) 正在節點 1 (Clean 目標) 上建置 "C:\Program Files (x86)\Microsoft Visual Studi
o\2019\Community\AirSim\MavLinkCom\MavLinkCom.vcxproj" (4)。
_PrepareForClean:
  正在刪除檔案 "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\MavLinkCom\temp\MavLinkCom\Win32\Debug\MavLinkCom.tlog\MavLinkCom.lastbuildstate"。
CoreClean:
  正在刪除檔案 "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\MavLinkCom\lib\Win32\Debug\MavLinkCom.pdb"。
  正在刪除檔案 "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\MavLinkCom\temp\MavLinkCom\Win32\Debug\MavLinkCom.pdb"。
專案 "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\MavLinkCom\MavLinkCom.vcxproj" (Clean 目標) 建置完成。

專案 "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\AirLib\AirLib.vcxproj" (Clean 目標) 建置完成。

專案 "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj.metaproj" (Clean 目標) 建置完成。

專案 "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\Unity\AirLibWrapper\AirsimWrapper.sln" (1) 正在節點 1 (預設目標) 上建置 "C:\Program Files (x86)\Microsoft
Visual Studio\2019\Community\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj.metaproj" (2:2)。
專案 "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj.metaproj" (2:2) 正在節點 1 (預設目標) 上建置 "C:\P
rogram Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\AirLib\AirLib.vcxproj" (3:4)。
專案 "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\AirLib\AirLib.vcxproj" (3:4) 正在節點 1 (預設目標) 上建置 "C:\Program Files (x86)\Microsoft Visual Studio\
2019\Community\AirSim\MavLinkCom\MavLinkCom.vcxproj" (4:2)。
InitializeBuildStatus:
  由於指定了 "AlwaysCreate"，因此正在建立 "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\MavLinkCom\temp\MavLinkCom\Win32\Debug\MavLinkCom.tlog\unsuccessfulb
  uild"。
ClCompile:
  C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.28.29333\bin\HostX86\x86\CL.exe /c /Icommon_utils /Iinclude /ZI /JMC /nologo /W3 /WX /d
  iagnostics:column /Od /Oy- /D _WIN32_WINNT=0x0600 /D _WIN32 /D WIN32 /D _SCL_SECURE_NO_WARNINGS /D _CRT_SECURE_NO_WARNINGS /D _DEBUG /D _LIB /D _UNICODE /D UNICODE /G
  m- /EHsc /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /std:c++17 /Fo"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\MavLinkCom
  \temp\MavLinkCom\Win32\Debug\\" /Fd"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\MavLinkCom\temp\MavLinkCom\Win32\Debug\MavLinkCom.pdb" /FR"C:
  \Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\MavLinkCom\temp\MavLinkCom\Win32\Debug\\" /Gd /TP /analyze- /FC /errorReport:queue common_utils\Fil
  eSystem.cpp common_utils\ThreadUtils.cpp src\UdpSocket.cpp src\impl\UdpSocketImpl.cpp src\AdHocConnection.cpp src\impl\AdHocConnectionImpl.cpp src\impl\MavLinkFtpClie
  ntImpl.cpp src\impl\MavLinkNodeImpl.cpp src\impl\MavLinkTcpServerImpl.cpp src\impl\onecore\OneCoreFindSerialPorts.cpp src\impl\windows\WindowsFindSerialPorts.cpp src\
  MavLinkFtpClient.cpp src\MavLinkLog.cpp src\MavLinkMessageBase.cpp src\MavLinkMessages.cpp src\MavLinkNode.cpp src\Semaphore.cpp src\MavLinkTcpServer.cpp src\MavLinkV
  ehicle.cpp src\impl\MavLinkVehicleImpl.cpp src\MavLinkConnection.cpp src\impl\MavLinkConnectionImpl.cpp src\MavLinkVideoStream.cpp src\impl\MavLinkVideoStreamImpl.cpp
   src\serial_com\SerialPort.cpp src\serial_com\SocketInit.cpp src\serial_com\TcpClientPort.cpp src\serial_com\UdpClientPort.cpp src\serial_com\wifi.cpp
  FileSystem.cpp
  ThreadUtils.cpp
  UdpSocket.cpp
  UdpSocketImpl.cpp
  AdHocConnection.cpp
  AdHocConnectionImpl.cpp
  MavLinkFtpClientImpl.cpp
  MavLinkNodeImpl.cpp
  MavLinkTcpServerImpl.cpp
  OneCoreFindSerialPorts.cpp
  WindowsFindSerialPorts.cpp
  MavLinkFtpClient.cpp
[... cut ...]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\Unity>if ERRORLEVEL 1 goto :buildfailed

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\Unity>echo(

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\Unity>echo #### Build failed - see messages above.  1>&2
#### Build failed - see messages above.

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\Unity>exit /b 1

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\AirSim\Unity>`

ahmed-elsaharti, 2021-01-12:
@david9705 It looks to me link the guide you're using misses the step of building Airsim using `build.cmd` __under the Airsim main directory__ (ie: going to the Airsim directory with the developer command prompt and running build.cmd) . Try following [this guide](https://microsoft.github.io/AirSim/Unity/) and see if the problem persists.

david9705, 2021-01-12:
Hello, @ahmed-elsaharti   [this guide](https://microsoft.github.io/AirSim/Unity/) tell use `build.cmd` in the **AirSim\Unity directory**

ahmed-elsaharti, 2021-01-12:
Hi @david9705, this is the section i'm refering to:
[image]

ahmed-elsaharti, 2021-01-12:
tbh at this point I would recommend just recloning the repo and starting from scratch while following the official guide. I'm trying to figure out what issue your prompt is having with the download_install section but with no luck.

david9705, 2021-01-12:
@ahmed-elsaharti  Thanks your suggestion. I'll recloning the repo later and report the result in this issue.  I guess the x64 command prompt maybe relevant.

david9705, 2021-01-14:
I solve the bug and `build.cmd` success now.

Open **AirSim\AirLib\AirLib.vcxproj** in VS2019 and we can see two project .One is **AirLib**  and the other is **AirSimWrapper**.

<br>

_(The follow chinese word I'm not sure those in english. )_

<br><br>

### In AirLib project
You should generate **rpc.lib**  first.
<br>
Right click the AirLib project and click **properties**.
Go to **manifest Tool > General > Additional Dependencies** and add the **rpc.lib** file path.
Finally, rebuild AirLib or use  `build.cmd` in command prompt .

<br>

[image]

<br>
In AirSimWrapper project has similar problem but only to set the build configurations and platform and rebuild MavLinkCom.lib.

<br><br><br>

Now I open project in Unity2019.4 , and there still have bugs....
If I cannot solve those problem, maybe open a new issue later.

closed 2022-01-14 by zimmy87