microsoft/AirSim #3297: Low FPS in ROS with AirSim drone

giovannifagotti, 2021-01-13:
## Question
### What's your question?
Hello everyone, 
I'm using the Unreal Plugin of Airsim in combination with ROS to record a rosbag with the drone.
Since I have a very low frame rate in ROS, I would like to increase it reducing the parameter "ClockSpeed" in settings.json.
After this setting, I effectively noticed that the simulator speed has become slower, but in ROS I continue to record with the old frame rate.
There are some parameters to set also in ROS that I have missed? 

### Include context on what you are trying to achieve
I would like to record a rosbag with the AirSim drone at 20fps for each camera.

#### Context details

### Include details of what you already did to find answers
I'm using the "NoDisplay" mode in the settings.json and, with a clock speed of 1.0, I can obtain 10 fps.
Consequntly, I set the clock speed to 0.5, but I continue to get 10 fps while recording the rosbag.
In the launch file, I have setted the "update_airsim_img_response_every_n_sec" at 0.01.
Can someone explain how to use the parameter "ClockSpeed" to get a higher frame rate in ROS?

Thank you for your support, 

GF

GimpelZhang, 2021-01-14:
From my experience, in the "Multirotor" SimMode, the ClockSpeed setting does work (AirSim version until early September, 2020) with ROS. Which means ClockSpeed lower than 1.0 could make the image freq "more dense". 

Maybe you can check the timestamps of the image rostopics directly to see if they get higher frequency instead of a "rostopic hz" command from a terminal.

giovannifagotti, 2021-01-14:
Thanks for your answer,

I have recorded 2 different rosbags with "ClockSpeed" 1.0 and 0.5 and then, as suggested, I have checked the time interval bethween every consecuteve frames: 
- ClockSpeed 1.0 -> 0,1 time interval average, around 10 fps
- ClockSpeed 0.5 -> 0,1 time interval average, around 10 fps

Again, if I play the rosbag recorded with 0.5 ClockSpeed and I watch the image topic with rqt_image_view, I notice that the drone goes slower, instead of having a higher frame rate.

alberto1995, 2021-01-14:
I have the same problem

GimpelZhang, 2021-01-14:
Maybe have a try at the settings:

> "PhysicsEngineName": "FastPhysicsEngine",
"ClockType": "SteppableClock",

And have you seen an screen output similar to the one in the red rectangle? 
[image]

alberto1995, 2021-01-14:
[image]

Tried those settings, nothing changed... I can see the right clockspeed in the simulator, but in ROS I have only a slower recorded video instead of a higher frame rate.

GimpelZhang, 2021-01-14:
> [image]
> 
> Tried those settings, nothing changed... I can see the right clockspeed in the simulator, but in ROS I have only a slower recorded video instead of a higher frame rate.

I'm sorry I forgot those are default settings.

alberto1995, 2021-01-18:
Any suggestion?

Thank you

77788888, 2022-04-26:
how to increate the frame per second of the depth image in AirSim