microsoft/AirSim #3303: Acquisition of GPS  in Computer Vision Mode (C++)

fangice, 2021-01-14:
AirSim Version/#commit: v1.3.1
UE4 version: 4.23

Hi,  Is there anyone who can tell me how to get GPS value in computer vision mode?

ahmed-elsaharti, 2021-01-14:
Hi @fangice , AFAIK a vehicle needs to exist in Airsim to be able to grab the GPS position. Since no vehicles are created in CV mode, I dont think a GPS position would be obtainable (with the APIs atleast)

jonyMarino, 2021-01-14:
Hi @fangice! and welcome! Why you need CV mode?

fangice, 2021-01-15:
> Hi @fangice , AFAIK a vehicle needs to exist in Airsim to be able to grab the GPS position. Since no vehicles are created in CV mode, I dont think a GPS position would be obtainable (with the APIs atleast)
Thank you very much for your reply. I have given up using API to get GPS value，and decided to calculate GPS by using the obtained Pose and the GPS of the playstart.

fangice, 2021-01-15:
> Hi @fangice! and welcome! Why you need CV mode?
Thank you very much for your reply. I want to use CV mode to collect images and GPS data of the scene. Considering that the motion jitter of UAV in Multirotor mode may cause some errors between the image and GPS value, CV mode is adopted.

closed 2021-01-15 by fangice