microsoft/AirSim #3311: Changing physics aspects of Drones

EdwinMeriaux, 2021-01-14:
Hello,

So I was trying to modify the physics of the basic drone to replicate a scenario with a stronger engine and a heavier frame.

So if I understand correctly it is just this file to modify right:
https://github.com/Microsoft/AirSim/blob/v1.1.10/AirLib/include/vehicles/multirotor/configs/SimpleFlightQuadX.hpp
for the physical aspects (drone dimmensions)

and this one for the actual thrust of the drone:
https://github.com/microsoft/AirSim/blob/v1.1.10/AirLib/include/vehicles/multirotor/RotorParams.hpp

So for the weight can it  really only be between .8kg and 1.6kg?

ahmed-elsaharti, 2021-01-15:
Hi @EdwinMeriaux,
On quickly scanning through both files it seems that you're right. However, I recommend giving this a try with the latest version of AirSim (1.4.0 as of a couple of days ago) since the structure has changed since version 1.1.10.

EdwinMeriaux, 2021-01-20:
Awesome thank you!

Out of pure curiosity is would you know if any work is being done to modify the physics engine to allow heavier drones?
Or even allow the ability to switch out the fast physics engine to something that can support heavier drones?

ahmed-elsaharti, 2021-01-20:
Glad to be of help ^_^ 

Also, this would be a question better suited for @jonyMarino and the team. But based on #3284 it seems like there could be something in the works related to the physics engine

EdwinMeriaux, 2021-01-21:
Ok Great!

Just one more clarification. When I modify those files is there anything to be done? Say I change the weight of the drone from 1kg to 1.2kg do I need to rebuild the airsim file? what is there to do other than change the value of the drone?

ahmed-elsaharti, 2021-01-21:
@EdwinMeriaux Correct. You would have to rebuild AirSim. 
If you're working on the blocks environment, you would have to rebuild AirSim and run the update_from_git script.
If you're working on a custom environment, you would need to rebuild AirSim and copy the new Plugins folder over (if I remember the steps correctly 😂)

EdwinMeriaux, 2021-01-21:
@ahmed-elsaharti what update_from_git script?

EdwinMeriaux, 2021-01-21:
The one in "AirSim\Unreal\Environments\Blocks"

Ok so just to make sure the steps to change drone mass or rotor strength are:
1. make the modification to the file
2. rebuild airsim
3. update_from_git

Is this accurate?

ahmed-elsaharti, 2021-01-21:
This is the update_from_git script:
[image]
its the same one from [the blocks environment docs](https://microsoft.github.io/AirSim/unreal_blocks/).
And yup, this is assuming that you're modifying the MAIN files in the airsim directory/solution (Airsim.sln) and not the Blocks solution.

If you're modifying the files in the Blocks project, you would need to:
1. make the modification to the file in the blocks solution (Blocks.sln file)
2. run update_**to**_git
3. rebuild airsim
4. run update_from_git
5. open the Blocks.sln and run the simulation

EdwinMeriaux, 2021-01-21:
ok when you say rebuild just to confirm is that the ./build.sh command from airsim install process or the build button in the blocks environment?

ahmed-elsaharti, 2021-01-21:
If youre on linux, yes rebuilding would be through build.sh.

EdwinMeriaux, 2021-01-21:
The file that should be moddifed I am slightly confused now.
Say the blocks file for the environment with the drone I would like to change mass is called blocks-4

You said modifying the main files does that mean the:
/home/edwinm/AirSim/AirLib/include/vehicles/multirotor/firmwares/simple_flight
/home/edwinm/AirSim/Unreal/Plugins/AirSim/Source/AirLib/include/vehicles/multirotor/firmwares/simple_flight
/home/edwinm/AirSim/Unreal/Environments/Blocks 4.24 - 4/Plugins/AirSim/Source/AirLib/include/vehicles/multirotor/firmwares/simple_flight

cause everytime I modify one of the first 2 it gets over written with update_to_git
and when I modify the 3rd one I don't see a difference in the program
When I change the mass it's position curve stays constant
I would assume when it works the drone would take longer to accelerate to the designated velocity?

ahmed-elsaharti, 2021-01-21:
What update_to_git does is that it copies the changes under the Blocks environment:
/home/edwinm/AirSim/Unreal/Environments/Blocks 4.24 - 4/
to the main directory
/home/edwinm/AirSim
what update_from_git does it the opposite, it updates the blocks environment from the main directory
for example, if i want to make changes to a particular file in AirLib,
I could either make those changes directly in /home/edwinm/AirSim/AirLib/, rebuild airsim (ie: the main airsim project) then do update_from_git to update the blocks environments' AirLib (/home/edwinm/AirSim/Unreal/Environments/Blocks 4.24 - 4/Plugins/AirSim/Source/AirLib/) with the new files and run.
OR
Make the changes under the blocks environment (/home/edwinm/AirSim/Unreal/Environments/Blocks 4.24 - 4/Plugins/AirSim/Source/AirLib/) and directly run the environment. However, from what I've seen sometimes this results in compilation errors so if you want to do it this way, make your changes under the blocks environment (/home/edwinm/AirSim/Unreal/Environments/Blocks 4.24 - 4/Plugins/AirSim/Source/AirLib/) then update_to_git (to copy the files over from the blocks environment to the main project), rebuild airsim (this builds the MAIN airsim project that just got updated with our new files), then run update_from_git to make sure everything is synced and run the blocks project.

Have a look at https://microsoft.github.io/AirSim/dev_workflow/#updating-and-changing-airsim-code

EdwinMeriaux, 2021-01-22:
Ok I'm running into a problem with those steps:
./update_from_git.sh: line 10: ./clean.sh: Permission denied

I used your first method:
1. modified mass of drone at /home/edwinm/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/vehicles/multirotor/firmwares/simple_flight to 1.5kg
2. in ~/Airsim I used ./build.sh
3. when in ~/AirSim/Unreal/Environments/Blocks 4.24 - 4 I tried to ./update_from_git.sh I then get the error

ahmed-elsaharti, 2021-01-22:
Are any of the files within ~/AirSim/Unreal/Environments/Blocks 4.24 - 4 open? that might be preventing the script from cleaning.
I'm not an expert on the topic though so my guess is probably waaaay off 😂

EdwinMeriaux, 2021-01-22:
No such files were open.

I realized there was a file in the same folder as the ./update_from_git.sh called clean.sh
I gave the execute privileges to that and the next time I ran ./update_from_git.sh it worked.

I tested the new drone and it appears to work.

EdwinMeriaux, 2021-01-22:
Thanks for the help!

closed 2021-01-22 by EdwinMeriaux

paobuliao, 2023-08-19:
Hi,
I wonder whether the control APIs are work correctly after changing the parameters, such as weight or length.

AlexanderZ3, 2023-09-25:
> Hi, I wonder whether the control APIs are work correctly after changing the parameters, such as weight or length.

same question? do we need to re-tune pid gains after changing the parameters?