microsoft/AirSim #3314: Cannot run Airsim on Azure VM - cannot find a compatible vulkan device or driver

harrynvfreeman, 2021-01-15:
## Question
### What's your question?

Is it possible to run Airsim in Microsoft Azure as described in https://microsoft.github.io/AirSim/azure/

### Include context on what you are trying to achieve

I'm trying to run AirSim on an Azure vm following https://microsoft.github.io/AirSim/azure/, where the vm used in the provided template is a Standard_NV6 with Tesla M60 GPU

#### Context details

Ubuntu 18.04, Airsim version 1.2.6, Unreal 4.24

### Include details of what you already did to find answers
I'm trying to run AirSim on an Azure vm following https://microsoft.github.io/AirSim/azure/, where the vm used in the provided template is a Standard_NV6 with Tesla M60 GPU.
However, when I launch unreal editor I get the error "cannot find a compatible vulkan device or driver", and had no success trying every solution I could find online.  
Turns out, vulkan does not support the Tesla M60 GPU (https://developer.nvidia.com/vulkan-driver), and when I try running with opengl I get "failed to create context" because the M60 also does not support opengl 3 or 4.
Is it possible run AirSim on a VM given that the gpu in the instructions is not supported by vulkan/opengl or is there something I am doing wrong?

goran-j, 2021-01-16:
We have the same problem! Anyone managed to run AirSim (Unreal Engine Editor) on Azure GPU machine?

guadacasuso, 2021-01-24:
Unreal using Vulkan rendering on a Linux VM through SSH only works in offscreen rendering mode (same thing for running inside a docker container). 

To enable offscreen rendering, the binary executable needs to be launched with the -renderoffscreen switch option. 

The start-airsim.ps1 script in the AirSim azure/ folder does include this switch option, but this script and the main documentation is for deploying a Windows VM. 

Also the Vulkan libraries need to be installed by running:  sudo apt install libvulkan1

guadacasuso, 2021-01-24:
Summarizing, to run on a Linux VM (doesn't have a display attached), you need:
        Nvidia graphics driver VM extension
	Vulkan libraries installed by sudo apt install libvulkan1
	Launch binary with offscreen rendering like ./Blocks.sh -renderoffscreen 

Let me know If that works.

goran-j, 2021-01-24:
thanks @guadacasuso 
What you described works, but we wanted to use the Unreal editor as well on Azure GPU machine with Linux (with desktop). We managed to make it work by changing the driver to opengl in Unreal config file, but for some reason it was extremely slow and actually unusable (over remote desktop). At the end we gave up because we found a way how to cross-compile/package it for Linux from Windows machine

guadacasuso, 2021-01-25:
Yes, enabling the deprecated OpenGL rendering will not work for long and has been fully disabled in UE 4.26. 
Unfortunately, Vulkan can’t forward its rendering over Remote Desktop or ssh as far as I know so Linux is limited to offscreen rendering for now.

closed 2021-01-25 by guadacasuso