microsoft/AirSim #3315: Problems with the latest release (v1.4.0) Linux assets

rajat2004, 2021-01-18:
## Bug report 

- AirSim Version/#commit: Release [`v1.4.0`](https://github.com/microsoft/AirSim/releases/tag/v1.4.0-linux) 
- UE/Unity version: NA
- autopilot version: NA
- OS Version: Ubuntu 18.04

### What's the issue you encountered?

I was trying out the latest release and had various problems, so this issue in order to know if these are isolated errors or others are also having these. Will keep updating the issue as I test more.

There seems to be some common errors from missing assets -

```
[2021.01.18-11.09.37:879][  0]LogStreaming: Error: Couldn't find file for package /Engine/EditorMaterials/AssetViewer/EpicQuadPanorama_CC+EV1 requested by async loading code. NameToLoad: /Engine/EditorMaterials/AssetViewer/EpicQuadPanorama_CC+EV1
[2021.01.18-11.09.37:879][  0]LogStreaming: Error: Found 1 dependent packages...
[2021.01.18-11.09.37:879][  0]LogStreaming: Error:   /AirSim/VehicleAdv/SUV/AutomotiveMaterials/Materials/Glass/M_Glass
```

 - Blocks - `Blocks.sh` to run the binary is missing, also the packaged Blocks env seems to be not present, only the Editor stuff is present.

- AbandonedPark - Car or Multirotor vehicle gets created, however the Car doesn't seem to move at all when trying using the arrow keys. Subwindows are empty except for Scene, and the car disappears when running `hello_car.py` script and later crashes with a seg fault. Stacktrace -

```
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
[2021.01.18-11.11.07:798][209]LogCore: === Critical error: ===
Unhandled Exception: SIGSEGV: unaligned memory access (SIMD vectors?)

[2021.01.18-11.11.07:798][209]LogCore: Fatal error!

0x0000000005f7044e AbandonedPark!AActor::SetActorLocationAndRotation(FVector, FRotator, bool, FHitResult*, ETeleportType)()
0x0000000002c89d2d AbandonedPark!PawnSimApi::resetImplementation() [/datadrive2/Marketplace/AbandonedPark/Plugins/AirSim/Source/PawnSimApi.cpp:296]
0x0000000002cae769 AbandonedPark!CarPawnSimApi::resetImplementation() [/datadrive2/Marketplace/AbandonedPark/Plugins/AirSim/Source/Vehicles/Car/CarPawnSimApi.cpp:167]
0x0000000002c89de7 AbandonedPark!msr::airlib::UpdatableObject::reset() [/datadrive2/Marketplace/AbandonedPark/Plugins/AirSim/Source/AirLib/include/common/UpdatableObject.hpp:43]
0x0000000002d30e11 AbandonedPark!UE4Function_Private::TFunctionRefCaller<ASimModeBase::reset()::$_19, void ()>::Call(void*) [/datadrive/ue_425/ue-4.25.1-linux-debugeditor/Engine/Source/Runtime/Core/Public/Templates/Function.h:548]
0x0000000002c7572c AbandonedPark!UAirBlueprintLib::RunCommandOnGameThread(TFunction<void ()>, bool, TStatId) [/datadrive2/Marketplace/AbandonedPark/Plugins/AirSim/Source/AirBlueprintLib.cpp:337]
0x0000000002c9f349 AbandonedPark!ASimModeBase::reset() [/datadrive2/Marketplace/AbandonedPark/Plugins/AirSim/Source/SimMode/SimModeBase.cpp:391]
0x0000000002cbb681 AbandonedPark!TGraphTask<FFunctionGraphTask>::ExecuteTask(TArray<FBaseGraphTask*, TSizedDefaultAllocator<32> >&, ENamedThreads::Type) [/datadrive/ue_425/ue-4.25.1-linux-debugeditor/Engine/Source/Runtime/Core/Public/Async/TaskGraphInterfaces.h:849]
0x00000000036b8342 AbandonedPark!FNamedTaskThread::ProcessTasksNamedThread(int, bool)()
0x00000000036b75c3 AbandonedPark!FNamedTaskThread::ProcessTasksUntilQuit(int)()
0x00000000036b580f AbandonedPark!FTaskGraphImplementation::WaitUntilTasksComplete(TArray<TRefCountPtr<FGraphEvent>, TInlineAllocator<4u, TSizedDefaultAllocator<32> > > const&, ENamedThreads::Type)()
0x0000000006b87039 AbandonedPark!FTickTaskSequencer::ReleaseTickGroup(ETickingGroup, bool)()
0x0000000006b8274b AbandonedPark!FTickTaskManager::RunTickGroup(ETickingGroup, bool)()
0x000000000656f982 AbandonedPark!UWorld::Tick(ELevelTick, float)()
0x00000000063e8110 AbandonedPark!UGameEngine::Tick(float, bool)()
0x0000000002c4e821 AbandonedPark!FEngineLoop::Tick()()
0x0000000002c555da AbandonedPark!GuardedMain(char16_t const*)()
0x00000000074ea80b AbandonedPark!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)())()
0x00007f92ccc95bf7 libc.so.6!__libc_start_main(+0xe6)
0x0000000002c49029 AbandonedPark!_start()
```

- MSBuild - Directly crashes during startup, in all simmodes. Error -
```
[2021.01.18-11.18.42:766][  0]LogActor: Warning: BP_CameraDirector_C /Game/Maps/Soccer_Field.Soccer_Field:PersistentLevel.CameraDirector has natively added scene component(s), but none of them were set as the actor's RootComponent - picking one arbitrarily
[2021.01.18-11.18.42:767][  0]LogStreaming: Error: Couldn't find file for package /AirSim/VehicleAdv/SUV/SuvCarPawn requested by async loading code. NameToLoad: /AirSim/VehicleAdv/SUV/SuvCarPawn
[2021.01.18-11.18.42:767][  0]LogStreaming: Error: Found 0 dependent packages...
terminating with uncaught exception of type std::invalid_argument: Failed to load asset class - Class'/AirSim/VehicleAdv/SUV/SuvCarPawn.SuvCarPawn_C'
Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=6
[2021.01.18-11.18.42:794][  0]LogCore: === Critical error: ===
Unhandled Exception: SIGABRT: abort() called

[2021.01.18-11.18.42:794][  0]LogCore: Fatal error!

0x00007f30f1749fb7 libc.so.6!gsignal(+0xc7)
0x00007f30f174b921 libc.so.6!abort(+0x140)
[... 36 lines cut ...]
[2021.01.18-11.22.19:271][  0]LogStreaming: Error: Couldn't find file for package /AirSim/Blueprints/BP_ComputerVisionPawn requested by async loading code. NameToLoad: /AirSim/Blueprints/BP_ComputerVisionPawn
[2021.01.18-11.22.19:271][  0]LogStreaming: Error: Found 0 dependent packages...
terminating with uncaught exception of type std::invalid_argument: Failed to load asset class - Class'/AirSim/Blueprints/BP_ComputerVisionPawn.BP_ComputerVisionPawn_C'
Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
...
```
Stack trace similar to MSBuild therefore omitted

LandscapeMountains seems to working, but haven't tested thoroughly.
Also haven't yet used on Windows

### Settings
Default settings, sometimes only to change SimMode

### How can the issue be reproduced?

1. Download the specific release asset from [here](https://github.com/microsoft/AirSim/releases/tag/v1.4.0-linux) 
2. Unzip and run

### Include full error message in text form

Included above

**What's better than filing an issue? Filing a pull request :).**

rambodrahmani, 2021-01-23:
I am having the same exact problem on ```Debian 4.19.160-2```.
```bash
[2021.01.23-13.17.13:720][  0]LogStreaming: Error: Couldn't find file for package /AirSim/VehicleAdv/SUV/SuvCarPawn requested by async loading code. NameToLoad: /AirSim/VehicleAdv/SUV/SuvCarPawn
[2021.01.23-13.17.13:720][  0]LogStreaming: Error: Found 0 dependent packages...
terminating with uncaught exception of type std::invalid_argument
Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=6
[2021.01.23-13.17.13:769][  0]LogCore: === Critical error: ===
Unhandled Exception: SIGABRT: abort() called

[2021.01.23-13.17.13:769][  0]LogCore: Fatal error!

[...]

Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.
Aborted (core dumped)
```

zimmy87, 2021-02-01:
Hi @rajat2004 and @rambodrahmani, I have uploaded new binaries for both windows that should address most of the issues you've been experiencing:

- Errors related to missing assets should all be fixed now. This was happening because the Unreal Build Tool was skipping assets that weren't explicitly referenced. A level referencing all vehicle pawns was included in the build to fix this.
- Blocks: I am not able to repro this issue personally. I tried unzipping the binary with 7zip, Thunar's archive manager, and Windows Explorer's extract options; with each extraction I saw both the Blocks.sh script and the binary in the archive. I strongly suspect this was happening due to a corrupted download.
- AbandonedPark: this environment was crashing due to a bug in Unreal's landscape rendering. I replaced the offending landscape with a static mesh and the environment is now running for me.
- MSBuild: this environment has been updated with the fix for the missing assets error.
- Africa: I checked an older release of this environment and the prompt was present, so this isn't anything introduced in this release. Because the prompt disappears after pressing the ok button, I don't think this a major issue that blocks any functionality. 

The Building_99, TrapCamera, and Zhangjiajie environments were also updated due to related issues that are now fixed. Please test again with the latest version of these binaries and let me know if you see any errors.

rajat2004, 2021-02-02:
@zimmy87 Thanks a lot for looking into this issue and updating the binaries! I haven't tested all the envs, just tried Blocks and few others and they seem to be working correctly.

- Blocks, AbandonedPark, MSBuild tested to working and not crashing
- Regarding UE4 skipping assets, I believe it might even fix the problem of https://github.com/microsoft/AirSim/pull/3009, after the game thread fix is applied (currently using the Weather API crashes due to not running in Game thread). Though personally, I believe its best to fix the problem of Weather assets by properly loading them such that they get referenced, that's becoming a much bigger task than thought, this can be good way currently to fix the problem
- About the missing Blocks files, could be a corrupted download, but it unzipped correctly and the size was also correct, so no idea.
- Had a strange behaviour in the MSBuild env, the trees and buildings doesn't seem to have collisions. The depth rendering works on the outside, but the interior is missing. Is this an intended behaviour?
[image]
[image]

zimmy87, 2021-02-09:
Hi @rajat2004, the Africa and MSBuild2018 environments have been updated with the following changes:

- the default map used in Africa was changed to one that doesn't have the "Select Day:" prompt and there's a wider variety of animals that spawn closer to the player start as well
- the collisions were fixed in MSBuild2018 so that vehicles now collide with buildings and trees

Since these were updates to the environments and not the AirSim code, these binaries have been updated in place and are available at the same location for v1.4.0

zimmy87, 2021-02-18:
The issues being reported look have been resolved by the re-uploaded binaries so closing this.

closed 2021-02-18 by zimmy87