microsoft/AirSim #3316: Point cloud generation from depth images

tharindurmt, 2021-01-19:
## Question
I am working on adding a lidar sensor using the `DepthPerspective` images as per my proposal described in [Depth camera-based lidar](https://github.com/microsoft/AirSim/issues/3272). It's straightforward to obtain a `DepthPerspective` and convert it to the corresponding point cloud (please refer the following script). 
However, the obtained point cloud exhibit radial distortions and no matter what I tried, I can't get rid of that. I was wondering that I'm doing wrong here and how can I rectify this.

Related issues: [1198](https://github.com/microsoft/AirSim/issues/1198), [745](https://github.com/microsoft/AirSim/issues/745)
it's reported that `depth_image_proc` ROS package accurately reconstruct a point cloud [2980](https://github.com/microsoft/AirSim/issues/2980) the depth image - [lines 313 to 339](http://docs.ros.org/en/melodic/api/depth_image_proc/html/point__cloud__xyzrgb_8cpp_source.html). However, I can't seems to find any difference between its implementation and the one that I'm using.
Furthermore, I think the code given in [point_cloud.py](https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/point_cloud.py) is incorrect as `cv2.reprojectImageTo3D` accept a disparity map [(documentation)](https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html#ga1bc1152bd57d63bc524204f21fde6e02), not a `DepthPerspective` image.

Code: 

```
import math
import airsim
import numpy as np

Width = 32
Height = 18
CameraFOV = 90
Fx = Fy = Width / (2 * math.tan(CameraFOV * math.pi / 360))
Cx = Width / 2
Cy = Height / 2
Colour = (0, 255, 0)
RGB = "%d %d %d" % Colour # Colour for points

def generatepointcloud(depth):
    rows, cols = depth.shape
    c, r = np.meshgrid(np.arange(cols), np.arange(rows), sparse=True)
    valid = (depth > 0) & (depth < 255)
    z = 1000 * np.where(valid, depth / 256.0, np.nan)
    x = np.where(valid, z * (c - Cx) / Fx, 0)
    y = np.where(valid, z * (r - Cy) / Fy, 0)
    return np.dstack((x, y, z))

def savepointcloud(image, filename):
    f = open(filename, "w+")
    for x in range(image.shape[0]):
        for y in range(image.shape[1]):
            pt = image[x, y]
            if math.isinf(pt[0]) or math.isnan(pt[0]) or pt[0] > 10000 or pt[1] > 10000 or pt[2] > 10000:
                None
            else:
                f.write("%f %f %f %s\n" % (pt[0], pt[1], pt[2] - 1, RGB))
    f.close()

client = airsim.CarClient()
client.confirmConnection()

responses = client.simGetImages(
    [airsim.ImageRequest('front_left', airsim.ImageType.DepthPerspective, pixels_as_float=True, compress=False)])
response = responses[0]
img1d = np.array(response.image_data_float, dtype=np.float)
img1d[img1d > 255] = 255
img2d = np.reshape(img1d, (responses[0].height, responses[0].width))
pcl = generatepointcloud(img2d)
savepointcloud(pcl, 'pcl.asc')
```

Airsim settings:
```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "ClockSpeed": 1,
  "SimMode": "Car",
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 2,
        "Width": 32,
        "Height": 18,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 100,
        "MotionBlurAmount": 0
      }
    ]
  },
  "Cameras": {
    "front_left": {
      "CaptureSettings": [
        {
          "ImageType": 2,
          "Width": 32,
          "Height": 18,
          "FOV_Degrees": 90
        }
      ]
    }
  }
}
```

Point cloud as viewed through cloudcompare.
[image]

Note that the vehicle was right in front of a cube in the blocks environment. 

Can anyone please shed some light here? 
@rajat2004, @sytelus, @jonyMarino, @saihv 
Thank you.

GimpelZhang, 2021-01-19:
Hi @tharindurmt , as I mentioned in #2980, I could get the point cloud from the depth images recorded through the ROS wrapper. However, the point cloud also has radial distortions: 

[image]

tharindurmt, 2021-01-19:
@GimpelZhang  Thanks for confirming that it doesn't work with ROS as well. So, this could be an issue with the `DepthPerspective` images then.
Were you able to rectify this issue?

GimpelZhang, 2021-01-19:
> @GimpelZhang Thanks for confirming that it doesn't work with ROS as well. So, this could be an issue with the `DepthPerspective` images then.
> Were you able to rectify this issue?

No, I didn't notice this problem until you raised this issue.

ahmed-elsaharti, 2021-01-19:
Hi @tharindurmt, my first guess here would be that your script might not be taking the lens' radial distortion into account (forgive me if that is not the case, I quickly skimmed through the code), that could be the reason behind this.

tharindurmt, 2021-01-19:
@ahmed-elsaharti Yes, the radial distortion coefficients are not used here. However, I'm under the assumption that the standard PIPCamera does not exhibit any radial distortions. I will double check on this though. Thanks for the suggestion.

ahmed-elsaharti, 2021-01-19:
@tharindurmt Since the camera allows you to set the FOV, it would be safe to assume that radial distortion would occur.
Also, I did a quick couple of tests using your scripts. On setting FOV to 20 degrees, the distortion disappears from the point cloud. 
[image]
as opposed to the 90 degrees FOV:
[image]

I would suggest using openCVs `undistort()` function prior to processing the image. That should theoretically fix the issue.

tharindurmt, 2021-01-19:
@ahmed-elsaharti Thank you very much for conducting that experiment and feel silly for not thinking about it myself. I'll definitely try it out tomorrow and see how it goes.

tharindurmt, 2021-01-20:
@ahmed-elsaharti I ran a test with 20 degrees FoV and still seems to be some radial distortions (please see attached screenshot).

[image]

Can you please confirm whether you used `DepthPerspective` or `DepthPlanner` image type? When I use `DepthPlanner`, I can get the following result at 20 degrees FoV. 

[image]

However, to get to my original goal (that is to approximate the lidar sensor using a depth camera which is described in [3272](https://github.com/microsoft/AirSim/issues/3272)), it seems that I will have to obtain a high resolution depth image and then sample from it to create the velodyne point pattern.  The following is when I set the FoV to 90 degrees and use `DepthPlanner`.

[image]

What do you think of this proposal?

ahmed-elsaharti, 2021-01-20:
Hi @tharindurmt, I used your code as it is (as far as I remember) so it was `DepthPerspective`. 
Not entirely sure about the usage of a planar view here. To get as close to a realistic system as possible I would stick with a perspective camera but would calibrate the camera and undistort the image.
This is me assuming that you're simulating a real-life scenario. Especially since your method of converting a depth image to a point cloud is pretty valid for real life use. Which is why I would recommend keeping this as 'non-ideal' as possible. But this mainly depends on your use case.

tharindurmt, 2021-01-20:
@ahmed-elsaharti Thanks for confirming that.
I agree that the perspective image should be used for better realism (which is why I was using it from the beginning). I conducted some further experiments and I think there are two strategies to model the lidar using `DepthPerspective` images (as opposed to my original proposal).

- Obtain multiple high resolution `DepthPerspective` images with very narrow FoVs (something like 5-10 degrees) and then sample from them to create the corresponding velodyne pattern. So, at each sensor call, multiple such images will be obtained to cover the lidar sweep needed.
- Build a custom projection matrix at each sensor call and obtain a single high resolution `DepthPerspective` image to cover the entire lidar seep needed and then sample from it to create the corresponding velodyne pattern. 

Will try both and report back.

virtualRooom, 2021-04-13:
Hi @tharindurmt , it seems like the depth image obtained by camera is the distance from the scene to the camera center, you need to convert it to the distance to image plane. Just  use DepthConversion before you `generatepointcloud` function. The original answer and more discussion can be found in [this link](https://github.com/unrealcv/unrealcv/issues/14#issuecomment-307028752).
```Python
def depthConversion(PointDepth, f):
    H = PointDepth.shape[0]
    W = PointDepth.shape[1]
    i_c = float(H) / 2 - 1
    j_c = float(W) / 2 - 1
    columns, rows = np.meshgrid(np.linspace(0, W-1, num=W), np.linspace(0, H-1, num=H))
    DistanceFromCenter = ((rows - i_c)**2 + (columns - j_c)**2)**(0.5)
    PlaneDepth = PointDepth / (1 + (DistanceFromCenter / f)**2)**(0.5)
    return PlaneDepth

...

responses = client.simGetImages(
    [airsim.ImageRequest('front_left', airsim.ImageType.DepthPerspective, pixels_as_float=True, compress=False)])
response = responses[0]
img1d = np.array(response.image_data_float, dtype=np.float)
img1d[img1d > 255] = 255
img2d = np.reshape(img1d, (responses[0].height, responses[0].width))
img2d_converted = DepthConversion(img2d, Fx)
pcl = generatepointcloud(img2d)
savepointcloud(pcl, 'pcl.asc')
```
Scene
[image]

Before conversion
[image]

After conversion
[image]

tharindurmt, 2021-05-11:
@virtualRooom  Thaks for this and sorry, I just saw this. 
I will have a look and see how this goes.
Since this didn't work out, we divereged from this a bit.

tharindurmt, 2021-05-11:
@virtualRooom  I just tested this and the artifacts similar to what I have described initially. Please see attached. There is a clear radial distortion which is similar to what I have explained above. 

Code:

```import math
import airsim
import numpy as np

Width = 32
Height = 18
CameraFOV = 90
Fx = Fy = Width / (2 * math.tan(CameraFOV * math.pi / 360))
Cx = Width / 2
Cy = Height / 2
Colour = (0, 255, 0)
RGB = "%d %d %d" % Colour # Colour for points

def depthConversion(PointDepth, f):
    H = PointDepth.shape[0]
    W = PointDepth.shape[1]
    i_c = float(H) / 2 - 1
    j_c = float(W) / 2 - 1
    columns, rows = np.meshgrid(np.linspace(0, W-1, num=W), np.linspace(0, H-1, num=H))
    DistanceFromCenter = ((rows - i_c)**2 + (columns - j_c)**2)**(0.5)
    PlaneDepth = PointDepth / (1 + (DistanceFromCenter / f)**2)**(0.5)
    return PlaneDepth

def generatepointcloud(depth):
    rows, cols = depth.shape
    c, r = np.meshgrid(np.arange(cols), np.arange(rows), sparse=True)
    valid = (depth > 0) & (depth < 255)
    z = 1000 * np.where(valid, depth / 256.0, np.nan)
    x = np.where(valid, z * (c - Cx) / Fx, 0)
    y = np.where(valid, z * (r - Cy) / Fy, 0)
    return np.dstack((x, y, z))

def savepointcloud(image, filename):
    f = open(filename, "w+")
    for x in range(image.shape[0]):
        for y in range(image.shape[1]):
            pt = image[x, y]
            if math.isinf(pt[0]) or math.isnan(pt[0]) or pt[0] > 10000 or pt[1] > 10000 or pt[2] > 10000:
                None
            else:
                f.write("%f %f %f %s\n" % (pt[0], pt[1], pt[2] - 1, RGB))
    f.close()

client = airsim.CarClient()
client.confirmConnection()

responses = client.simGetImages(
    [airsim.ImageRequest('front_left', airsim.ImageType.DepthPerspective, pixels_as_float=True, compress=False)])
response = responses[0]
img1d = np.array(response.image_data_float, dtype=np.float)
img1d[img1d > 255] = 255
img2d = np.reshape(img1d, (responses[0].height, responses[0].width))
img2d_converted = depthConversion(img2d, Fx)
pcl = generatepointcloud(img2d)
savepointcloud(pcl, 'pcl.asc')```
```
Settings file:
```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "ClockSpeed": 1,
  "SimMode": "Car",
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 2,
        "Width": 32,
        "Height": 18,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 100,
        "MotionBlurAmount": 0
      }
    ]
  },
  "Cameras": {
    "front_left": {
      "CaptureSettings": [
        {
          "ImageType": 2,
          "Width": 32,
          "Height": 18,
          "FOV_Degrees": 90
        }
      ]
    }
  }
}
```

The view that was used to generate the pointcloud and the resultant poincloud are as bellow. 
[image]
[image]

AmitNativ1984, 2021-08-10:
For those still struggling with this issue..
ros depth_image_proc has a nodelet: **depth_image_proc/point_cloud_xyz_radial**. using it instead of depth_image_proc/point_cloud_xyz fixed the problem for me

closed 2021-08-11 by jonyMarino