microsoft/AirSim #3323: Fragmented Camera Scene Image

icisneros, 2021-01-20:
## Bug report 

- AirSim Version/#commit: 46a9b3a53811175e198e3608316fa4c983f7c53c
- UE/Unity version: 4.25.4
- autopilot version: n/a
- OS Version: MacOS Catalina 10.15.7

### What's the issue you encountered?

The "scene" camera (ImageType 0 in the settings.json) provides a broken and fragmented image. This happens both in the simulator and in the png images that are written when recording.  I have tested with SimMode of "Multirotor" and "ComputerVision", and in both "FPV" modes and "Fly with Me" camera modes. However, the fragmented image bug does not affect the depth or segmentation views (ImageType 3 and 5 respectively).  I am using the default settings.json, and working in the provided Blocks environment.  But this bug also happens if I create new cameras that are also outputting ImageType 0 (an example settings.json is provided below).  I always start off the project from the xcode workspace, because I am working on a mac.

[image]
[image]

### Settings

{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 1,
    "Cameras": [
        { "CameraName": "leftcamera", "ImageType": 0, "PixelsAsFloat": false, "Compress": true },
		{ "CameraName": "rightcamera", "ImageType": 0, "PixelsAsFloat": false, "Compress": true },
		{ "CameraName": "frontcamera", "ImageType": 0, "PixelsAsFloat": false, "Compress": true },
		{ "CameraName": "backcamera", "ImageType": 0, "PixelsAsFloat": false, "Compress": true },
		{ "CameraName": "upcamera", "ImageType": 0, "PixelsAsFloat": false, "Compress": true },
		{ "CameraName": "downcamera", "ImageType": 0, "PixelsAsFloat": false, "Compress": true }
    ]
  },
  "Vehicles": {
    "360Drone": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "AutoCreate": true,
      "PawnPath": "",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      },
     "Cameras": {
        "leftcamera": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 256,
              "Height": 144,
              "FOV_Degrees": 100,
              "TargetGamma": 1.5
            }
          ],
          "X": 0, "Y": 0, "Z": -0.40,
          "Pitch": 0, "Roll": 0, "Yaw": -90
        },
        "rightcamera": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 256,
              "Height": 144,
              "FOV_Degrees": 100,
              "TargetGamma": 1.5
            }
          ],
          "X": 0, "Y": 0, "Z": -0.40,
          "Pitch": 0, "Roll": 0, "Yaw": 90
        },
		"frontcamera": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 256,
              "Height": 144,
              "FOV_Degrees": 100,
              "TargetGamma": 1.5
            }
          ],
          "X": 0, "Y": 0, "Z": -0.40,
          "Pitch": 0, "Roll": 0, "Yaw": 0
        },
		"backcamera": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 256,
              "Height": 144,
              "FOV_Degrees": 100,
              "TargetGamma": 1.5
            }
          ],
          "X": 0, "Y": 0, "Z": -0.40,
          "Pitch": 0, "Roll": 0, "Yaw": 180
        },
		"upcamera": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 256,
              "Height": 144,
              "FOV_Degrees": 100,
              "TargetGamma": 1.5
            }
          ],
          "X": 0, "Y": 0, "Z": 0,
          "Pitch": 90, "Roll": 0, "Yaw": 0
        },
		"downcamera": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 256,
              "Height": 144,
              "FOV_Degrees": 100,
              "TargetGamma": 1.5
            }
          ],
          "X": 0, "Y": 0, "Z": 0,
          "Pitch": -90, "Roll": 0, "Yaw": 0
        }
      }
    }
  },
  "SubWindows": [
    {"WindowID": 0, "CameraName": "leftcamera", "ImageType": 0, "Visible": false},
    {"WindowID": 1, "CameraName": "frontcamera", "ImageType": 0, "Visible": false},
    {"WindowID": 2, "CameraName": "upcamera", "ImageType": 0, "Visible": false}
  ]
}

rajat2004, 2021-01-20:
Hmm, pretty strange, I'm not seeing such behaviour on Linux at least, using UE 4.25.3, used the settings you provided but no such problem. Someone should probably test on Windows & OSX also if possible but definitely strange. 
Maybe try rebuilding AirSim from scratch in a different directory, You could also try clearing the intermediate folders in the Blocks directory, there's a script `clean.sh` inside `Unreal/Environments/Blocks` directory, which will delete those. A further step would be `clean_rebuild.sh` script in the root directory of the repo.

ahmed-elsaharti, 2021-01-20:
Tested this on Windows with a new-ish build (not the latest) and a 1.4.0 binary and both worked fine:
[image]

Could be a MacOS specific bug?

icisneros, 2021-01-21:
I have followed rajat2004 suggestions and tried `clean.sh` in the Blocks directory, as well as running `clean_rebuild.sh` in the repo root, as well as rebuilding in a different directory.  But the issue persists. I am running with default settings, on the Blocks environment immediately after building.  I have also tried creating a new Unreal project, and copying in the `Unreal\Plugins` folder as detailed at the bottom of the page [here](https://github.com/microsoft/AirSim/blob/master/docs/unreal_custenv.md) (as well as modifying the `.uproject`), but still the problem persists in that environment.

Would it be possible for someone to send me a zip of their macOS-built `Unreal\Plugins` folder to see if replacing this in my machine would solve the issue?