microsoft/AirSim #3327: Implementing A3C Reinforcement Learning with Airsim

Deepakgthomas, 2021-01-22:
## Question
### 

### Include context on what you are trying to achieve

#### Context details

### Include details of what you already did to find answers

ahmed-elsaharti, 2021-01-22:
@Deepakgthomas Is each instance a compiled/cooked binary or is all this running through the UE editor?
I just ran a couple of tests and it seems (to my surprise) like running the same environment through the UE editor consumes less CPU power.

Deepakgthomas, 2021-01-22:
I am not sure what a cooked binary is (I am a newbie to this stuff). I have a Unreal environment that I have integrated with the Airsim API. In order to check how well it works, I opened multiple instances of the Environment (by launching the .uproject file).
Does that answer your question?

ahmed-elsaharti, 2021-01-22:
No worries. A cooked binary is essentially a compiled 'game' executable. Something like the ones you can download from the [releases section](https://github.com/microsoft/AirSim/releases). But yes, what you just said does answer my question, it seems like you're running the environment within the editor. I'm not sure if moving to unity would improve things drastically (haven't tried Unity yet tbh). Try setting the viewmode in [settings.json](https://microsoft.github.io/AirSim/settings/) to 'NoDisplay' and see if it makes any difference.

Deepakgthomas, 2021-01-22:
Wow, there definitely seems to be an improvement. But I distinctly another Github post saying that running 'headless' will disable the collision avoidance capabilities. Is that true?

Deepakgthomas, 2021-01-22:
Here is a comment from the this [post](https://github.com/microsoft/AirSim/issues/563) - "I think you don't want render but you still want all objects in environment and collision detection. Unfortunately Unreal doesn't allow turning off rendering. It might be possible (its open source engine) but I don't know... But in any case, if you turn off rendering then you won't have ability to get screenshots or depth view etc." Do you know if this issue has been resolved?

ahmed-elsaharti, 2021-01-22:
Hmm..I'm not sure if that's still the case.
I just tried retrieving an RGB image with NoDisplay mode and it worked. Try testing the collision feature using your script to see if that works too.

Deepakgthomas, 2021-01-22:
I will do that! Thank you very much.

closed 2021-01-22 by Deepakgthomas