microsoft/AirSim #3331: 'roslaunch airsim_ros_pkgs airsim_node.launch;' is stuck at 'process[ned_to_enu_pub-3]: started with pid [5052]'

LuciusLuF, 2021-01-22:
## Question
### 'roslaunch airsim_ros_pkgs airsim_node.launch;' is stuck at 'process[ned_to_enu_pub-3]: started with pid [5052]'

### I'm following the tutorial: https://microsoft.github.io/AirSim/airsim_ros_pkgs/. Everything went on well until 'roslaunch airsim_ros_pkgs airsim_node.launch;' .

#### The program got stuck at 'process[ned_to_enu_pub-3]: started with pid [5052]'. I have tried using 'catkin clean' and run it again several times but the same issue happened every time.  
Details are provided below:

Laptop:~/AirSim/ros$ roslaunch airsim_ros_pkgs airsim_node.launch;
... logging to /home/lufe/.ros/log/70e59aba-5d00-11eb-856c-94e6f7b226e5/roslaunch-lufe-Laptop-5026.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lufe-Laptop:42707/

SUMMARY
========

PARAMETERS
 * /airsim_node/host_ip: localhost
 * /airsim_node/is_vulkan: False
 * /airsim_node/publish_clock: False
 * /airsim_node/update_airsim_control_every_n_sec: 0.01
 * /airsim_node/update_airsim_img_response_every_n_sec: 0.05
 * /airsim_node/update_lidar_every_n_sec: 0.01
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    airsim_node (airsim_ros_pkgs/airsim_node)
    ned_to_enu_pub (tf/static_transform_publisher)

auto-starting new master
process[master]: started with pid [5037]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 70e59aba-5d00-11eb-856c-94e6f7b226e5
process[rosout-1]: started with pid [5048]
started core service [/rosout]
process[airsim_node-2]: started with pid [5051]
process[ned_to_enu_pub-3]: started with pid [5052]

### The last time it ran over a night. Stuck at the same step. I would appreciate any help!

ahmed-elsaharti, 2021-01-23:
Are you running this on WSL? or is this being done purely on Linux?

LuciusLuF, 2021-01-23:
On Ubuntu 18.04.

ahmed-elsaharti, 2021-01-23:
And did you run the Airsim environment before launching the node?

LuciusLuF, 2021-01-23:
Yeah… I followed the tutorial and used setup, build, catkin build, source. There was no error. When it first got stuck, I used catkin clean. But it didn't work.

ahmed-elsaharti, 2021-01-23:
Hmm..Thats odd. The reason I'm asking though is that it is expected to get stuck on that step only if its not able to communicate with the instance or Airsim that is running. Otherwise, it should say
```
Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
Waiting for connection -
Connected!
```

LuciusLuF, 2021-01-23:
It's weird… I didn't see anything like this

ahmed-elsaharti, 2021-01-23:
Try running `roslaunch airsim_ros_pkgs airsim_node.launch output:=screen` that should display the connection status.
Also check if the topics are being created and are being published. It seems like it being stuck on `process[ned_to_enu_pub-3]: started with pid [5052]` is normal.

LuciusLuF, 2021-01-23:
Got it. I will try it. Thank you very much!

LuciusLuF, 2021-01-23:
Hi ahmed-elsaharti,
Good afternoon! I just ran your code and the details are listed below:

Laptop:~/AirSim/ros$ roslaunch airsim_ros_pkgs airsim_node.launch output:=screen
... logging to /home/lufe/.ros/log/007c522e-5db3-11eb-aa29-94e6f7b226e5/roslaunch-lufe-Laptop-10169.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lufe-Laptop:43743/

SUMMARY
========

PARAMETERS
 * /airsim_node/host_ip: localhost
 * /airsim_node/is_vulkan: False
 * /airsim_node/publish_clock: False
 * /airsim_node/update_airsim_control_every_n_sec: 0.01
 * /airsim_node/update_airsim_img_response_every_n_sec: 0.05
 * /airsim_node/update_lidar_every_n_sec: 0.01
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    airsim_node (airsim_ros_pkgs/airsim_node)
    ned_to_enu_pub (tf/static_transform_publisher)

auto-starting new master
process[master]: started with pid [10179]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 007c522e-5db3-11eb-aa29-94e6f7b226e5
process[rosout-1]: started with pid [10190]
started core service [/rosout]
process[airsim_node-2]: started with pid [10197]
process[ned_to_enu_pub-3]: started with pid [10198]
simmode_name: 
[ INFO] [1611430896.039163315]: Setting ROS wrapper to DRONE mode
Waiting for connection - XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

ahmed-elsaharti, 2021-01-23:
If your AirSim simulation is running in the background then this tells us that the ROS node/wrapper is unable to connecto to the simulation through the APIs. Are you able to connect to the simulation through the APIs in general through Python/C++?

LuciusLuF, 2021-01-23:
Thank you for your response! I have to say I'm a beginner on Ubuntu and AirSim. Could you give me some instructions on how to realize that?

ahmed-elsaharti, 2021-01-23:
Could you attach a screenshot of what the AirSim simulation shows when you see `Waiting for connection - XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX`?
Also, to test API connectivity: 
Assuming you have python installed, run `pip install msgpack-rpc-python` and then `pip install airsim`. Then run the hello_drone.py script under the PythonClient/multirotor directory

LuciusLuF, 2021-01-24:
Hi ahmed-elsaharti,
Good morning! 
This is the screenshot of my hello_drone.py. It works well. 
https://ibb.co/0s41W5P

This is the screenshot when I input roslaunch airsim_ros_pkgs airsim_node.launch. It's still stuck. 
https://ibb.co/vcp4zPT
https://ibb.co/WGsX35Q

LuciusLuF, 2021-01-24:
And when I tried roslaunch airsim_ros_pkgs airsim_node.launch output:=screen while AirSim simulation was active, the code is different.

Laptop:~/AirSim/ros$ roslaunch airsim_ros_pkgs airsim_node.launch output:=screen
... logging to /home/lufe/.ros/log/a9248448-5e15-11eb-ac08-94e6f7b226e5/roslaunch-lufe-Laptop-22658.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lufe-Laptop:34423/

SUMMARY
========

PARAMETERS
 * /airsim_node/host_ip: localhost
 * /airsim_node/is_vulkan: False
 * /airsim_node/publish_clock: False
 * /airsim_node/update_airsim_control_every_n_sec: 0.01
 * /airsim_node/update_airsim_img_response_every_n_sec: 0.05
 * /airsim_node/update_lidar_every_n_sec: 0.01
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    airsim_node (airsim_ros_pkgs/airsim_node)
    ned_to_enu_pub (tf/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

process[airsim_node-1]: started with pid [22673]
process[ned_to_enu_pub-2]: started with pid [22674]
simmode_name: 
[ INFO] [1611473634.821928913]: Setting ROS wrapper to DRONE mode
[ WARN] [1611473634.843543485]: Shutdown request received.
[ WARN] [1611473634.846018751]: Reason given for shutdown: [[/ned_to_enu_pub] Reason: new node registered with same name]
Waiting for connection - 
Connected!

Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
Waiting for connection - 
Connected!

Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
Waiting for connection - 
Connected!

Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
Exception raised by the API, something went wrong.
rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle API for 'ComputerVision' is not available. This could either because this is simulation-only API or this vehicle does not exist.
AirsimROSWrapper Initialized!
error
Exception raised by the API:
rpclib: function 'simGetGroundTruthEnvironment' (called with 1 arg(s)) threw an exception. The exception contained this information: Vehicle Sim-API for 'ComputerVision' is not available. This could either because this is not a simulation or this vehicle does not exist.
error
Exception raised by the API:
rpclib: function 'simGetGroundTruthEnvironment' (called with 1 arg(s)) threw an exception. The exception contained this information: Vehicle Sim-API for 'ComputerVision' is not available. This could either because this is not a simulation or this vehicle does not exist.
error
Exception raised by the API:
rpclib: function 'simGetGroundTruthEnvironment' (called with 1 arg(s)) threw an exception. The exception contained this information: Vehicle Sim-API for 'ComputerVision' is not available. This could either because this is not a simulation or this vehicle does not exist.
error
Exception raised by the API:
rpclib: function 'simGetGroundTruthEnvironment' (called with 1 arg(s)) threw an exception. The exception contained this information: Vehicle Sim-API for 'ComputerVision' is not available. This could either because this is not a simulation or this vehicle does not exist.
error

LuciusLuF, 2021-01-24:
And after I got the code listed below, the output of  `roslaunch airsim_ros_pkgs airsim_node.launch` has changed:

Laptop:~/AirSim/ros$ roslaunch airsim_ros_pkgs airsim_node.launch;
... logging to /home/lufe/.ros/log/a9248448-5e15-11eb-ac08-94e6f7b226e5/roslaunch-lufe-Laptop-22026.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lufe-Laptop:40419/

SUMMARY
========

PARAMETERS
 * /airsim_node/host_ip: localhost
 * /airsim_node/is_vulkan: False
 * /airsim_node/publish_clock: False
 * /airsim_node/update_airsim_control_every_n_sec: 0.01
 * /airsim_node/update_airsim_img_response_every_n_sec: 0.05
 * /airsim_node/update_lidar_every_n_sec: 0.01
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    airsim_node (airsim_ros_pkgs/airsim_node)
    ned_to_enu_pub (tf/static_transform_publisher)

auto-starting new master
process[master]: started with pid [22036]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a9248448-5e15-11eb-ac08-94e6f7b226e5
process[rosout-1]: started with pid [22047]
started core service [/rosout]
process[airsim_node-2]: started with pid [22050]
process[ned_to_enu_pub-3]: started with pid [22051]
[ WARN] [1611473634.841140625]: Shutdown request received.
[ WARN] [1611473634.842695608]: Shutdown request received.
[ WARN] [1611473634.842739332]: Reason given for shutdown: [[/airsim_node] Reason: new node registered with same name]
[ WARN] [1611473634.844854605]: Reason given for shutdown: [[/ned_to_enu_pub] Reason: new node registered with same name]
[ned_to_enu_pub-3] process has finished cleanly
log file: /home/lufe/.ros/log/a9248448-5e15-11eb-ac08-94e6f7b226e5/ned_to_enu_pub-3*.log
[airsim_node-2] process has finished cleanly
log file: /home/lufe/.ros/log/a9248448-5e15-11eb-ac08-94e6f7b226e5/airsim_node-2*.log

ahmed-elsaharti, 2021-01-24:
Hi @LuciusLuF, 
Firstly, 
```
Shutdown request received.
Reason given for shutdown: [new node registered with same name]
```
means that the node is already running in the background. So you might want to close everything down before starting again.
Also, 
```
Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
Waiting for connection -
Connected!
```
is a good sign, it means that the API is able to connect to the simulation. In general the node shouldn't work if the simulation is not active since it tries to connect to AirSim via the API.

LuciusLuF, 2021-01-24:
You are right! The reason might be I ran two terminals at the same time. I just restarted Ubuntu and ran `roslaunch airsim_ros_pkgs airsim_node.launch output:=screen`. Then I got:

 lufe@lufe-Laptop:~/AirSim/ros$ roslaunch airsim_ros_pkgs airsim_node.launch output:=screen
... logging to /home/lufe/.ros/log/13887eec-5e67-11eb-b2c5-94e6f7b226e5/roslaunch-lufe-Laptop-5414.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lufe-Laptop:37069/

SUMMARY
========

PARAMETERS
 * /airsim_node/host_ip: localhost
 * /airsim_node/is_vulkan: False
 * /airsim_node/publish_clock: False
 * /airsim_node/update_airsim_control_every_n_sec: 0.01
 * /airsim_node/update_airsim_img_response_every_n_sec: 0.05
 * /airsim_node/update_lidar_every_n_sec: 0.01
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    airsim_node (airsim_ros_pkgs/airsim_node)
    ned_to_enu_pub (tf/static_transform_publisher)

auto-starting new master
process[master]: started with pid [5425]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 13887eec-5e67-11eb-b2c5-94e6f7b226e5
process[rosout-1]: started with pid [5436]
started core service [/rosout]
process[airsim_node-2]: started with pid [5439]
process[ned_to_enu_pub-3]: started with pid [5440]
simmode_name: 
[ INFO] [1611508241.610019155]: Setting ROS wrapper to DRONE mode
Waiting for connection - 
Connected!

Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
Waiting for connection - 
Connected!

Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
Waiting for connection - 
Connected!

Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
Exception raised by the API, something went wrong.
rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle API for 'ComputerVision' is not available. This could either because this is simulation-only API or this vehicle does not exist.
AirsimROSWrapper Initialized!
error
Exception raised by the API:
rpclib: function 'simGetGroundTruthEnvironment' (called with 1 arg(s)) threw an exception. The exception contained this information: Vehicle Sim-API for 'ComputerVision' is not available. This could either because this is not a simulation or this vehicle does not exist.
error
Exception raised by the API:
rpclib: function 'simGetGroundTruthEnvironment' (called with 1 arg(s)) threw an exception. The exception contained this information: Vehicle Sim-API for 'ComputerVision' is not available. This could either because this is not a simulation or this vehicle does not exist.
error

ahmed-elsaharti, 2021-01-24:
The 
```
rpclib: function 'simGetGroundTruthEnvironment' (called with 1 arg(s)) threw an exception. The exception contained this information: Vehicle Sim-API for 'ComputerVision' is not available. This could either because this is not a simulation or this vehicle does not exist.
```
error suggests that the ROS node is grabbing a settings.json file that has the `SimMode` set to `ComputerVision`.
This is odd since your simulation (from the screenshot) seems to be running in Multirotor mode.
If I remember correctly, in Linux, both the simulation and the ROS node grab the same settings.json file (unlike WSL where each of them looks in a different directory).

LuciusLuF, 2021-01-24:
Thank you for your information! If I chose to run in the vehicle mode, the same issue would arise:
https://ibb.co/1qp5jjc

ahmed-elsaharti, 2021-01-24:
Can you attach the settings.json file under the documents directory? Its very odd that the node is reading a different mode than the one specified for the simulation

LuciusLuF, 2021-01-24:
I'm not sure. Which directory is it in?

ahmed-elsaharti, 2021-01-24:
After a bit of testing, I think your issue is with the settings.json file.
You need to specify a drone and a couple of sensors (rather than use the default blank settings.json file) for that error not to show up.
Try:
```json
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings_json.md",
  "SettingsVersion": 1.2,
  "ClockSpeed": 1,
  "SimMode": "Multirotor",
  "Vehicles": {
  "Drone1": {
    "VehicleType": "simpleflight",
    "AutoCreate": true,
    "X": 0, "Y": 0, "Z": 0,
    "Sensors": {
      "LidarSensor2": { 
         "SensorType": 6,
        "Enabled" : true,
        "NumberOfChannels": 1,
        "X": 0, "Y": 0, "Z": -1,
        "DrawDebugPoints": true,
        "DataFrame": "SensorLocalFrame"
      }
    }
  }
  }
}
```
or maybe one of the settings in the tutorials:
https://microsoft.github.io/AirSim/airsim_tutorial_pkgs/

This file ([settings.json](https://microsoft.github.io/AirSim/settings/)) would be under `~/Documents/AirSim` on Linux systems

LuciusLuF, 2021-01-24:
OK... it only has:
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2
}
in the setting.json.

I will copy your code in it.

LuciusLuF, 2021-01-24:
For `roslaunch airsim_ros_pkgs airsim_node.launch;` it's still stuck:
https://ibb.co/dmHDLQZ

but for `roslaunch airsim_ros_pkgs airsim_node.launch output:=screen` , it shows:
https://ibb.co/pv6wcyQ

Does that mean it's OK to use?

ahmed-elsaharti, 2021-01-24:
Yup its ok to use it now. Try to echo any of the topics and you'll see that its being published

LuciusLuF, 2021-01-24:
Great! Thank you so much for your time and help!

ahmed-elsaharti, 2021-01-24:
Glad to be of help ^_^

closed 2021-03-01 by jonyMarino