microsoft/AirSim #3338: LIDAR doesn't detect detailed information  in the CityEnviron

dongfeng12, 2021-01-25:
Hello,
I'm trying to get lidar points in CityEnviron 1.2.2. However, using getLidarData() only return the coarse frame of the car models.
[image]

This problem is similar to issue #2948 , but the two ways mentioned cannot be implemented to the CityEnviron. I wonder can I change the collision standard to complex or get detailed LiDAR points in CityEnviron.

Ryton, 2021-04-20:
This answer probably a bit late, but (as far as I know):
The Airsim lidar raytracing acts/collides on the player-collision world, not on the visibility collision world (which would be much more realistic, but much more computational).

You CAN view the used collision models in the editor through the 'view' button
[image]

Also, you can set the meshes to use "complex collision model as simple", for the objects you want to query in more detail, by editing the static mesh (or a used copy).

I hope this can help you?

jonyMarino, 2021-07-06:
This is now a dup of #3855