microsoft/AirSim #3339: AbandonedPark Environment does not run on Windows 10

adamdai, 2021-01-25:
## Bug report 

- AirSim Version/#commit: latest (9415f8b7546f6a02ffaebe16d77a2ea0c636bcbe)
- UE/Unity version: N/A
- autopilot version: N/A
- OS Version: Windows 10

### What's the issue you encountered?

I downloaded the newly released AbandonedPark environment binary and ran the Abandonedpark.exe executable and received an Assertion failed error:

[image]

### Settings

```
{
	"SettingsVersion": 1.2,
	"SimMode": "Car",

	"CameraDefaults": {
		"CaptureSettings": [
		  {
			"ImageType": 0,
			"Width": 1920,
			"Height": 1080,
			"FOV_Degrees": 90,
			"AutoExposureSpeed": 75,
			"AutoExposureBias": 0,
			"AutoExposureMaxBrightness": 0.64,
			"AutoExposureMinBrightness": 0.03,
			"MotionBlurAmount": 0,
			"TargetGamma": 0.5,
			"ProjectionMode": "",
			"OrthoWidth": 5.12
		  }
		]
	},
	
	"Vehicles": {
		"Car1": {
		  "VehicleType": "PhysXCar",
		  "X": 4, "Y": 0, "Z": -2
		},
		"Car2": {
		  "VehicleType": "PhysXCar",
		  "X": -4, "Y": 0, "Z": -2
		}
    }
}
```

### How can the issue be reproduced?

1. Download the AbandonedPark environment from https://github.com/microsoft/AirSim/releases/tag/v1.4.0-windows, unzip, and run the executable.

### Include full error message in text form
See screenshot above.

**What's better than filing an issue? Filing a pull request :).**

EdwinMeriaux, 2021-01-29:
@adamdai I just downloaded the file and it works?

When you say you run the exe file are you running the unreal exe file?
If it still does not work maybe redownload and make sure the file is extracted properly.

zimmy87, 2021-02-01:
Hi @adamdai, there was an issue with the binary encountering a bug in Unreal's landscape rendering. I replaced the landscape causing the bug with a static mesh and was able to work around the bug that way. I uploaded a fixed binary around 4:00-5:00pm on 1/28, which looks like around the same time you commented @EdwinMeriaux, so I think you were one of the first ones to see the fixed binary. Please continue testing and let me know if you see any other issues.

closed 2021-02-05 by zimmy87

rafaellcampos, 2021-03-23:
Hi. I also have the same issue. I tried to run the AbandonedPark.exe file and I got this error:

[image]