microsoft/AirSim #3348: How to display texts in airsim

Han-Sin, 2021-01-27:
Hi guys, I want to display some texts on the screen of airsim. 
And I know there is a python api simPrintLogMessage() can do this. But this API can not change the text size and position. I want to decide the texts' position, color and size by myself. So is there any way to do this? 
Please, thanks so much

rajat2004, 2021-01-27:
Have you tried [`simPlotStrings`](https://microsoft.github.io/AirSim/api_docs/html/#airsim.client.VehicleClient.simPlotStrings), it has the options which you require. Also see [plot_markers.py](https://github.com/microsoft/AirSim/blob/master/PythonClient/environment/plot_markers.py) for some examples of all the Plotting APIs

Han-Sin, 2021-01-30:
Thanks so much~~
really help me~

rajat2004, 2021-01-30:
If the problem has been fixed, best to close the issue :)