microsoft/AirSim #3349: OptionsMenu.uasset not correctly removed in release 1.4.0?

CarloCloet, 2021-01-29:
## Bug report 

- AirSim Version/#commit:1.4.0
- UE/Unity version: 4.25.4
- autopilot version: N/A
- OS Version: Windows 10

### What's the issue you encountered?

When running the Blocks environment, I notice the following warning in the output log window:
`LogLinker: Warning: Failed to load '/AirSim/Weather/UI/OptionsMenu': Can't find file.`

The application works fine though, including the weather menu (F10).

I noticed the `OptionsMenu.uasset` file was removed from `AirSim\Unreal\Plugins\AirSim\Content\Weather\UI` between release 1.3.1 and 1.4.0. (see https://github.com/microsoft/AirSim/pull/2877)

Adding the file back manually clears the warnings, but I assume that is not the proper way to fix this. Do others also experience this problem in 1.4.0?

The issue is caused by  the `MenuActor.uasset` file:

`LogLinker: Warning: While loading '<path omitted>/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/UI/MenuActor.uasset' failed to load '/AirSim/Weather/UI/OptionsMenu': Can't find file. `

When opening the MenuActor asset and checking its references, it indeed lists OptionsMenu as a referenced asset, but I do not understand Unreal sufficiently at this point to debug any further.

### Settings

{
  "SettingsVersion": 1.2,
  "SimMode": "Car"
}

zimmy87, 2021-02-11:
Hi @CarloCloet, thank you for reporting this issue. It looks like this happening due to a removed reference inside Blueprint'/AirSim/Weather/UI/MenuActor.MenuActor'. These warnings go away after I open this asset, select 'File -> Refresh All nodes', and recompile. Please retest after refreshing the MenuActor asset and let me know if you still see these warnings.

zimmy87, 2021-03-02:
Hi @CarloCloet, I created a PR #3432 to address this issue. Please retest with this PR applied and let me see if you still experience this issue.

CarloCloet, 2021-03-04:
@zimmy87 , this solves the issue indeed. Tx for the fix!

(Apologies for the late response, had to send in my PC for repair and was not able to test earlier.)

PR #3432 (2021-03-23): refresh nodes in MenuActor.uasset

Fixes: #3349

## About
This removes an outdated reference to '/AirSim/Weather/UI/OptionsMenu' which has been removed

## How Has This Been Tested?
The Blocks environment was run with and without this fix and it was confirmed that the warning about failure to load '/AirSim/Weather/UI/OptionsMenu' was removed

## Screenshots (if appropriate):
closed 2021-03-23 by jonyMarino