microsoft/AirSim #3352: Custom and normal drone

EdwinMeriaux, 2021-01-31:
Hi!

So I know how to get multiple of the same drones flying. But I was curious if there was anyway I could get both a custom and a default drone flying or even 2 custom drones of different type?

Thanks in advance

rajat2004, 2021-02-01:
I believe the `PawnPath` setting is there for per-vehicle as well, that should work

EdwinMeriaux, 2021-02-01:
I've been trying to put multiple different drones in the pawnpaths.
But when I spawn the drones how do I specify which drone is which path?

rajat2004, 2021-02-01:
Something like below I guess -
```json
"Vehicles": {
  "Drone1": {
    ...
    "PawnPath": "Path1"
  },
  "Drone2": {
    ...
    "PawnPath": "Path2"
  }
}
```

EdwinMeriaux, 2021-02-02:
Perfect Thanks!

closed 2021-02-02 by EdwinMeriaux