microsoft/AirSim #3361: Error in ./build.sh

wq1148927746, 2021-02-02:
## Bug report 

- AirSim Version/#commit: git clone https://github.com/Microsoft/AirSim.git
- UE/Unity version: 4.25.4
- autopilot version: N/A
- OS Version: N/A

### What's the issue you encountered?

I do like [this](https://microsoft.github.io/AirSim/build_linux/), everything is well before I run the command **./build.sh**.

************************************
AirSim setup completed successfully!
************************************

+ debug=false
+ [[ 0 -gt 0 ]]
+ '[' '!' -d ./external/rpclib/rpclib-2.2.1 ']'
+ '[' -d ./cmake_build ']'
++ which cmake
+ CMAKE=/usr/bin/cmake
+ false
+ build_dir=build_release
++ uname
+ '[' Linux == Darwin ']'
+ export CC=clang-8
+ CC=clang-8
+ export CXX=clang++-8
+ CXX=clang++-8
+ [[ -d ./AirLib/deps/eigen3/Eigen ]]
+ echo 'putting build in build_release folder, to clean, just delete the directory...'
putting build in build_release folder, to clean, just delete the directory...
+ [[ -f ./cmake/CMakeCache.txt ]]
+ [[ -d ./cmake/CMakeFiles ]]
+ folder_name=
+ [[ ! -d build_release ]]
+ mkdir -p build_release
+ pushd build_release
+ false
+ folder_name=Release
+ /usr/bin/cmake ../cmake -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is Clang 8.0.0
-- The CXX compiler identification is Clang 8.0.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/clang-8 - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - failed
-- Check for working CXX compiler: /usr/bin/clang++-8
-- Check for working CXX compiler: /usr/bin/clang++-8 - broken
CMake Error at /usr/share/cmake-3.19/Modules/CMakeTestCXXCompiler.cmake:59 (message):
  The C++ compiler

    "/usr/bin/clang++-8"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/wqf/AirSim/build_release/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make cmTC_8b979/fast && /usr/bin/make  -f CMakeFiles/cmTC_8b979.dir/build.make CMakeFiles/cmTC_8b979.dir/build
    make[1]: 进入目录“/home/wqf/AirSim/build_release/CMakeFiles/CMakeTmp”
    Building CXX object CMakeFiles/cmTC_8b979.dir/testCXXCompiler.cxx.o
    /usr/bin/clang++-8    -o CMakeFiles/cmTC_8b979.dir/testCXXCompiler.cxx.o -c /home/wqf/AirSim/build_release/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
    Linking CXX executable cmTC_8b979
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_8b979.dir/link.txt --verbose=1
    /usr/bin/clang++-8 CMakeFiles/cmTC_8b979.dir/testCXXCompiler.cxx.o -o cmTC_8b979 
    /usr/bin/ld: 找不到 -lstdc++
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    CMakeFiles/cmTC_8b979.dir/build.make:105: recipe for target 'cmTC_8b979' failed
    make[1]: *** [cmTC_8b979] Error 1
    make[1]: 离开目录“/home/wqf/AirSim/build_release/CMakeFiles/CMakeTmp”
    Makefile:140: recipe for target 'cmTC_8b979/fast' failed
    make: *** [cmTC_8b979/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "/home/wqf/AirSim/build_release/CMakeFiles/CMakeOutput.log".
See also "/home/wqf/AirSim/build_release/CMakeFiles/CMakeError.log".
+ popd
~/AirSim ~/AirSim
+ rm -r build_release
+ exit 1

### Settings
 

### How can the issue be reproduced?

1.
2.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

wq1148927746, 2021-02-02:
I think it's the cmake error

but I don't know how to solve this.

Thans!

rajat2004, 2021-02-02:
Seems to be a linker error from the line - `usr/bin/ld: 找不到 -lstdc++`
Try installing `libstdc++-8-dev` package and run the script again

wq1148927746, 2021-02-02:
> Seems to be a linker error from the line - `usr/bin/ld: 找不到 -lstdc++`
> Try installing `libstdc++-8-dev` package and run the script again

I have done this,but it still error.

+ debug=false
+ [[ 0 -gt 0 ]]
+ '[' '!' -d ./external/rpclib/rpclib-2.2.1 ']'
+ '[' -d ./cmake_build ']'
++ which cmake
+ CMAKE=/usr/bin/cmake
+ false
+ build_dir=build_release
++ uname
+ '[' Linux == Darwin ']'
+ export CC=clang-8
+ CC=clang-8
+ export CXX=clang++-8
+ CXX=clang++-8
+ [[ -d ./AirLib/deps/eigen3/Eigen ]]
+ echo 'putting build in build_release folder, to clean, just delete the directory...'
putting build in build_release folder, to clean, just delete the directory...
+ [[ -f ./cmake/CMakeCache.txt ]]
+ [[ -d ./cmake/CMakeFiles ]]
+ folder_name=
+ [[ ! -d build_release ]]
+ mkdir -p build_release
+ pushd build_release
+ false
+ folder_name=Release
+ /usr/bin/cmake ../cmake -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is Clang 8.0.0
-- The CXX compiler identification is Clang 8.0.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/clang-8 - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - failed
-- Check for working CXX compiler: /usr/bin/clang++-8
-- Check for working CXX compiler: /usr/bin/clang++-8 - broken
CMake Error at /usr/share/cmake-3.19/Modules/CMakeTestCXXCompiler.cmake:59 (message):
  The C++ compiler

    "/usr/bin/clang++-8"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/wqf/AirSim/build_release/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make cmTC_17ef3/fast && /usr/bin/make  -f CMakeFiles/cmTC_17ef3.dir/build.make CMakeFiles/cmTC_17ef3.dir/build
    make[1]: 进入目录“/home/wqf/AirSim/build_release/CMakeFiles/CMakeTmp”
    Building CXX object CMakeFiles/cmTC_17ef3.dir/testCXXCompiler.cxx.o
    /usr/bin/clang++-8    -o CMakeFiles/cmTC_17ef3.dir/testCXXCompiler.cxx.o -c /home/wqf/AirSim/build_release/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
    Linking CXX executable cmTC_17ef3
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_17ef3.dir/link.txt --verbose=1
    /usr/bin/clang++-8 CMakeFiles/cmTC_17ef3.dir/testCXXCompiler.cxx.o -o cmTC_17ef3 
    /usr/bin/ld: 找不到 -lstdc++
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    CMakeFiles/cmTC_17ef3.dir/build.make:105: recipe for target 'cmTC_17ef3' failed
    make[1]: *** [cmTC_17ef3] Error 1
    make[1]: 离开目录“/home/wqf/AirSim/build_release/CMakeFiles/CMakeTmp”
    Makefile:140: recipe for target 'cmTC_17ef3/fast' failed
    make: *** [cmTC_17ef3/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "/home/wqf/AirSim/build_release/CMakeFiles/CMakeOutput.log".
See also "/home/wqf/AirSim/build_release/CMakeFiles/CMakeError.log".
+ popd
~/AirSim ~/AirSim
+ rm -r build_release
+ exit 1

rajat2004, 2021-02-02:
Checking my packages, since its related to clang, you might be missing `libc++-8-dev` instead of `libstdc++-8-dev` which is for GCC, try installing that. Otherwise check you packages and see if anything important seems to be missing. (The `-7` files are for GCC 7)

```
$ dpkg -l | grep c++           
ii  lib32stdc++-8-dev                                                8.4.0-1ubuntu1~18.04                             amd64        GNU Standard C++ Library v3 (development files)
ii  lib32stdc++6                                                     8.4.0-1ubuntu1~18.04                             amd64        GNU Standard C++ Library v3 (32 bit Version)
ii  libc++-8-dev:amd64                                               1:8-3~ubuntu18.04.2                              amd64        LLVM C++ Standard library (development files)
ii  libc++1-8:amd64                                                  1:8-3~ubuntu18.04.2                              amd64        LLVM C++ Standard library
ii  libc++abi-8-dev:amd64                                            1:8-3~ubuntu18.04.2                              amd64        LLVM low level support for a standard C++ library (development files)
ii  libc++abi1-8:amd64                                               1:8-3~ubuntu18.04.2                              amd64        LLVM low level support for a standard C++ library
ii  libnetcdf-c++4                                                   4.2-8                                            amd64        legacy NetCDF C++ interface
ii  libsigc++-2.0-0v5:amd64                                          2.10.0-2                                         amd64        type-safe Signal Framework for C++ - runtime
ii  libstdc++-7-dev:amd64                                            7.5.0-3ubuntu1~18.04                             amd64        GNU Standard C++ Library v3 (development files)
ii  libstdc++-8-dev:amd64                                            8.4.0-1ubuntu1~18.04                             amd64        GNU Standard C++ Library v3 (development files)
ii  libstdc++6:amd64                                                 8.4.0-1ubuntu1~18.04                             amd64        GNU Standard C++ Library v3
ii  libstdc++6:i386                                                  8.4.0-1ubuntu1~18.04                             i386         GNU Standard C++ Library v3
ii  libx32stdc++-8-dev                                               8.4.0-1ubuntu1~18.04                             amd64        GNU Standard C++ Library v3 (development files)
ii  libx32stdc++6                                                    8.4.0-1ubuntu1~18.04                             amd64        GNU Standard C++ Library v3 (x32)
ii  ros-melodic-xmlrpcpp                                             1.14.10-1bionic.20201017.071853                  amd64        XmlRpc++ is a C++ implementation of the XML-RPC protocol.

```

wq1148927746, 2021-02-03:
[image]

wq1148927746, 2021-02-03:
> Checking my packages, since its related to clang, you might be missing `libc++-8-dev` instead of `libstdc++-8-dev` which is for GCC, try installing that. Otherwise check you packages and see if [...]

I have done this, too.
[image]

I think the last package is about ROS, so I didn't install it.

wq1148927746, 2021-02-03:
However, it is still error.

[image]

wq1148927746, 2021-02-03:
I encounter another problem about GPU.

**When I try to run UE, There was a mistake, too.** 

[image]

**Maybe this is related to the settings of GPU?**

However, I can use GPU in PyTorch. And `nivdia-smi` is perfectly normal.

[image]

But, the bar of graph in setting displays as follows:

[image]

I am new to ubuntu and GPU.

Thank you!

wq1148927746, 2021-02-03:
I use a GPU of  **nvidia 3060TI**

nikhil-sethi, 2021-02-11:
What kernel version are you using @wq1148927746? There are known GPU issues with an AMD-Nvidia configuration. I have a similar configuration and a lot of the gpu issues were resolved after 5.6-5.7 kernel versions. Also, in my case the 'llvmpipe' was showing when I had Nvidia-optimus enabled in the BIOS. Try setting it to only the dedicated GPU option and see if it changes.

wq1148927746, 2021-02-22:
> What kernel version are you using @wq1148927746? There are known GPU issues with an AMD-Nvidia configuration. I have a similar configuration and a lot of the gpu issues were resolved after 5.6-5.7 [...]

Thanks! I solved this problem. But I still error in ./build.sh.

lovettchris, 2021-04-01:
This solved the problem for me:
```
sudo apt-get install libstdc++-8-dev
```
I see this in the setup.sh so I'm not sure what that didn't work, but running it manually fixed the build.

leimo2011, 2021-09-22:
same issue with mac os 11.2.2 

+ debug=false
+ gcc=false
+ [[ 0 -gt 0 ]]
+ RPC_VERSION_FOLDER=rpclib-2.3.0
+ '[' '!' -d ./external/rpclib/rpclib-2.3.0 ']'
+ '[' -d ./cmake_build ']'
++ which cmake
+ CMAKE=/usr/local/bin/cmake
+ false
+ build_dir=build_release
++ uname
+ '[' Darwin == Darwin ']'
+ export CC=/usr/local/opt/llvm@8/bin/clang
+ CC=/usr/local/opt/llvm@8/bin/clang
+ export CXX=/usr/local/opt/llvm@8/bin/clang++
+ CXX=/usr/local/opt/llvm@8/bin/clang++
+ [[ -d ./AirLib/deps/eigen3/Eigen ]]
+ echo 'putting build in build_release folder, to clean, just delete the directory...'
putting build in build_release folder, to clean, just delete the directory...
+ [[ -f ./cmake/CMakeCache.txt ]]
+ [[ -d ./cmake/CMakeFiles ]]
+ [[ ! -d build_release ]]
+ mkdir -p build_release
+ pushd build_release
+ false
+ folder_name=Release
+ /usr/local/bin/cmake ../cmake -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is Clang 8.0.1
-- The CXX compiler identification is Clang 8.0.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /usr/local/opt/llvm@8/bin/clang
-- Check for working C compiler: /usr/local/opt/llvm@8/bin/clang - broken
CMake Error at /usr/local/Cellar/cmake/3.21.3/share/cmake/Modules/CMakeTestCCompiler.cmake:69 (message):
  The C compiler

    "/usr/local/opt/llvm@8/bin/clang"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/xyz/Airsim/AirSim/build_release/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make -f Makefile cmTC_8502c/fast && /Applications/Xcode.app/Contents/Developer/usr/bin/make  -f CMakeFiles/cmTC_8502c.dir/build.make CMakeFiles/cmTC_8502c.dir/build
    Building C object CMakeFiles/cmTC_8502c.dir/testCCompiler.c.o
    /usr/local/opt/llvm@8/bin/clang   -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk -mmacosx-version-min=11.2 -MD -MT CMakeFiles/cmTC_8502c.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_8502c.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_8502c.dir/testCCompiler.c.o -c /Users/xyz/Airsim/AirSim/build_release/CMakeFiles/CMakeTmp/testCCompiler.c
    clang-8: error: invalid version number in '-mmacosx-version-min=11.2'
    make[1]: *** [CMakeFiles/cmTC_8502c.dir/testCCompiler.c.o] Error 1
    make: *** [cmTC_8502c/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/xyz/Airsim/AirSim/build_release/CMakeFiles/CMakeOutput.log".
See also "/Users/xyz/Airsim/AirSim/build_release/CMakeFiles/CMakeError.log".
+ popd
~/Airsim/airsim ~/Airsim/airsim
+ rm -r build_release
+ exit 1

derekwacks, 2021-11-17:
For anyone who comes across this thread with @leimo2011 's problem above:

(i.e. problem running ./build.sh for AirSim on MacOS Big Sur getting clang-8: error: invalid version number in '-mmacosx-version-min=11.2')

in `AirSim/build.sh` ~ line 57, 

Change

```
if [ "$(uname)" == "Darwin" ]; then 
    export CC=/usr/local/opt/llvm@8/bin/clang 
    export CXX=/usr/local/opt/llvm@8/bin/clang++ 
else
```

to 

```
if [ "$(uname)" == "Darwin" ]; then
    export CC=clang
    export CXX=clang++
else
```

(Get rid of the absolute paths)

zimmy87, 2022-01-14:
Are you still experiencing this issue @wq1148927746?

thejeshk, 2022-02-17:
Hi, I am facing similar issue

Please guide me to solve this issue: 

Ubuntu 20.04
UE 4.25 

if [ "$(uname)" == "Darwin" ]; then 
    export CC=/usr/local/opt/llvm@8/bin/clang 
    export CXX=/usr/local/opt/llvm@8/bin/clang++ 
else