microsoft/AirSim #3362: server could not find function 'simGetDistortionParams' with argument count 2

Chanchanhan, 2021-02-02:
## Question
### Why 'server could not find function 'simGetDistortionParams' with argument count 2' ?

When I try to use get distrot by [simGetDistortionParams](https://github.com/microsoft/AirSim/pull/3039) , it print error like this. And I can't find same question on web
Can any one help me? Thank you for your time!

Chanchanhan, 2021-02-02:
What a silly question!
You should use DepthPlanner rather than DepthPerspective!

ahmed-elsaharti, 2021-02-02:
I just tested this and did not receive an error.
Are you sure you're using an environment built with the latest commit?

Chanchanhan, 2021-02-05:
> I just tested this and did not receive an error.
> Are you sure you're using an environment built with the latest commit?

Thanks for your relpy!
And Yes, I try use this api with C++(1.40) and python(1.28), but both failed.
So maybe the error comes from a wrong way to call this api?
Can you show me an easy example ?

ahmed-elsaharti, 2021-02-05:
Hi @Chanchanhan ,
This is the code I used for testing along with the settings mentioned in the issue:
```python
import setup_path
import airsim
client = airsim.MultirotorClient()
distort_depth = client.simGetDistortionParams('depth_image', 'Rotor')
```
Btw, when you say python 1.28 are you referring to the version of the airsim library? ie: are you using the pip installed airsim module or are you importing the latest one in within the repo under PythonClient/airsim ?
Also, are you testing this on the blocks environment within the repo or is this being tested with a prebuilt environment (the ones found in the releases section).

Chanchanhan, 2021-02-09:
> Hi @Chanchanhan ,
> This is the code I used for testing along with the settings mentioned in the issue:
>
> ```python
> import setup_path
> import airsim
> client = airsim.MultirotorClient()
> distort_depth = [...]

It still fail with same log with your code.
The lastest version of airsim for python is 1.30(installed with pip), but it return 'AttributeError: 'MultirotorClient' object has no attribute 'simGetDistortionParams' when I call it.
So I use repo under PythonClient/airsim, whos version is 1.28.
I test with a prebuilt environment, is this the reason for this problems?

ahmed-elsaharti, 2021-02-09:
The airsim library you get through pip would probably not have the DistortionParams feature yet, so its better to use the library within PythonClient/airsim.
Which prebuilt environment are you using?

Chanchanhan, 2021-02-09:
> The airsim library you get through pip would probably not have the DistortionParams feature yet, so its better to use the library within PythonClient/airsim.
> Which prebuilt environment are you using?

 I use the library within PythonClient/airsim.

Chanchanhan, 2021-02-09:
> The airsim library you get through pip would probably not have the DistortionParams feature yet, so its better to use the library within PythonClient/airsim.
> Which prebuilt environment are you using?

 I use the library within PythonClient/airsim. And environment is 'Building_99'.

ahmed-elsaharti, 2021-02-09:
I just tested a couple (Blocks and City environment) of the precompiled binaries and it seems like the Distortion feature was not included in those either although the [changelog](https://microsoft.github.io/AirSim/CHANGELOG/) mentions that they were added before the release. It does, however, work with AbandonedPark which was [repackaged recently](https://github.com/microsoft/AirSim/issues/3339#issuecomment-770987790).
You could just use [the blocks environment](https://microsoft.github.io/AirSim/unreal_blocks/) within the repo (not the one in the releases) if you want to use this feature until the issue is fixed.

@jonyMarino @zimmy87 it seems like the binaries in the latest release are missing a couple of commits.

Chanchanhan, 2021-02-09:
> I just tested a couple (Blocks and City environment) of the precompiled binaries and it seems like the Distortion feature was not included in those either although the [...]

OK, thanks for your help!

zimmy87, 2021-02-11:
Hi @ahmed-elsaharti and @Chanchanhan, we believe the latest release was built with an incorrect target commit. We need to do some investigation to determine which environments are affected. Once we know exactly what happened and how to fix it, we will publish a new release.

zimmy87, 2021-02-18:
Hi @ahmed-elsaharti and @Chanchanhan, we investigated which environments were built with an incorrect target commit and found both the Blocks and ZhangJiajie were affected. These environments have been reuploaded as [Blocks.correct.zip](https://github.com/microsoft/AirSim/releases/download/v1.4.0-windows/Blocks.correct.zip) and [ZhangJiajie.correct.zip](https://github.com/microsoft/AirSim/releases/download/v1.4.0-windows/ZhangJiajie.correct.zip) to differentiate from their original uploads.

ahmed-elsaharti, 2021-02-18:
Hi @zimmy87 Blocks.correct works perfectly now. However, I'm unable to run ZhangJiajie. Not entirely sure if it's a problem with the binary or if it's something on my end though (probably the latter).

zimmy87, 2021-03-02:
Hi @ahmed-elsaharti, I am not able to repro your issue with ZhangJiajie on either Windows or Ubuntu. Can you attach one of the logs saved to ZhangJiajie\Saved\Logs after running the binary so I can take a look at the specific error you're running into?

ahmed-elsaharti, 2021-03-02:
@zimmy87, I just tried again and everything worked perfectly. Must have been a hiccup on my machine (sorry about that 😅). Given that the `simGetDistortionParams` works now on both binaries, I think the issue is resolved.

zimmy87, 2021-03-05:
Thanks for testing this again and confirming the behavior @ahmed-elsaharti! I'm closing this as it's now resolved.

closed 2021-03-05 by zimmy87