microsoft/AirSim #3368: Cannot change the orientation of the distance sensor

Alexwei92, 2021-02-04:
## Bug report 

- AirSim Version/#commit: 1.4.0
- UE/Unity version: UE 4.26
- autopilot version: N/A
- OS Version: Ubuntu 18.04

### What's the issue you encountered?

I want to use the distance sensor as a rangefinder to perform the terrain following tasks. However, the custom X,Y,Z and Yaw,Roll,Pitch in the settings.json do not work. The distance sensor is always set to the default (X=0,Y=0,Z=0) and (Roll=0, Pitch=0, Yaw=0) no matter which values I changed in the settings.json file.

### Settings

{
  "SettingsVersion": 1.4,
  "SimMode": "Multirotor",
  "Vehicles": {
      "drone0": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Disarmed",
      "AllowAPIAlways": true,
      "Sensors": {
        "Distance": {
          "SensorType": 5,
          "Enabled": true,
          "X": 0, "Y": 0, "Z": 0,
          "Yaw": 0, "Pitch": -90, "Roll": 0,
          "MinDistance": 0.2,
          "MaxDistance": 40.0,
          "DrawDebugPoints": true
        }
      }
    }
 }
}

rajat2004, 2021-02-08:
Working for me with the 1.4.0 release Blocks binary, using the settings you've put and the code -

```python
import setup_path 
import airsim
import time

client = airsim.MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)

client.armDisarm(True)

landed = client.getMultirotorState().landed_state
if landed == airsim.LandedState.Landed:
    print("taking off...")
    client.takeoffAsync()
else:
    print("already flying...")
    client.hoverAsync()

while True:
    data = client.getDistanceSensorData()
    print(data.distance)
    time.sleep(0.1)
```

Alexwei92, 2021-02-08:
Thanks! I mistakenly used the binaries compiled from an eariler UE version. It is working now in the 1.4.0 binaries with UE 4.26.

closed 2021-02-08 by Alexwei92