microsoft/AirSim #3387: Custom Environment simpleflightquadx Question

EdwinMeriaux, 2021-02-11:
Hello,

So I have created a custom environment for Airsim simulation in Windows. I wanted to modify the drone's mass in simpleflightquadx file. The file is empty (in the custom environment). The basic drone flies in the new environment the file to modify it's properties is just empty. 

Is there something I need to do to be able to have to files and modify those files in a custom environment?

ahmed-elsaharti, 2021-02-12:
Hi @EdwinMeriaux , what version of AirSim are you using here?
Also, what directory is the empty file under? is it under `\Plugins\AirSim\Source\AirLib\.....` ?

jonyMarino, 2021-02-16:
Hi @EdwinMeriaux! AirSim drones do not use unreal engine physics. You can change the mass of the drone between 0.8 and 1.6 from here:
https://github.com/microsoft/AirSim/blob/4be4d1eb0c7503197a19bbda0898ae082cc9f050/AirLib/include/vehicles/multirotor/MultiRotorParams.hpp#L226

closed 2021-04-08 by EdwinMeriaux