microsoft/AirSim #3389: HI Team,

chiragkhanna28, 2021-02-14:
## Question
### What's your question?

### Include context on what you are trying to achieve

#### Context details

### Include details of what you already did to find answers

chiragkhanna28, 2021-02-14:
Hi Team,
We are trying to use Microsoft AirSim in one of our projects. We would like to simulate the farm images using drone and derive the crop and soil quality of it by sending the images captured to custom vision.
Is it possible to achieve above use case using AirSim. If yes can you please help with the templates which i should use in Epic Games launcher, which can give me nice images for the farm.
Looking forward to hear from you.
Regards,
Chirag Khanna

jonyMarino, 2021-02-17:
Hi, @chiragkhanna28, and welcome! Yes, you can simulate all that with AirSim, you will need to manage to send the images to a server. Unfortunately, we don't have environment templates to provide, you will have to search on your own.
With regard to "And Is it possible to have different images which depicts bad areas to cultivate crop." yes, of course, you can model that inside unreal.

chiragkhanna28, 2021-02-21:
Hi @jonyMarino ,

Thanks for your response. We have created the custom project on which we have simulated farming environment as per our requirement.

We are struggling on how to move the drone over the rectangular area farm, so that we can capture the completed area over which drone is flying. We tried to use the hello_drone.py file inside Airsim master repository to simulate flying of drone using python script, but could not find how to get exact coordinates of the rectangular area and which method from the core api can be used once we get the coordinates. Can you please help us on the same.

Thanks in advance,
Chirag Khanna

jonyMarino, 2021-02-22:
Hi @chiragkhanna28!
With this script, you can fly (with an Xbox controller, for example) and collect the points you are interested in:
https://gist.github.com/zimmy87/fd2ff063823d532a68388ecee803ff69. Or take the from the Unreal editor and transform them to NED coordinates.

Then you can use moveToPositionAsync or moveOnPath to follow the collected points in a path.
And you can take images in the middle with another script or thread or between calls to moveToPositionAsync. 
I will ask about IOT edge.

chiragkhanna28, 2021-02-22:
Hi @jonyMarino,

Thanks for your response. We are trying to split the area over which drone is flying into smaller pieces. For that we need that when our drone is running in the simulator mode, then in the first time it will capture image of block size let say 10x10, then the second time it again capture image of block size 10*10 but this time the captured image should not get overlapped with the  first image, so that we have 2 mutually exclusive and continuous section of field. Similarly we keep on repeating this process and we can get mutually exclusive bunch of images of the area.
Please let us know how can we achieve this.
Thanks in advance,
Chirag Khanna