microsoft/AirSim #3392: SimGetImages() is VERY slow

TomAvrech, 2021-02-15:
## What feature are you suggesting?
### Overview:
Make SimGetImages() fast again !

### Smaller Details: 
When plotting many lines/points, while using low resolution, SimGetImages() is slow (2 Frames per minute(!)).

If I were to try to fix this my self, what would you think I should try ?

### Nature of Request:  

- Change
  - 

## Why would this feature be useful?
Less time

rajat2004, 2021-02-16:
That is very very slow. The problem's happening after lots of plot lines? You could try using [`simFlushPersistentMarkers`](https://microsoft.github.io/AirSim/api_docs/html/#airsim.client.VehicleClient.simFlushPersistentMarkers) to clear them if plotted with `is_persistent=True` if no longer required, but suspect that's not the case, and you want them to be present?

Also, are you using some cameras very infrequently by any chance, especially high-resolution ones?

TomAvrech, 2021-02-16:
Hey,

Yes a lot of lines, and 99%+ of them should be visible in any image, therefore can't flush \ it won't matter much \ just detecting which lines don't appear may cost more.

I'm using a low-resolution (200x200) camera, as still camera, and each capture takes 25-30 seconds.

rajat2004, 2021-02-16:
How many is a lot? Here's an interesting post which indicates that `DrawDebug*` isn't the most efficient way to do it, and using other method is faster - https://answers.unrealengine.com/questions/73788/whats-the-most-efficient-way-to-draw-a-bunch-of-li.html