microsoft/AirSim #3395: Default value of ApiServerPort (RpcLibPort)

asherikov, 2021-02-16:
## Bug report 

Airsim should not use 41451 port number by default, it is an ephemeral port on most modern systems -> https://en.wikipedia.org/wiki/Ephemeral_port, i.e., it may be taken by another networking application leading to random connection issues. Something between 5001 and 32767 that does not conflict with https://www.iana.org/assignments/service-names-port-numbers/service-names-port-numbers.txt should be ok.

ahmed-elsaharti, 2021-02-16:
Hi @asherikov, although it seems like the port is out of the ephemeral port ranges set by IANA and Windows systems, I'm guessing that if the port ends up being already occupied for a user they'll probably receive an error along the lines of:
[image]
and would be able to change `ApiServerPort` through settings.json (documented in #3196)

asherikov, 2021-02-16:
1. The issue can be solved by endusers of course, but why not change the default value to something that wouldn't fail randomly?
2. Connection between the error message and `ApiServerPort` is not obvious.

jonyMarino, 2021-02-22:
Hi @asherikov, and welcome! Thanks for catching this. I have searched where that port is being referenced:
https://github.com/microsoft/AirSim/search?q=41451
do you dare to submit a pull request for changing this?

asherikov, 2021-02-24:
Hello, sorry I dont have time atm.