microsoft/AirSim #3397: Installation Error on Windows 10 -- identifier 'FObjectProperty' is undefined in SimModeBase.cpp

denmonz, 2021-02-16:
## Bug report 

- AirSim Version/#commit: 40a2bbab9e5faebab74c044f240a68b32a16b055
- UE/Unity version: 4.24.3
- autopilot version: N/A
- OS Version: Windows 10
- Visual Studio Version: 2019

### What's the issue you encountered?

I successfully built AirSim using build.cmd, moved the AirSim plugin folder from Airsim/Unreal/Plugins to my modified Blocks.uproject environment, generated the project files, and attempted to build the solution with Development Editor > Win64 configuration. The main error I receive is, 'FObjectProperty': undeclared identifier. No documentation online on how to resolve.

[image]

### Settings

N/A

### How can the issue be reproduced?

1. Downloaded AirSim source (v1.4.0-windows) from https://github.com/Microsoft/AirSim/releases
2. Ran build.cmd with Developer Command Prompt through VS 2019.
3. Copied compiled Plugin folder to my .uproject directory.
4. Generated the files for the .uproject.
5. Opened the solution, set configuration to Development Editor > Win64.
6. Attempted to build project.

### Include full error message in text form
Severity	Code	Description	Project	File	Line	Suppression State
Error (active)	E0020	identifier "FObjectProperty" is undefined	Blocks	C:\Users\Zane - PC\Documents\GitHub\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\SimMode\SimModeBase.cpp	222	
Error (active)	E0020	identifier "sun_prop" is undefined	Blocks	C:\Users\Zane - PC\Documents\GitHub\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\SimMode\SimModeBase.cpp	222	
Error (active)	E0020	identifier "CastFieldChecked" is undefined	Blocks	C:\Users\Zane - PC\Documents\GitHub\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\SimMode\SimModeBase.cpp	222	
Error	C2065	'FObjectProperty': undeclared identifier	Blocks	C:\Users\Zane - PC\Documents\GitHub\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\SimMode\SimModeBase.cpp	222	
Error	C2065	'sun_prop': undeclared identifier	Blocks	C:\Users\Zane - PC\Documents\GitHub\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\SimMode\SimModeBase.cpp	222	
Error	C2065	'CastFieldChecked': undeclared identifier	Blocks	C:\Users\Zane - PC\Documents\GitHub\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\SimMode\SimModeBase.cpp	222	
Error	C2065	'FObjectProperty': undeclared identifier	Blocks	C:\Users\Zane - PC\Documents\GitHub\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\SimMode\SimModeBase.cpp	222	
Error	C2065	'sun_prop': undeclared identifier	Blocks	C:\Users\Zane - PC\Documents\GitHub\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\SimMode\SimModeBase.cpp	223	
Error	MSB3075	The command ""C:\Program Files\Epic Games\UE_4.24\Engine\Build\BatchFiles\Build.bat" BlocksEditor Win64 DebugGame -Project="C:\Users\Zane - PC\Documents\GitHub\AirSim\Unreal\Environments\Blocks\Blocks.uproject" -WaitMutex -FromMsBuild" exited with code 5. Please verify that you have sufficient rights to run this command.	Blocks	C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.MakeFile.Targets	46	

**What's better than filing an issue? Filing a pull request :).**

ahmed-elsaharti, 2021-02-16:
Hi @denmonz , this probably is because of the UE4 version you're using (4.24). As of recently, the version recommended for use with AirSim has become 4.25 so switching over to 4.25 should fix the errors. Also, you might want to set the build configuration to `DebugGame_Editor` as per the [guide](https://microsoft.github.io/AirSim/unreal_blocks/)

denmonz, 2021-02-16:
Thank you for your comment! I'll upgrade my UE4 from 4.24 to 4.25. I'm trying to port my Mac UE4 environment to a Windows environment, and what's weird is that 4.24 seems to work just fine on my Mac. I'll be sure to update this issue if I run into any more problems.

ahmed-elsaharti, 2021-02-17:
My guess would be that the environment was originally built on your mac using an old AirSim commit (ie: before the official migration to 4.25) but now its being rebuilt with the latest AirSim commit which requires 4.25. Nonetheless, if the issue persists please do update us 👍

nethtrick, 2021-02-17:
I am not currently able to upgrade to UE 4.25 so I found a hack for UE 4.24 which is to change the offending line to what was in a previous version of AirSim:

`UObjectProperty* sun_prop = Cast<UObjectProperty>(p);`

rajat2004, 2021-02-18:
Yup, this was changed in https://github.com/microsoft/AirSim/pull/3268. It was mentioned that it would break compilation on UE 4.24. If there are many people having this problem then a PR could be done to revert it, or maybe a UE4 version check in the code

denmonz, 2021-02-18:
@rajat2004 an AirSim version compatible with UE 4.24 would be preferable, as @nethtrick mentioned, he and I are unable to upgrade to UE 4.25.

rajat2004, 2021-02-18:
Can you test #3404 and see if fixes the problem

denmonz, 2021-02-22:
@rajat2004 works on my Mac environment! Cheers!

jonyMarino, 2021-02-22:
Hi @denmonz, and welcome!
Thanks for the report. And thanks, @rajat2004, for the pr!
@denmonz, Do you dare to submit a pull request for changing the documentation back to 4.24 again so I can merge Rajat's pr?

nethtrick, 2021-02-22:
Worked for me on Windows. Thank you!

This part maybe doesn't belong here - but I am also stuck with VS2017 still so I had to include the following line in build.cmd:

cmake -G"Visual Studio 15 2017 Win64" ..

Incidentally, I noticed the Win64 part was missing from the VS2019 version - but somehow still worked (on my other computer that has VS2019). It seems necessary for VS2017.

rajat2004, 2021-02-23:
Great, thanks everyone for testing! 
@jonyMarino Is changing the docs to again back to 4.24 necessary? I remember some asset changes which mostly won't get loaded on 4.24, and regressing the version might cause some problems. We could possibly leave 4.25 as the recommended version but just say that >=4.24 is supported

jonyMarino, 2021-02-23:
Hi @rajat2004! Yes, you're right. 4.25 should be recommended and >= 4.24 the supported ones.

rajat2004, 2021-02-23:
Okay, added the commit for doc change

PR #3404 (2021-02-24): Add UE4 version check for 4.24 compatibility

Fixes: #3397    

## About

Adds a UE4 version check to fix compilation with 4.24, update to #3268 after users had issues e.g. https://github.com/microsoft/AirSim/discussions/3388

## How Has This Been Tested?

Compilation on Linux with 4.25 and 4.24

## Screenshots (if appropriate):
closed 2021-02-24 by jonyMarino