microsoft/AirSim #3409: Exception Thrown in Python Plotting API

jusevitch, 2021-02-21:
## Bug report 

- AirSim Version/#commit: v1.4.0
- UE/Unity version: N/A
- autopilot version: N/A
- OS Version: Ubuntu 20.04

### What's the issue you encountered?

The [plotting API][API] in Python is throwing an exception when I try to call it.

### Settings

I'm using the default `settings.json` in the `AirSim/docker` folder.

### How can the issue be reproduced?

1. Run the default docker image for AirSim using the [commands in the documentation](https://microsoft.github.io/AirSim/docker_ubuntu/#running-an-unreal-binary-inside-a-docker-container). I ran this after freshly cloning the AirSim repo--no modifications.
2. Open Python3 and set up the AirSim client:
```
import airsim
client = airsim.MultirotorClient()
client.confirmconnection() # Output: Connected!
```
3. Plot a single `Vector3r` point:
```
client.simPlotPoints(airsim.Vector3r(1,0,0), is_persistent=True)
```

### Include full error message in text form

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/james/.local/lib/python3.8/site-packages/airsim/client.py", line 742, in simPlotPoints
    self.client.call(
  File "/home/james/.local/lib/python3.8/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/james/.local/lib/python3.8/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'simPlotPoints' (called with 5 arg(s)) threw an exception. The exception is not derived from std::exception. No further information available.
```

**What's better than filing an issue? Filing a pull request :).**

[API]: https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/client.py#L683

rajat2004, 2021-02-21:
It's likely due to the old Blocks release being downloaded in `download_env_binary.sh`. Instead, download the latest 1.4.0 binary.
There's a PR #2942 which updates the scripts, but need to update the PR as well for the new release

jusevitch, 2021-02-22:
Thanks for the tip. Unfortunately the 1.4.0 release for Linux throws the exact same error. I downloaded the binary [here](https://github.com/microsoft/AirSim/releases/tag/v1.4.0-linux) about 10 minutes ago and tried running the same command both with and without Docker.

I also tried running the same command with the v1.4.0 releases for  `LandscapeMountains` and `ZhangJiajie` environments (without Docker--just using the *.sh binary). The same error was thrown each time.

rajat2004, 2021-02-22:
I'm on mobile right now so can't run and check, but the docstring indicates that it's supposed to be a list of points, can you try that like `[Vecotr3r(...)]`

jusevitch, 2021-02-22:
Yikes, totally missed that detail. It works perfectly when I use `[Vector3r(...)]` inside of a list--tested on both the v1.4.0 Blocks and Zhang Jiajie environments. Thank you for your help!

closed 2021-02-22 by jusevitch