microsoft/AirSim #3415: Bug: PX4 SITL EKF-Failure/GPS-loss

ahmed-elsaharti, 2021-02-22:
(Similar to the post over at #3406 but a bug report is more appropriate at this point)
## Bug report 

- AirSim Version/#commit: 1.4 0096b5e
- UE/Unity version: 4.25
- autopilot version: PX4 1.10.1 SITL
- OS Version: Windows 10 x64 (Running SITL on Cygwin 0.8)

### What's the issue you encountered?
As seen in #3167, #3213, #3087 and many more, after launching AirSim in PX4 SITL mode (with lockstep mode enabled by default) the terminal displays the following errors continuously:
[image]
Although they do not affect the flight instantaneously, an EKF error shows up, QGC throws an 'no global position' error and the drone flies diagonally into the ground (and a line is actually traced on QGC showing that the GPS estimate just threw itself way off). As this happens `timeout on HilActuatorControlMessage, resetting lock step mode` appears on UE's side.
Here's a gif showing the problem in action:
[image]

A workaround that fixes this by disabling lockstep mode is discussed in #3406. **If you are facing a similar issue and need a temp fix have a look at that post**

Also, I'm curious as to whether this problem occurs on Linux systems too. If not then this might be caused by Cygwin and AirSim not communicating well?
Anyway, it would be great if someone could test this out on a Linux system.

### Settings
```json
{
  "SettingsVersion": 1.2,
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SimMode": "Multirotor",
  "DefaultVehicleConfig": "PX4",
  "Vehicles": {
    "PX4": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPort": 14570
        }
  },
  "OriginGeopoint": {
    "Latitude": 47.641468,
    "Longitude": -122.140165,
     "Altitude": 0
    }
}

```

### How can the issue be reproduced?
**Note** this error is inconsistent, ie: sometimes it happens as soon as the drone takes off and sometimes it takes a good 2-3 mins of flying before it happens.

To reproduce:

1. Follow [the docs](https://microsoft.github.io/AirSim/px4_sitl/) to set up and launch PX4 SITL using release 1.10.1
2. Launch QGroundControl
3. Launch AirSim with the settings shown above
4. Fly around for a bit using QGroundControl's 'go to location' (click somewhere on the map in QGC and it'll give u the option to go there) and/or upload a mission and run it.

### Include full error message in text form
[image]
and 
`timeout on HilActuatorControlMessage, resetting lock step mode`

**What's better than filing an issue? Filing a pull request :).**
Will probably file a PR with some edits and scripts to make disabling lockstep easier till we figure this out 😂

jonyMarino, 2021-03-04:
This may be a PX4 bug on windows, given that on Linux doesn't seem to appear.

lovettchris, 2021-06-04:
Lockstep has been fixed in latest AirSim bits, can you try again using these fixes. Thanks.