microsoft/AirSim #3422: build own executables failed with CMake on ubuntu

Astaxuqichao, 2021-02-25:
## Question
### What's your question?

I try to use Airsim-api to control car.But i follow the docs and write a CMakeLists.txt, it build failed.

### Include context on what you are trying to achieve

The code is copied from HelloCar example.It is my CMakelist:
```
cmake_minimum_required(VERSION 3.11.0)
project(test)

set(AIRSIM_ROOT /home/zhilu/oproj/AirSim-1.4.0-linux)
set(CMAKE_CXX_COMPILER /usr/bin/g++)

if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 11)
endif()

include_directories(
    ${AIRSIM_ROOT}/AirLib/deps/eigen3
    ${AIRSIM_ROOT}/AirLib/deps/rpclib/include
    ${AIRSIM_ROOT}/AirLib/include
    ${AIRSIM_ROOT}/AirLib/deps/MavLinkCom/include
    include
)

link_directories(
    # ${AIRSIM_ROOT}/AirLib/deps/MavLinkCom/lib/
    # ${AIRSIM_ROOT}/AirLib/deps/rpclib/lib/
    ${AIRSIM_ROOT}/AirLib/lib/x64/Release/
)

add_executable(test test.cpp)
target_link_libraries(test 
    AirLib
    rpc
    MavLinkCom)
```
There is error:
```
/home/zhilu/oproj/AirSim-1.4.0-linux/AirLib/lib/x64/Release/librpc.a(client.cc.o): In function `client':
/home/zhilu/oproj/AirSim-1.4.0-linux/external/rpclib/rpclib-2.2.1/lib/rpc/client.cc:174: undefined reference to `std::__1::to_string(int)'
/home/zhilu/oproj/AirSim-1.4.0-linux/AirLib/lib/x64/Release/librpc.a(client.cc.o): In function `clmdep_asio::ip::basic_resolver_query<clmdep_asio::ip::tcp>::basic_resolver_query(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, clmdep_asio::ip::resolver_query_base::flags)':
/home/zhilu/oproj/AirSim-1.4.0-linux/external/rpclib/rpclib-2.2.1/dependencies/include/asio/ip/basic_resolver_query.hpp:150: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)'
/home/zhilu/oproj/AirSim-1.4.0-linux/external/rpclib/rpclib-2.2.1/dependencies/include/asio/ip/basic_resolver_query.hpp:151: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)'
/home/zhilu/oproj/AirSim-1.4.0-linux/AirLib/lib/x64/Release/librpc.a(client.cc.o): In function `std::__1::error_code::error_code()':
```
#### Context details

OS: ubuntu 18.04
Arisim version : 1.4.0

### Include details of what you already did to find answers

jonyMarino, 2021-03-19:
Hi @Astaxuqichao! And welcome!
I'm getting the same errors.
Did you find any solution?

Astaxuqichao, 2021-03-27:
> Hi @Astaxuqichao! And welcome!
> I'm getting the same errors.
> Did you find any solution?

I solute it. I am  following  https://github.com/kevindehecker/airsim-test.
You can try this.
If something compile wrong , you can note it in CmakeList of  Airsim(I member that is MavLinkComTest)
I will close the issue.

closed 2021-03-27 by Astaxuqichao

jonyMarino, 2021-04-07:
Thanks, the root cause was because you were using g++, but AirSim is compiled with clang.